pvtom/rscp2mqtt #4: Feature Request: Set limits - Higher Values for Supporting PRO Series

Treasy79, 2022-10-12:
Is it possible to implement higher Values for Set limits ?
The Pro Series supports up to 12000W in Charging and Discharging.

The same whit the manual Loading, there are also 12000Wh supported.

pvtom, 2022-10-14:
Implemented with release v2.2

closed 2022-10-14 by pvtom

Treasy79, 2022-10-18:
After the upgrade I can still only set the max 4500, all other values are not taken. I can see the custom regex line that allows the higher value, still it doesn't work.

Is there anything special to consider when upgrading?

pvtom, 2022-10-20:
The call mosquitto_pub -t "e3dc/set/power_limits" -m true sets the mode in which values can be specified. Then you can set the values e.g. with mosquitto_pub -t "e3dc/set/max_charge_power" -m 4000. Observe the output value to e3dc/ems/max_charge/status. "0" means the value fits, S10 accepts the value, "1" the value is in the not recommended range but is accepted, "-1" an invalid value was passed (e.g. higher than 4500 in case of a non-PRO serie), "-2" for set currently not possible. What value do you get in the different ranges allowed by your device? If I call mosquitto_pub -t "e3dc/set/power_limits" -m false, the S10 automatically sets the power value back to 4500 watts (e3dc/ems/max_charge/power).
Prerequisites: Have you checked that the program runs only once? (call pkill before starting the newly compiled program).

pvtom, 2022-10-21:
It looks like the old version is still active. Did you stop the program before? Please use the command pkill rscp2mqtt for this. Then follow the steps from the readme and start the program with ./rscp2mqtt -d.

Treasy79, 2022-10-21:
Problem solved :-) There were something wrong with the copy of the new compiled file to the /opt/rscp2mqtt directory.

Now also values bigger than 4500 are accepted and executed.

THX a lot for your support!