pvtom/rscp2mqtt #5: Feature Request: Support of REQ_SET_POWER

Treasy79, 2022-10-18:
Support of REQ_SET_POWER commands to control the charging/discharging of the battery storage directly.

Presumably REQ_SET_POWER_MODE and REQ_SET_POWER_VALUE should be sent in one container.
I am available for testing if needed

Treasy79, 2022-10-21:
Hi Thomas,

if it's ok for you I can make a fork and provide you then with a pull request for this Feature. Then not all the workload is on your side for that. Or are you already implementing it?

I'm working on a Module for my Home-Automation to connect to the E3DC via RSCP by using your RSCP2MQTT as a "gateway" since there is no RSCP implementation directly in php. And  these commands give an excellent possibility to directly control the loading/unloading process from the home automation system.

BR Philipp

Treasy79, 2022-10-21:
Hi Thomas,

I have added the following lines:
#define PAYLOAD_REGEX_MODE     "(^0$|^1$|^2$|^3$|^4$)"

{ TAG_EMS_REQ_SET_POWER_MODE, TAG_EMS_REQ_SET_POWER, "e3dc/set/power_mode", PAYLOAD_REGEX_MODE, "", "", "", "", RSCP::eTypeUChar8, true },
    { TAG_EMS_REQ_SET_POWER_VALUE, TAG_EMS_REQ_SET_POWER, "e3dc/set/power_value", PAYLOAD_REGEX, "", "", "", "", RSCP::eTypeInt32, true },

to the RscpMqttMapping.h file.

But I always get an Error from TAG_EMS_SET_POWER:
: Tag 0x01800030 received error code 3.

I tried different settings with both commands, but always that same error message.

Perhaps you have any Idea? Or more possibilities to debug on the RSCP Side?

Best regards

Philipp

pvtom, 2022-10-22:
Hi Philipp,
the implementation is correct. Problem is that both tags have to be written into the container, as you have already guessed. You can try it out by increasing the interval in the config file to e.g. 5 seconds. Then you have more time to set both tags in one cycle. The result is that the E3DC device goes into the desired mode, but only for a few seconds, after which it switches back to the normal mode. For a longer effect you have to set the mode and the value again and again. I am currently developing a solution for this.
Best regards
Thomas

Treasy79, 2022-10-22:
Hello Thomas,

that's right, I tried it and it works. I have now a method which simply sets both values directly after each other. I was just a bit confused, because even if I set Mode 0 (automatic) the Value Tag is expected, which actually does nothing. Same with mode 1 (idle), there the value is actually also no matter.

But great that you want to implement a "clean" solution for this. From my point of view the best would be to have an address where you can pass mode and value as payload at once, or how do you think about that? And for the Value we should have a new regex, which accepts values from 0 - 30000.

That the command remains only for a short time and must be set thus last repeatedly has however the advantage that if the house automation system has once an error/failure, the E3DC goes back after a few seconds into the completely normal operating mode, and not evtl. small load values then last until the problem recovery remains.
And this command is the only value how I can regulate the charging of the battery independently. Especially also during the day in hours with low electricity prices, in times where you do not reach 100% autarky, all PV income into the battery and the house consumption from the grid to be able to use the battery in hours with high electricity prices.

Greetings

Philipp

pvtom, 2022-10-23:
The solution is currently being tested. It will go like this (in one call :-) ):
mosquitto_pub -t "e3dc/set/power_mode" -m "auto"
mosquitto_pub -t "e3dc/set/power_mode" -m "idle:60" (60 seconds in idle mode)
mosquitto_pub -t "e3dc/set/power_mode" -m "charge:2000:60" (2000 watt, 60 seconds in charge mode)
... same for "discharge" and "grid_charge".

Treasy79, 2022-10-23:
Hi Thomas,

ok so you will repeat the instruction for the defined time ( in your example 60s), and must not be requested from my side all few seconds? But will it be possible to send a new request within that time with a new charge Value?

BR Philipp

Treasy79, 2022-10-23:
Short other question, do you know if there is a newer decimation of the RSCP Tags than the file from 24.03.16? I wonder because in that file it is described that the set_power instruction must be repeated at least every 30 seconds, but my tests shows something of about 10 seconds to return to auto mode.

So I think that there were perhaps some changes since 2016 ...

pvtom, 2022-10-24:
Hi Philipp,
yes, it will be possible to send new requests within the time.
No, I don't know a newer documentation. But I saw that user nischram uses further tags starting with TAG_SE_*. Maybe he has more informations.
Best regards Thomas

pvtom, 2022-10-25:
Done with release v2.3.
Please consider the new key "AUTO_REFRESH" in the .config file.

closed 2022-10-25 by pvtom

Treasy79, 2022-10-25:
Hi Thomas,

thx for that fast implementation. 
One short question concerning the AUTO_REFRESH: If this is false, are the commands then still functional, only without automatic repeat? Or are then the 4 commands not usable at all?

Philipp

pvtom, 2022-10-25:
Hi Philipp,
AUTO_REFRESH=false means, the 4 commands aren't usable. If you like to have "no refresh" use 1 cycle in the calls. But you will see that the E3DC system isn't very fast to switch between the modes.
Regards Thomas

Treasy79, 2022-10-25:
Hi Thomas,

ok thx for the clarification. 
I will test it and include it in my module.

Regards Philipp

Treasy79, 2022-10-26:
Hi Thomas,

may I ask you to include the following line into your Mapping?
{ 0, TAG_EMS_SET_POWER, "e3dc/ems/set_power/power", "%i", "", RSCP::eTypeInt32, 1, false },

It's the automatic Response of the req_set_power command.

Thx a a lot

Philipp

pvtom, 2022-10-26:
Hi Philipp,
I have added the tag.
Something else: It is interesting that the "auto" mode behaves exactly like "idle". The power also goes down to 0 before the power is regulated again.
Best regards
Thomas

Treasy79, 2022-10-27:
Hi Thomas,

yes I noted this also with the auto-mode, because I had always some grid consumption this night, even if the Automode was set. This happened because i process the commands in Auto & Idle mode all 1 Minutes. So every Minute the Batterie discharge went down to 0 to increase a few seconds later again...I check now if I already have Auto mode in place and then skip the command.

But would it be possible on your side, that if you receive the auto mode command, you only delete eventually existing cycles of a prior set_power command and do not forward anything to the E3DC ? Then nothings happens if you are already in Auto mode, and if you are in an other mode the system go back on its own to auto mode after a few seconds the no new command arrives.

BR
Philipp

pvtom, 2022-10-27:
Hi Philipp,
good idea.
Please load the file RscpMqttMain.cpp into an editor and delete line 105 "it->done = false;", store the file and recompile the software. "auto" will set cycle to "0" and won't send anything to the E3DC device.
If it's okay, I will change it in the repository.
Best regards
Thomas