pvtom/rscp2mqtt #7: enable discharge lock?

aquadrat2217, 2023-02-13:
Hi Thomas,

thanks for that nice project. 
I would like to block battery discharge when my car is being charged. Is there a set command to lock battery discharged? 
Or do you have another idea how to solve this?

BR Andreas

Treasy79, 2023-02-13:
Hi Andreas,

You can set the power_Mode to "idle" then the battery will not charge or decharge.

It's described in the Readme under "Power Management"

Regards
Philipp

aquadrat2217, 2023-02-13:
Hi Philipp,

thanks, I don't know if I understood the cycles correctly.  Does it mean that then e.g. every 5s the idle mode is written to memory?

BR Andreas

Treasy79, 2023-02-13:
Hi Andreas,

the Power_mode commands are not persistent in the EMS of the E3DC, that means the EMS requires an update within rd 10s otherwise it returns back to "Normal" Mode. (which is pretty good in case of an "breakdown" of the sending Home-Automation system, to prevent an uncontrolled status). 

That you don't have to send the command all few seconds, if you know you want to set that Mode for an longer time, Thomas has implemented that cycles, where you can define how often (count of the cycles) RSCP2MQTT will repeat the command towards the E3DC without the need of an new command from your side. The cycle duration depends on/is equal the INTERVAL Parameter from the Configuration File.

Example: If your Interval in the Settings is set to 5 Seconds and you send the Idel Command with 20 cycles, the the Command is automatically send to the E3DC 20 times with an break of 5 seconds between each command.

I Hope the explanation helps.

Regards
Philipp

aquadrat2217, 2023-02-13:
Hi, 
Thanks for the good explanation,
got it right. Are that many write cycles critical to memory? Do you know what kind of memory is used here? or is it maybe
better to reduce the maximum discharge power to the
minimum?

BR Andreas

pvtom, 2023-02-14:
Hi Andreas,
the cycle number is stored in an integer variable. The tool rscp2mqtt queries the values of the E3DC device every seconds configured in the INTERVAL (.config file). If the idle function is set, i.e. the value is greater than 0, the tool will send the idle command to the E3DC device in addition to the usual queries and then decrement the variable with each query until the value is 0. So there is no memory problem with the functionality. In the "idle phase" a few additional bytes are sent to the E3DC. With "auto" the sending of the "idle" command can be cancelled.
Another possibility is to set the maximum discharge power, maybe a value of 0 is possible or to increase the value of the battery emergency power reserve.
Regards Thomas

aquadrat2217, 2023-02-14:
Hi Thomas,

Thanks for the info. The emergency power reserve is also a good idea. Let's see what I implemented. Thank you again.

Best regards Andreas

pvtom, 2023-07-23:
The new release v3.6 supports idle periods (Lade-/Entlade-Sperrzeiten)... maybe the best way to lock charging / discharging ;-)

aquadrat2217, 2023-07-24:
Hi Thomas, thanks for the info.  I updated it right away and it works - very nice.  Do you know what kind of memory the data is stored in?  Can there be problems with too many write accesses?  Greetings Andreas

pvtom, 2023-07-24:
Hi Andreas, 
the memory is allocated at program start and reused afterwards. The number of calls has no influence on the memory consumption. A change in memory usage could not be observed in my tests, which ran for a few days.
Best regards Thomas

aquadrat2217, 2023-07-24:
Hi Thomas, I meant what kind of memory e.g.  flash or volatile memory such as RAM.  I'm worried if the E3DC flash memory is designed for so many writes.  Do you have any information about this?  Greetings Andreas

pvtom, 2023-07-25:
Hi Andreas, I don't know the hardware architecture of the E3DC devices, so unfortunately I can't say anything about the memory technologies used. rscp2mqtt only queries the values, generates mainly network load for the E3DC. The E3DC device itself generates a lot of data by measuring power values, etc. You would need to ask E3DC if this data is stored in flash memory or possibly in a cloud. rscp2mqtt does not store any data (except some logging information) on disk (on a Raspberry Pi on an SD card), but keeps the data in RAM. The MQTT broker probably stores its data only in RAM, unless it is in retain mode). An Influxdb will certainly write the data to disk.
I hope I could help a little bit?
Best regards Thomas

aquadrat2217, 2023-07-25:
Hi Thomas, thanks for your detailed explanation.  It'll be fine.  It's good that you can now block charging and discharging of the battery.  Best regards Andrew