pvtom/rscp2mqtt #8: problem when switching true / false does not work

hagene1505, 2023-02-25:
Hello,

Thank you for the bridge and the good documentation.
I have a problem switching the power limits in W works.

However, if I want to switch the power limits or the weather regulation, it doesn't work.

What can be the reason?

`e3dc/set/max_charge_power` -> works
`e3dc/set/weather_regulation` -> **it does not work**
`e3dc/set/power_limits` -> **it does not work**

I haven't tested more yet.

Thanks and Regards

P.S. Sorry for my bad English, German would be better.

pvtom, 2023-02-25:
Hallo,

bei mir funktionieren die Aufrufe für power_limits und weather_regulation.

Hast Du evtl. auch das Dashboard (pvtom/rscp2mqtt-dashboard) laufen?

Bei mir ist es so, dass das Dashboard die Werte immer wieder anpasst, d.h. Werte, die ich mit mosquitto_pub geändert habe, werden automatisch auf den Wert zurückgesetzt, der der Schalterposition im Dashboard entspricht.

Beobachten kann man das z.B. mit 
mosquitto_sub -h mqtt-broker-server -p 1883 -t "e3dc/ems/weather_regulation"

Gruß
Thomas

hagene1505, 2023-02-25:
Hallo Thomas,

ahhh auf deutsch, prima.

Danke nochmal für die tolle Bridge. Ich habe nur diese, ohne Dashboard am laufen. Und es geht alles außer die Änderungen bei Werten von true / false?

Vermutlich liegt es aber an IP Symcon, ich sende aus dem internen MQTT Server. Wenn ich z.B. den MQTT Explorer verwende klappt es einwandfrei. Sehr komisch mal wieder.
Liegt aber dann nicht an deiner Bridge. Mal sehen ob ich irgendwie herausfinden kann woran es liegt das die einen gehen und die anderen nicht.

Viele Grüße
Stephan

pvtom, 2023-02-25:
Hallo Stephan,
man kann statt true auch 1 und statt false auch 0 nehmen.
Das versteht die Bridge auch. Vielleicht hilft Dir das weiter.
Gruß Thomas

hagene1505, 2023-02-25:
Hi Thomas,

Weltklasse das war es. Du hast meinen Tag / Abend gerettet.
Vielen Dank für deine Arbeit, die Bridge ist echt ein Brett.

Schönen Abend und viele Grüße
Stephan

closed 2023-02-25 by hagene1505

Treasy79, 2023-02-25:
Hallo Stephan,

Ich habe für Symcon gerade ein Modul für die RSCP2MQTT fertig gestellt. 

Ich werde das Modul die kommenden Tage als Beta bereitstellen.

VG
Philipp

hagene1505, 2023-02-25:
Hi Philipp,

das ist ja der Wahnsinn. Super, vielen herzlichen Dank. Da freue ich mich drauf.
Dann warte ich da noch drauf.

Grüße
Stephan

hagene1505, 2023-03-01:
Hallo Philipp,

kannst du schon sagen wann du das Modul für IP Symbol veröffentlichen wirst?
Ich warte schon gespannt .

Freut mich schon, viele Grüße
Stephan

Treasy79, 2023-03-01:
Hi, wurde soeben als Beta veröffentlicht.

@pvtom : Danke für die Unterstützung mit den SET_POWER Funktionen!

Grüße 
Philipp

Treasy79, 2023-03-03:
Hallo Thomas,

kannst du dir das bitte noch mal anschauen?
Wenn ein true gesendet wird, wird es von der Bridge auch als empfangen angezeigt, nur passiert nichts???

[5052B4CC-624E-48D3-A68A-F417EF984AB2.png]

Ich baue das sonst bei mir auch auf 0 und 1 um, ich bin mir aber sicher dass die Befehle beim Testen von mir schon erfolgreich mit true und false ausgeführt wurden.

Vielen Dank & Grüße

Philipp

pvtom, 2023-03-03:
Hallo Philipp,
ich habe gerade bei strahlendem Sonnenschein die max. Ladeleistung reduziert (mit true), hat sofort reagiert.
Probier's mal mit 1 und 0 aus. Ist ja ein bisschen komisch das Ganze.
Gruß
Thomas

hagene1505, 2023-03-03:
Hi Thomas,
Hi Philipp,

bei mir geht es eben aus IP Symcon heraus nur mit 1 und 0 warum auch immer. 

Im MQTT explorer klappt es mit true und false, daher dachte ich das es vielleicht an IPS liegt?

Spricht eigentlich was gegen die Änderung auf 1 und 0?

Grüße
Stephan

Treasy79, 2023-03-03:
Hi Stephan,

Ich hab ein Update eingestellt.
Hab auf 1 und 0 gewechselt.

Viele Grüße
Philipp

hagene1505, 2023-03-03:
Hi Philipp,

vielen Dank schon mal, werde es testen.

Grüße
Stephan