pvtom/rscp2mqtt #9: Data from yesterday and the current week, month and year - Add Power (external Source) Missing

Treasy79, 2023-03-08:
Hallo Thomas,

bei den Historischen Werten fehlen bei allen Zeiträumen der Wert für die externe Quelle.
Kannst du das noch ergänzen?

Danke & Grüße

Philipp

pvtom, 2023-03-09:
Hallo Philipp,

die Dokumentation der Tags ist nicht gut an der Stelle. Die Namen der Tags für die historischen Werte lauten oft ein bisschen anders als die Tags zu den zugehörigen Power-Werten. So finde ich kein "Energie-Tag" für ADD.
Was es gibt sind Tags wie TAG_DB_PM_0_POWER und TAG_DB_PM_1_POWER. Evtl. sind das die richtigen.
In der Datei RscpMqttMapping.h sind entsprechende Zeilen für heute und gestern enthalten (Zeilen 119,120,131,132), allerdings inaktiv, da auskommentiert. Wenn Du möchtest, könntest Du die Zeilen aktivieren, die Bridge neu kompilieren und es ausprobieren.
Bei positivem Ergebnis, passe ich die Bridge dann an, auch für Phasen Woche, Monat und Jahr.

Gruß Thomas

closed 2023-05-05 by pvtom

Treasy79, 2023-11-14:
Hallo Thomas,

TAG_DB_PM_1_POWER entspricht bei mir dem ADD also externe Produktion. Allerdings kommen die Werte nicht wie bei dir im Mapping definiert in kWh sondern in Wh. Kannst du das bitte noch mal checken, ob da ein Fehler mit dem Faktor ist.

Danke 
Philipp

Treasy79, 2024-11-12:
Hallo Thomas,

wäre es möglich dass du TAG_DB_PM_1_POWER für den Zusatzwechselrichter wieder in deinen Code aufnimmst?
Hatte damals bis auf den Faktor der Einheit bei mir funktioniert.

Danke & Grüße
Philipp

pvtom, 2024-11-15:
Hallo Philipp,
klar, mach ich. Ich weiß auch gar nicht mehr, warum ich die Zeilen rausgenommen habe.

Wenn Du willst, kannst Du folgende Zeilen in die Datei RscpMqttMapping.h einbauen und testen, ob es funktioniert:

    // CONTAINER TAG_DB_HISTORY_DATA_...
// DEV Issue #9
    { TAG_DB_HISTORY_DATA_DAY, TAG_DB_PM_0_POWER, 0, "pm_0/energy", "", F_FLOAT_2, UNIT_KWH, 1000, 0, false, false, false },
    { TAG_DB_HISTORY_DATA_DAY, TAG_DB_PM_1_POWER, 0, "pm_1/energy", "", F_FLOAT_2, UNIT_KWH, 1000, 0, false, false, false },

Bzgl. kWh oder Wh: Der aus dem Hauskraftwerk ausgelesene Wert wird durch 1000 geteilt und hat als Einheit kWh.
Passt das?

Gruß
Thomas

pvtom, 2025-02-02:
Mit Release v3.32 werden die Werte wieder ausgegeben.
Defauftmäßig werden die Werte, die das Hauskraftwerk liefert, durch 1000 geteilt und als kWh betrachtet.
Wenn das nicht passt, kann man das Verhalten (pro PM) in der .config anpassen:

#-Addon Power Meter
#--correct unit (Wh or kWh) or divisor (1 or 1000) if the values do not fit
CORRECT_PM_0_UNIT=Wh
CORRECT_PM_0_DIVISOR=1
CORRECT_PM_1_UNIT=kWh
CORRECT_PM_1_DIVISOR=1000