pvtom/rscp2mqtt #10: feat | Logging to stdout / stderr

goebelmeier, 2023-03-16:
Hi @pvtom,
vielen Dank für das coole Projekt. Ich würde rscp2mqtt gerne als Container bereitstellen um es als Add-On in Home Assistant OS bereitzustellen. Voraussetzung für das Deployment in Containern ist, dass das Logging nicht in ein Logfile geschieht, sondern nach `stdout` und `stderr`. Warum das so ist wird erklärt in https://12factor.net/logs / https://12factor.net/de/logs 
Wäre es ein großer Aufwand, nach `stdout` und `stderr` zu loggen, wenn zum Beispiel `LOGFILE=` leer gelassen wird?

martinholters, 2023-03-17:
>  Ich würde rscp2mqtt gerne als Container bereitstellen

https://hub.docker.com/r/martinholters/rscp2mqtt ? (Löst allerdings nicht das Logfile-Problem, wobei neben den Ausgaben in das Logfile aber sowieso schon sehr ausführliche Ausgaben nach stdout (oder stderr?) erfolgen.)

goebelmeier, 2023-03-18:
> > Ich würde rscp2mqtt gerne als Container bereitstellen
> 
> https://hub.docker.com/r/martinholters/rscp2mqtt ? (Löst allerdings nicht das Logfile-Problem, wobei neben den Ausgaben in das Logfile aber sowieso schon sehr ausführliche Ausgaben nach stdout (oder stderr?) erfolgen.)

Ah das ist schon mal ganz schick. Für Home Assistant benötige ich noch ein wenig Framework drumherum, insbesondere da ich die Config Optionen im Home Assistant UI konfigurierbar machen möchte und dann on-the-fly die `.config` erstelle. Zudem kann Home Assistant die Daten wie mqtt Host, Username und Passwort dynamisch bereitstellen.

Das mit dem Logging ist mir auch schon aufgefallen, es wird relativ verbose nach stdout geloggt und nur ganz wenig landet zusätzlich im Logfile. Das könnte man schicker homogenisieren und ggf Log Level einbauen. Alle Values die in MQTT geschrieben werden zusätzlich loggen ist in meinen Augen im Normalbetrieb überflüssig.

pvtom, 2023-03-18:
Hallo zusammen,
ich habe gerade das Logging umgestellt.
Wenn stdout umgeleitet wird (pipe), werden nur wenige Informationen ausgegeben. Das Logfile wird dann nicht mehr geschrieben.
Gruß Thomas

martinholters, 2023-03-19:
Ok, ich habe https://hub.docker.com/r/martinholters/rscp2mqtt auf 2.0.8 aktualisiert.

@goebelmeier bzgl. der Konfiguration mach gern ein Issue in https://github.com/martinholters/rscp2mqtt-docker auf, dann können wir dort diskutieren, welche Anpassungen an dem Container-Image sinnvoll sind.

closed 2023-05-05 by pvtom