pvtom/rscp2mqtt #19: missing some Wallboxdata

lorbetzki, 2023-05-19:
Thank you for including the wallboxdata. 

but I am missing two Datapoints. 

To access them, we need to set "TAG_EMS_BATTERY_BEVORE_CAR_MODE" and "TAG_EMS_BATTERY_TO_CAR_MODE". Here we can enter the discharge level of the battery in sunmode. Maybe you have an clu how we get the data? 

The 2nd Datapoint is to dis/allow to discharge the battery in mixed mode.
Best regards
[image]

pvtom, 2023-05-21:
Hi Kristian,
I have added two new topics to get/set the additional values. I hope this is the solution.

e3dc/wallbox/battery_discharge_until
e3dc/set/wallbox/battery_discharge_until
e3dc/wallbox/enforce_power_assignment
e3dc/set/wallbox/enforce_power_assignment

Best regards
Thomas

lorbetzki, 2023-05-21:
thank you,

```
e3dc/wallbox/battery_discharge_until
e3dc/set/wallbox/battery_discharge_until
```
this both runs as expectet, 

```
e3dc/wallbox/enforce_power_assignment
```
it looks like true and false are reversed, if i setup on the E3DC "erlaubt" which means allow, i get "false" flag from Mqtt, and on "unterbunden" which means disallow a "true" flag.

```
e3dc/set/wallbox/enforce_power_assignment
```
i can only send a false flag. if i'm sending true, nothing happens. 

[image]

pvtom, 2023-05-22:
Hi Kristian,

kannst Du bitte die Zeile in der Datei RscpMqttMapping.h von
    { 0, TAG_EMS_REQ_SET_WALLBOX_ENFORCE_POWER_ASSIGNMENT, "e3dc/set/wallbox/enforce_power_assignment", "^true|on|1$", "1", "^false|off|0$", "0", "", RSCP::eTypeBool, -1, true },
in
    { 0, TAG_EMS_REQ_SET_WALLBOX_ENFORCE_POWER_ASSIGNMENT, "e3dc/set/wallbox/enforce_power_assignment", "^true|on|1$", "true", "^false|off|0$", "false", "", RSCP::eTypeBool, -1, true },
ändern und erneut testen?

Wenn es funktioniert, haben "true" und "false" logisch die gleiche "verdrehte" Aussage wie bei e3dc/wallbox/enforce_power_assignment?

Gruß
Thomas

lorbetzki, 2023-05-22:
Moin,

```
{ 0, TAG_EMS_REQ_SET_WALLBOX_ENFORCE_POWER_ASSIGNMENT, "e3dc/set/wallbox/enforce_power_assignment", "^true|on|1$", "true", "^false|off|0$", "false", "", RSCP::eTypeBool, -1, true },
```
passt, ich kann damit schreiben.

> Wenn es funktioniert, haben "true" und "false" logisch die gleiche "verdrehte" Aussage wie bei e3dc/wallbox/enforce_power_assignment?
ja, wenn ich ein True sende, ist der button bei "unterbunden", bei false auf "erlaubt". Drücke ich am Speicher erhalte ich auch entsprechende Rückmeldung.
Viele Grüße

pvtom, 2023-05-22:
Das ist prima.
Die Logik würde ich ungern drehen.
Idee wäre, die Topics umzubenennen.
Würden z.B. "e3dc/wallbox/disable_battery_at_mix_mode" und "e3dc/set/wallbox/disable_battery_at_mix_mode" passen?

closed 2023-05-25 by pvtom