pvtom/rscp2mqtt #21: Probleme mit Battery_to_car

lorbetzki, 2023-05-31:
Hi,

```   { 0, TAG_EMS_REQ_SET_BATTERY_BEFORE_CAR_MODE, "e3dc/set/wallbox/battery_before_car", "^true|on|1$", "1", "^false|off|0$", "0", "", RSCP::eTypeUChar8, -1, true },
```
ich habe noch Probleme mit dieser Zeile. Wenn ich am Speicher die Einstellung ändere bekomme ich
0 = zuerst Wallbox
1 = zuerst Hausbatterie

Das gleiche Ergebnis, erhalte ich, wenn ich über das E3DC Portal schalte.

Sende ich über MQTT eine 0 wird auf "zuerst Wallbox" geschaltet.
Sende ich  über MQTT eine 1 wird auch auf "zuerst Hausbatterie" geschaltet. ABER. Leider nicht immer. Ich kann manchmal mehrmals hintereinander umschalten, manchmal bleibt aber der Wert immer bei "zuerst Wallbox". In diesem Zustand komme ich nur raus, wenn ich am Speicher oder per Portal auf "zuerst Hausbatterie" schalte. 

Danach klappen die Umschaltvorgänge meistens wieder für kurze zeit. Hast Du da eine idee?

lorbetzki, 2023-06-01:
Hi,

ok, es ist KEIN Problem. 

Es besteht eine Wechselwirkung zwischen 
TAG_EMS_REQ_SET_BATTERY_BEFORE_CAR_MODE und TAG_EMS_REQ_SET_BATTERY_TO_CAR_MODE

Um TAG_EMS_REQ_SET_BATTERY_BEFORE_CAR_MODE einstellen zu können, muss auf false.

[image]

Ich denke es reicht ein Hinweis für die Readme oder wie siehst du das?

pvtom, 2023-06-01:
Hi Kristian,
gut, dass es kein Fehler im rscp2mqtt ist :-)
Ich denke auch, dass es reicht, einen Hinweis ins Readme aufzunehmen.
Ich werde das bei einem nächsten Update machen.
Gruß Thomas

closed 2023-06-23 by pvtom