pvtom/rscp2mqtt #22: Logging: Error: Tag 0x05840000 received error code 6.

dhelleberg, 2023-06-25:
Hi,
thanks for providing this great tool. I'm running it for a couple of hours and noted the following log entry:

```
Error: Tag 0x05840000 received error code 6.
{"code":"invalid","message":"unable to parse 'e3dc,topic=e3dc/system/software value=S10_2022_046': invalid boolean\nunable to parse 'e3dc,topic=e3dc/system/production_date value=KW46 2020': invalid boolean\nunable to parse 'e3dc,topic=e3dc/system/serial_number value=XXXXXXXX': invalid boolean\nunable to parse 'e3dc,topic=e3dc/time/zone value=Europe/Berlin': invalid boolean"}{"code":"invalid","message":"unable to parse 'e3dc,topic=e3dc/battery/name value=LG_BAT': invalid boolean"}[2023-06-25 22:09:19] pid=1 ppid=0 RscpMqttMain.cpp(789) Error: Tag 0x05840000 received error code 6.
```
I removed the serial number from the bug report. It doesn't seem to hurt much as it seems to provide most of the data I was looking for. But maybe you want to investigate / fix it .
In case of any questions, please let me know.

pvtom, 2023-06-26:
Hallo Dominik,

0x05840000 ist der TAG_PM_DATA - Tag.

Über ihn kann man vom Power Meter Leistungs-, Spannungs- und Stromwerte der drei Phasen abfragen.

Ich habe im Netz gelesen, dass es Fälle gibt, z.B. beim S10 Mini, dass das Power Meter extern sitzt und mit einem anderen Index abgefragt werden muss, mit Index = 6 statt mit Index = 0.

Welches S10-Gerät hast Du im Einsatz?

Kannst Du in der Datei RscpMqttMain.cpp bitte Zeile 501

protocol.appendValue(&PMContainer, TAG_PM_INDEX, (uint8_t)0);

in

protocol.appendValue(&PMContainer, TAG_PM_INDEX, (uint8_t)6);

ändern, das Paket neu kompilieren und das Verhalten testen?

Gruß
Thomas

dhelleberg, 2023-06-26:
Hi,
ich hab einen S10 Mini mit externem Power Meter. Der Patch scheint das Problem zu lösen, ich sehe zumindest die Fehlermeldung nicht mehr im stdout nachdem ich neu kompiliert hab.
Hoffe das hilt eine config oder flag darein zu bauen für das S10 Mini
Vielen Dank schonmal! Dominik

pvtom, 2023-07-23:
Fixed by release v3.6.
Please set the new parameter PM_EXTERN=true in your .config file.

closed 2023-07-23 by pvtom