pvtom/rscp2mqtt #29: Question: Aktive Phasen der Wallbox setzen?

Dirk070, 2023-11-10:
Hallo Thomas,

besteht die Möglichkeit, die Menge der Phasen (1 oder 3) zu setzen?
Ich nutze eine E3/DC Multi Connect Wallbox und schalte bisher manuell auf 1 Phase für die Überschussladung und auf 3 Phasen im Mischmodus.

Danke vorab und schöne Grüße
Dirk

pvtom, 2023-11-12:
Hallo Dirk,
ich habe ein Tag für die RSCP-Schnittstelle gefunden, mit dem man die Phasen voraussichtlich steuern kann.
Leider habe ich keine E3/DC-Wallbox, so dass ich die Funktionalität nicht testen kann.
Ich bringe demnächst ein neues Release raus. Da ist der "Phasenschalter" für die Wallbox dann enthalten.

Gruß
Thomas

Dirk070, 2023-11-12:
Hallo Thomas,

super, vielen Dank.
Ich teste natürlich gerne und gebe dann Feedback. Ich habe ein Pro, falls Du diesbezüglich etwas testen möchtest, gib gerne Bescheid.

Viele Grüße 
Dirk

Dirk070, 2023-11-14:
Hallo Thomas,

sofern es nicht zu aufwändig wird, lässt sich die Position des Schlüsselschalters an der Wallbox auslesen?

Anwendungsfall: der Schalter kann entweder die Wallbox sperren oder zwischen Sonnen- und Mischmodus schalten.
Letzteres habe ich eingestellt. Wenn es schnell gehen muss, drehe ich den Schlüssel und bin im Mischmodus. Dann muss ich bisher per App (my3E) von 1 auf 3 Phasen schalten und schalte noch die Entladung der Akkus im Wallbox-Betrieb ab. So wird dann mit 22kW geladen und der Akku nicht leergesaugt.

Könnte man die Schalterstellung auslesen, könnte ich die Umschaltungen (Phase und Akkunutzung) automatisieren.

Danke und schöne Grüße
Dirk

Dirk070, 2023-11-14:
Hallo Thomas,

perfekt, vielen Dank. Ich gebe dann natürlich Rückmeldung.

Viele Grüße
Dirk

pvtom, 2023-11-26:
Hallo Dirk,
das neue Release v3.8 steht jetzt bereit.
In der Datei WALLBOX.md ist beschrieben, wie man die Phasenzahl setzen kann.
Gruß Thomas

Dirk070, 2023-11-28:
Hallo Thomas,

die Position des Schlüsselschalters wird ebenfalls korrekt erkannt, 1 für Mischmodus und 0 für Sonnenmodus (analog, wenn der Schalter zum Sperren der Wallbox konfiguriert ist).

Nochmals vielen Dank für die schnelle Ergänzung!!

Viele Grüße
Dirk

Dirk070, 2023-12-15:
Läuft perfekt, vielen Dank!!!

closed 2023-12-15 by Dirk070