pvtom/rscp2mqtt #31: AES Password ?

tognellap, 2023-11-22:
Hi together

I am just preparing the config file and wondering what i have to do with the AES Password. Do i have to encrypt my password?

Regards

Patrick

pvtom, 2023-11-22:
Hi Patrick,
you have to configure the RSCP password in your device (Personalisieren / Benutzerprofil).

[image]

Use the same password in the .config file for the attribute E3DC_AES_PASSWORD.

Best regards
Thomas

tognellap, 2023-11-22:
Thanks that means the Password in the config is the is the Password from the Portal?

pvtom, 2023-11-22:
No, it's a different password you have to create in the S10 device directly.
Please fill the text field "RSCP-Passwort (min. 6 Zeichen)".
.config:
E3DC_PASSWORD= (Portal password)
E3DC_AES_PASSWORD= (RSCP password)

Best regards
Thomas

tognellap, 2023-11-22:
Thanks that helps, regards from Switzerland,  Patrick

closed 2023-11-22 by tognellap

timmost, 2023-12-10:
Small add on - password needs to be 6 Char length min, otherwise E3DC adds some chars... making it invalid... (see docs).