pvtom/rscp2mqtt #32: Connection Error

tognellap, 2023-11-23:
Hi Thomas

Ich habe die Punkte von dir gestern umgestetzt. Ich habe einen Docker Container auf meiner Synology am laufen und erhalte nach dem Start die folgnede Ausgabe im Terminal

Beim Start habe ich folgendes im Log:
[image]

Ich bekomme dann aber Connection Error:

[image]

Mich erstaunt die Meldung **cannot be converted.1.230**. Dies sieht irgendwie wie Teile der IP

Im Config habe ich folgendes:

E3DC_IP=192.168.001.230
E3DC_PORT=5033
E3DC_USER=xxx
E3DC_PASSWORD=xxx
E3DC_AES_PASSWORD=xxx
MQTT_HOST=192.168.1.10
MQTT_PORT=1883
MQTT_AUTH=false
MQTT_USER=
MQTT_PASSWORD=
MQTT_QOS=0
MQTT_RETAIN=false
ENABLE_INFLUXDB=true
DISABLE_MQTT_PUBLISH=false
INFLUXDB_HOST=localhost
INFLUXDB_PORT=8086
INFLUXDB_VERSION=2
INFLUXDB_MEASUREMENT=e3dc
INFLUXDB_1_DB=e3dc
INFLUXDB_1_AUTH=false
INFLUXDB_1_USER=
INFLUXDB_1_PASSWORD=
INFLUXDB_2_ORGA=tognellacom
INFLUXDB_2_BUCKET=smartHome
INFLUXDB_2_TOKEN=xxx==
PREFIX=e3dc
LOGFILE=/tmp/rscp2mqtt.log
INTERVAL=1
PVI_REQUESTS=true
PVI_TRACKER=2
PM_REQUESTS=true
PM_EXTERN=false
DCB_REQUESTS=true
AUTO_REFRESH=false
WALLBOX=true
FORCE_PUB=e3dc/[a-z]+/power
FORCE_PUB=e3dc/battery/rsoc
FORCE_PUB=e3dc/pvi/.*/string_[1,2]

Ich sehe den Fehler nicht. Ich habe eine S10X mit Zusatzwechselrichter

pvtom, 2023-11-23:
Hallo Patrick,
ich denke, die IP-Adresse muss 192.168.1.230 sein und nicht 192.168.**00**1.230.
Lösch mal die zwei Nullen raus.
Gruß Thomas

tognellap, 2023-11-23:
Hallo Thomas
Dies habe ich auch schon gemacht. Hat zum selben ergebnis geführt. Da war die Meldung

[image]

pvtom, 2023-11-23:
Hallo Patrick,

hmm, gute Frage. Die Überprüfung der IP-Adresse auf ein gültiges Format scheitert.
Was komisch aussieht, ist, dass die Portnummer 5033 in der Log-Ausgabe ganz vorne in der Ausgabe steht, statt der Jahreszahl, wie bei den anderen Log-Ausgaben.
Kann es sein, dass irgendwo ein oder mehrere nicht sichtbare Steuerzeichen in der .config stehen?
Kannst Du versuchen, die .config neu zu erstellen?

Gruß
Thomas

pvtom, 2023-11-26:
Hallo Patrick,
das neue Release v3.8 steht jetzt bereit.
Ich habe eine zusätzliche Syntax-Prüfung für die IP-Adresse eingebaut.
Gruß Thomas

tognellap, 2023-11-26:
Hallo Thomas

Ich habe die Verbindung jetzt hinbekommen war warscheinlich wirklich die Problematik mit den Steuerzeichen.

Jetzt bekomme ich einfach noch den folgenden Fehler im Log:
[image]

Liebe Grüsse Patrick

closed 2024-01-12 by pvtom