pvtom/rscp2mqtt #33: Connection issue "RscpMqttMain.cpp(1143) RSCP authentitication level 0"

timmost, 2023-12-10:
Hi Thomas,
vielen Dank für Dein Modul (die Anleitung von e3DC ist ja ziemlich wild...)

Allerdings passt was mit meiner Connection noch nicht. Das Ganze läuft unter Docker compose. 
- Die Initiale Verbindung klappt
- Ich bekomme auch zwei MQTT Meldungen (Battery, Grid)
- Aber dann ist Ruhe...

Any hints? Vermute mal irgendwas in Richtung authorization oder reset connection?
VG timmo

Ich bekomme im Portainer log folgende Meldungen:
**Log:**

/rscp2mqtt [v3.8]
Connecting...
E3DC system >192.168.178.58:5033< user: >my.name<
MQTT broker >localhost:1883< qos = 0 retain = false
MQTT prefix: >e3dcrscp <
History year 2017 to 2023
Fetching data every 10 seconds.
Requesting PVI data = true (2 strings/trackers)
Requesting PM data = true
Requesting DCB (battery module) data = true
Battery SOC limiter active = false
Wallbox support = true
Auto refresh mode = false
Stdout to terminal

[2023-12-10 13:22:44] pid=1 ppid=0 RscpMqttMain.cpp(1970) Connecting to server 192.168.178.58:5033
[2023-12-10 13:22:44] pid=1 ppid=0 RscpMqttMain.cpp(1977) Success: E3DC connected.
[2023-12-10 13:22:44] pid=1 ppid=0 RscpMqttMain.cpp(725) Request authentication (1702210964)
[2023-12-10 13:22:46] pid=1 ppid=0 RscpMqttMain.cpp(1143) RSCP authentitication level 0
[2023-12-10 13:22:46] pid=1 ppid=0 RscpMqttMain.cpp(1622) Connecting to broker localhost:1883
[2023-12-10 13:22:46] pid=1 ppid=0 RscpMqttMain.cpp(1633) Error: Connection failed.
[2023-12-10 13:22:56] pid=1 ppid=0 RscpMqttMain.cpp(725) Request authentication (1702210976)
[2023-12-10 13:22:59] pid=1 ppid=0 RscpMqttMain.cpp(1143) RSCP authentitication level 0
[2023-12-10 13:22:59] pid=1 ppid=0 RscpMqttMain.cpp(1622) Connecting to broker localhost:1883
[2023-12-10 13:22:59] pid=1 ppid=0 RscpMqttMain.cpp(1631) Success: MQTT broker connected.
MQTT publish topic >e3dcrscp /battery/state< payload >EMPTY< unit ><
[2023-12-10 13:22:59] pid=1 ppid=0 RscpMqttMain.cpp(266) MQTT: starting listener loop
MQTT publish topic >e3dcrscp /grid/state< payload >IN< unit ><
[2023-12-10 13:23:09] pid=1 ppid=0 RscpMqttMain.cpp(725) Request authentication (1702210989)
[2023-12-10 13:23:12] pid=1 ppid=0 RscpMqttMain.cpp(1511) Error: Response receive timeout (retry)
[2023-12-10 13:23:22] pid=1 ppid=0 RscpMqttMain.cpp(725) Request authentication (1702211002)
[2023-12-10 13:23:22] pid=1 ppid=0 RscpMqttMain.cpp(1143) RSCP authentitication level 0
[2023-12-10 13:23:32] pid=1 ppid=0 RscpMqttMain.cpp(725) Request authentication (1702211012)
[2023-12-10 13:23:32] pid=1 ppid=0 RscpMqttMain.cpp(1143) RSCP authentitication level 0
[2023-12-10 13:23:42] pid=1 ppid=0 RscpMqttMain.cpp(725) Request authentication (1702211022)
[2023-12-10 13:23:42] pid=1 ppid=0 RscpMqttMain.cpp(1143) RSCP authentitication level 0
[2023-12-10 13:23:52] pid=1 ppid=0 RscpMqttMain.cpp(725) Request authentication (1702211032)
[2023-12-10 13:23:52] pid=1 ppid=0 RscpMqttMain.cpp(1143) RSCP authentitication level 0
[2023-12-10 13:24:02] pid=1 ppid=0 RscpMqttMain.cpp(725) Request authentication (1702211042)
[2023-12-10 13:24:02] pid=1 ppid=0 RscpMqttMain.cpp(1143) RSCP authentitication level 0
[2023-12-10 13:24:12] pid=1 ppid=0 RscpMqttMain.cpp(725) Request authentication (1702211052)
[2023-12-10 13:24:12] pid=1 ppid=0 RscpMqttMain.cpp(1143) RSCP authentitication level 0
[2023-12-10 13:24:22] pid=1 ppid=0 RscpMqttMain.cpp(725) Request authentication (1702211062)
[2023-12-10 13:24:22] pid=1 ppid=0 RscpMqttMain.cpp(1143) RSCP authentitication level 0

(etc.)

**.config File:**

 // IP address of the E3/DC S10 device
E3DC_IP=192.168.178.58
// Port of the E3/DC S10 device, default is 5033
E3DC_PORT=5033
// E3/DC account
E3DC_USER=my.name
E3DC_PASSWORD=my.PW
// AES password
E3DC_AES_PASSWORD=my.AESPW
// Target MQTT broker
MQTT_HOST=localhost
// Default port is 1883
MQTT_PORT=1883
// MQTT user / password authentication necessary? Depends on the MQTT broker configuration.
MQTT_AUTH=false
// if true, then enter here
MQTT_USER=
MQTT_PASSWORD=
// MQTT parameters
MQTT_QOS=0
MQTT_RETAIN=false
// Topic prefix (max 24 characters), default is e3dc
PREFIX=e3dcrscp
// log file
LOGFILE=/tmp/rscp2mqtt.log
// history start year (installation of the E3/DC device)
HISTORY_START_YEAR=2017
// log file for selected topics
TOPIC_LOG_FILE=/tmp/rscp2mqtt.history
// selected topics for logging (regular expressions)
TOPIC_LOG=e3dc/history/.*
TOPIC_LOG=e3dc/system/software
// Interval requesting the E3/DC S10 device in seconds (1..10)
INTERVAL=10
// enable PVI requests, default is true
PVI_REQUESTS=true
// number of available PV strings(trackers), default is 2
PVI_TRACKER=2
// enable PM requests, default is true
PM_REQUESTS=true
// use external PM (S10 Mini), default is false
PM_EXTERN=false
// enable DCB (details for battery modules) requests, default is true
DCB_REQUESTS=true
// enable battery SOC limiter, default is false
SOC_LIMITER=false
// Auto refresh, default is false
AUTO_REFRESH=false
// Disable MQTT publish support (dryrun mode)
DISABLE_MQTT_PUBLISH=false
// Wallbox, default is false
WALLBOX=true
// topics to be published in each cycle (regular expressions)
FORCE_PUB=e3dc/[a-z]+/power
FORCE_PUB=e3dc/battery/soc

**Docker Compose**

  rscp2mqtt:
    container_name: "rscp2mqtt"
    image: pvtom/rscp2mqtt:latest-with-influxdb
    restart: unless-stopped
    environment:
    - TZ=Europe/Berlin
    volumes:
    - ./volumes/rscp2mqtt/.config:/opt/rscp2mqtt/.config
    network_mode: "host"
    stdin_open: true
    tty: true

pvtom, 2023-12-10:
Hallo Timmo,

"RSCP authentitication level  0" bedeutet, dass der Login zum Hauskraftwerk nicht erfolgreich war.

Die beiden Topics, die gesendet werden, enthalten einen Startwert, der aber nicht vom Hauskrafktwerk kommt.

Gruß
Thomas

timmost, 2023-12-10:
Hi Thomas
Danke - Du hast recht - ich habe das Passwort vom "neuen" e3DC Portal eingegeben. Das alte Portal PW ist das richtige. 
VG timmo

closed 2023-12-10 by timmost