pvtom/rscp2mqtt #34: Influxdb2 - Definition von topic

timmost, 2023-12-11:
Hi Tom
noch eine Frage - bei der Configuration von Influxdb2 schreibt das Modul einfach alle Daten in die DB (Klappt prima). 

Gibt es eine Möglichkeit, hier nur ausgewählte Topics (entlang von FORCEPUB - aber halt als PUB2INFLUX) zu schreiben - oder empfiehlst Du, dann über den MQTT Broker zu gehen, und Telegraf die Auswahl zu überlssen?

Viele Grüße - Timmo

pvtom, 2023-12-12:
Hallo Timmo,
bisher kann man die Topics, die zur InfluxDB gehen, nicht einschränken. Das bedarf einer Anpassung der Software.
Die Idee ist aber super, da viele Daten übertragen werden, die man wahrscheinlich nicht in der InfluxDB haben möchte.
Gruß
Thomas

timmost, 2023-12-12:
Hi Thomas,

Hatte ich schon gedacht - dann mache ich das über MQTT und Telegraf, der
dann nur die paar Daten in Influxdb schreibt, die notwendig sind.

So wird's ein bisserl viel auf die Dauer für den Raspi - finde es spannend,
ein System aufzubauen, dass dann auch unbetreut ein paar Jahre läuft, da
muss man die Langfristfolgen im Blick halten… (könnte - schaffe nicht, die
Finger weg zu lassen)

closed 2023-12-12 by timmost

pvtom, 2023-12-23:
Hallo Timmo,
ich habe gerade Release v3.9 rausgebracht. Damit kann man die Topics, die an die InfluxDB gehen, einschränken.
In der .config kann dazu der Parameter INFLUXDB_TOPIC (auch mehrfach) benutzt werden.
Die Topics können mit der Regex-Syntax beschrieben werden. Siehe INFLUXDB.md
Gruß Thomas

timmost, 2023-12-24:
Cool, werde ich ausprobieren, vielen Dank und schöne Feiertage
Viele Grüße - Timmo

Timmo Sturm
Wittelsbacherstr. 8
85622 Feldkirchen
+49 151 2762 1280

timmost, 2023-12-28:
Hi Tom, 
funktioniert prima. 
Meine Config für InfluxDB2 - mal schauen, was wich wirklich anschaue. Insbesondere die Tageswerte sind für mich interessant. 

VG timmo

ENABLE_INFLUXDB=true
INFLUXDB_HOST=localhost
INFLUXDB_PORT=8086
INFLUXDB_VERSION=2
INFLUXDB_MEASUREMENT=e3dc
INFLUXDB_2_ORGA=home
INFLUXDB_2_BUCKET=e3dc_new
INFLUXDB_2_TOKEN="mein Token"
INFLUXDB_TOPIC=e3dc/[a-z]+/power
INFLUXDB_TOPIC=e3dc/solar/energy
INFLUXDB_TOPIC=e3dc/battery/soc
INFLUXDB_TOPIC=e3dc/battery/energy/discharge
INFLUXDB_TOPIC=e3dc/battery/energy/charge
INFLUXDB_TOPIC=e3dc/home/energy
INFLUXDB_TOPIC=e3dc/grid/energy/in
INFLUXDB_TOPIC=e3dc/grid/energy/out
INFLUXDB_TOPIC=e3dc/yesterday/battery/energy/charge
INFLUXDB_TOPIC=e3dc/yesterday/battery/energy/discharge
INFLUXDB_TOPIC=e3dc/yesterday/solar/energy
INFLUXDB_TOPIC=e3dc/yesterday/grid/energy/in
INFLUXDB_TOPIC=e3dc/yesterday/grid/energy/out
INFLUXDB_TOPIC=e3dc/yesterday/home/energy
INFLUXDB_TOPIC=e3dc/yesterday/autarky

timmost, 2023-12-28:
Noch eine Frage:
da ich unnötige Systemlast reduzieren will, würde ich gerne die MQTT Meldungen minimieren. Bin mir aber nicht sicher, wo man das macht - einfach die MQTT credentials löschen?

VG Timmo

pvtom, 2023-12-28:
Hallo Timmo,
Du kannst über folgende Parameter in der .config die Last reduzieren:
PVI_REQUESTS=false
PM_REQUESTS=false
DCB_REQUESTS=false
Damit werden dann weniger Werte vom Hauskraftwerk abgefragt.
Du kannst mit 
DISABLE_MQTT_PUBLISH=true
den gesamten Traffic zum MQTT Broker abschalten.

Gruß
Thomas