pvtom/rscp2mqtt #35: Make ClientID configurable

docolli, 2023-12-16:
I had problems to add rscp2mqtt to my FHEM installation via the build-in MQTT2 server. Finally I could do so after changing your mosquito connection string, which generates at every start a new random ClientID, to this fixed string (line 1621 in RscpMqttMain.cpp):

`mosq = mosquitto_new("rscp2mqtt", true, NULL);`

It would be fine, if it is possible to set a fixed ClientID in the config file.

pvtom, 2023-12-23:
Hi docolli,
release 3.9 is available. Now it's possible to configure the parameter MQTT_CLIENT_ID in the config file.
Best regards
Thomas

docolli, 2023-12-27:
Thank you! 
I did an implementation of this by myself after my request. I am not a good coder, so I compared what you did and how my solution looks like to learn. 
In my solution I did not need to use an extra flag `fix_mqtt_client_id` to have an argument for the necessary if-condition when connection is done. 
Here is how I did it:

```
    strcpy(cfg.mqtt_host, "");
    cfg.mqtt_port = MQTT_PORT_DEFAULT;
    strcpy(cfg.mqtt_cid, "");
    cfg.mqtt_auth = false;
    cfg.mqtt_qos = 0;
```

```
        // MQTT connection
        if (!mosq) {
            if (cfg.mqtt_cid=="") {
				mosq = mosquitto_new(NULL, true, NULL);
			} else {
				mosq = mosquitto_new(cfg.mqtt_cid, true, NULL);
			}
            logMessage(cfg.logfile, (char *)__FILE__, __LINE__, (char *)"Connecting to broker %s:%u\n", cfg.mqtt_host, cfg.mqtt_port);

```

pvtom, 2023-12-27:
Hi docolli,
correct, the flag fix_mqtt_client_id isn't really necessary. The comparison with an empty character string would be sufficient.
But your implementation isn't okay. If you like to compare two strings, you have to use strcmp in C, not "=="!
Best regards
Thomas

docolli, 2023-12-27:
Thanks for this hint!
However, it did compile without any error and I think, when I tested it without ClientID configured in the config file, the ClientID was generated randomly, so the code worked.
As I said, I am not very experienced in coding, so I checked on the web now and found this, at least for C++ the "==" is also possible to compare strings and leads to the same machine code: https://stackoverflow.com/questions/9158894/differences-between-c-string-and-compare

closed 2023-12-30 by pvtom