pvtom/rscp2mqtt #37: Semantics

JensHauser, 2023-12-20:
First of all - I love this piece of software :-)

No issues, just some perceptions beyond the scope of the well-functional programming.
Would be interesting to receive feedback about them.

**It´s about the meaning behind the parameters (semantics)**

The E3DC RSCP doc+examples (found the last tag update from 2016) give a little insights about semantics, but it seems to me that E3DC doesn´t want them to be self-explanatory.

1. It took a while to understand that "e3dc/grid/energy/in" seems to be the daily energy to/from grid.
2. I´d propose to enrich the documentation with a column like "readable" / "writable". 
3. First values of interest for me were the power values. Some kind of grouping, beyond the E3DC tags, would be appreciated.
4. Some semantics would be helpful. Say, what is the meaning behind the parameters
5. Typical behaviour of the parameter values (e.g. like in the docs - "e3dc/mode" give good hints, but what are the possible values?) Currently, I tested emergency power and this value raised to 10. On normal days it alternates between 0..1..2. Please understand this as an example. I guess, some experiences could be helpful for the community.

Maybe there is already some deeper pieces of information available somewhere to avoid reinventing the wheel?

If someone is interested in my contribution, let me know (along my personal learning curve with)

pvtom, 2023-12-23:
Hi Jens,
today I launched release 3.9 and I tried to improve the documentation as well :-)
1. "e3dc/grid/energy/in" means the energy to the grid "Einspeisung" of the current day ("out" means "from the grid" (Netzbezug))
2., 4. Take a look to TOPICS.md. Hope, it's a bit clearer now.
5. Bad news: I also don't know the possible values. The documentation from E3/DC (Excel in the sample program package) isn't perfect... :-(

Best regards
Thomas

JensHauser, 2023-12-24:
Installed and working like a charm :-)
Thanks for updating the docs

closed 2023-12-24 by JensHauser