pvtom/rscp2mqtt #38: Historical data

JensHauser, 2023-12-20:
Daily, weekly, yearly historica data - how to manage this?
Can I give a specific date or a range?

pvtom, 2023-12-23:
Hi Jens,

as mentioned in TOPICS.md. energy, autarky and SOC values are from today, yesterday, the current week, the current month and the current year.
You can configure the year of the installation of your device. rscp2mqtt will generate topics for past years e.g. "e3dc/history/2022/autarky".

Best regards
Thomas

JensHauser, 2023-12-24:
Thanks a lot.
I just asked myself, if the historical data of last [x] days, weeks would be possible to be grabbed :-)

closed 2023-12-24 by JensHauser

zutewas, 2024-01-01:
Thanks for working on rscp2mqtt. 

It is possible to implement history data for months of the past years.

e.g. like topics
e3dc/history/2022/01/autarky
e3dc/history/2022/01/battery/energy/charge
e3dc/history/2022/01/consumed
e3dc/history/2022/01/battery/energy/discharge
e3dc/history/2022/01/grid/energy/in
e3dc/history/2022/01/grid/energy/out
e3dc/history/2022/01/home/energy
e3dc/history/2022/01/solar/energy

A other possiblity could be to request a period e.g. via e3dc/set/force and the message "20220101-20220131" for special days or "202201" for this month or "202201-202203" for three months. The return similar like the yearly history.

For me it is important to see the past months to compare it with past years.