pvtom/rscp2mqtt #39: How to expand the code with other RSCP tags?

docolli, 2023-12-27:
I am also using rscpgui (https://github.com/rxhan/RSCPGui) to get familiar with RSCP communication to my E3DC S10E System. As you might know, this project has some more (maybe officially undocumented) tags implemented.
Escpecially I would be interested to to get some basic information about the installed battery capacity, etc.

```
    BAT_SPECIFIED_CAPACITY = 0x03800125
    BAT_SPECIFIED_DSCHARGE_POWER = 0x03800126
    BAT_SPECIFIED_CHARGE_POWER = 0x03800127
    BAT_SPECIFIED_MAX_DCB_COUNT = 0x03800128
    BAT_ROLE = 0x03800129
```

I already tried, but I failed, to implement other RSCP tags into your code.
My attempt was to define a new tag into RscpTags.h and then add a mapping into RscpMqttMapping.h, but I failed.

Can you help me?

pvtom, 2023-12-27:
Hi docolli,
the integration of additional tags is sometimes trial and error.
First you have to find out whether tags belong to a container or not. In this case, the tags belong to the "BAT_SPECIFICATION" container, which must be assigned to the "BAT_DATA" container. However, this is not documented.
I have succeeded and I will release the solution with a next release.

Best regards
Thomas

docolli, 2023-12-27:
Love to see the release soon, so I can compare and understand how you have added the topics.
Thanks for adding this so quick!

Best regards,
Oliver

closed 2024-01-09 by docolli