pvtom/rscp2mqtt #41: Setting reserve to 100% doesn't work

bondhook, 2024-01-07:
Hallo Thomas, 

wenn man bei der E3DC-Anlage die Reserve auf 100 % setzen möchte ("e3dc/set/reserve/percent 100"), dann reagiert die Anlage nicht. Bei Werten von 0 bis 99 funktioniert es. Ist das ein Bug bei rscp2mqtt oder liegt es an E3DC? 

Gruß, 
Oli

pvtom, 2024-01-07:
Hallo Oli,
die 100 lässt rscp2mqtt derzeit nicht zu. Ich ändere das zum nächsten Release.

Wenn Du willst, kannst Du es ausprobieren. Dazu Zeile 412 in der Datei RscpMqttMapping.h editieren:

{ TAG_SE_REQ_SET_EP_RESERVE, TAG_SE_PARAM_EP_RESERVE, "set/reserve/percent", PAYLOAD_REGEX_2_DIGIT, "", "", "", "", UNIT_PERCENT, RSCP::eTypeFloat32, -1, true },
in
{ TAG_SE_REQ_SET_EP_RESERVE, TAG_SE_PARAM_EP_RESERVE, "set/reserve/percent",PAYLOAD_REGEX_0_100, "", "", "", "", UNIT_PERCENT, RSCP::eTypeFloat32, -1, true },
ändern und neu komplilieren.

Gruß
Thomas

bondhook, 2024-01-07:
Hi Thomas, 

danke für die Info. 99% reicht mir und ich kann bis zum nächsten Release warten 😉.

Vielen Dank für das tolle Programm und die ganze Arbeit, die du geleistet hast 👍.

Gruß, 
Oli

pvtom, 2024-01-13:
fixed by v3.11