pvtom/rscp2mqtt #42: Change request interval of specific Tags

docolli, 2024-01-09:
Hallo Thomas,

wie kann man (im Code) das Abfrageinterval für best. Tags ändern? Ich nutze aktuell das EMS WB_DISCHARGE_BAT_UNTIL und setze dies per MQTT, um es synchron mit Werten aus der Autoladesoftware evcc zu halten.
Dabei ist mir aufgefallen, dass sich der MQTT Wert `e3dc/wallbox/battery_discharge_until` nach einem MQTT Set nicht ändert, da er wohl nur ab und zu (1x nach Start von rscp2mqtt ?) aktualisiert wird. 
Wenn der häufiger aktualisert werden würde, könnte ich auch eine Automatik aufsetzen, die bei Änderung dieses Wertes im E3DC System meine evcc Steuerung synchronisiert.

pvtom, 2024-01-09:
Hallo Oliver,
ich kann die Wallbox-Tags leider nicht testen, da ich (auch leider) keine E3/DC-Wallbox habe.
Alle Tags werden mit demselben Intervall abgefragt (Einstellung in .config).
Kann man den Wert im Display des S10 sehen? Ändert er sich da?
Eine Sache könnte aber auch im rscp2mqtt schief laufen: Wenn die S10 einen Fehler zu einem Wallbox-Tag meldet (warum auch immer), setzt rscp2mqtt den Wert cfg.wallbox auf false und fragt danach nie wieder die Wallbox-Tags ab, da dann angenommen wird, dass keine Wallbox angeschlossen ist. Vielleicht ist diese Entscheidung aber auch zu hart, da evtl. nur bestimmte Tags nicht unterstützt werden.
Du kannst, wenn Du die Version 3.10 einsetzt, mit mosquitto_pub -h <broker> -t "e3dc/set/health" -m 1 abfragen, welche Tags nicht unterstützt werden. rscp2mqtt schreibt diese Tags ins Log-File bzw. in die Logausgabe von systemd, je nachdem wie Du das Programm startest.
Wenn Du im Log Zeilen findest die "Container abc Tag xyz not supported" lauten, kann ich prüfen, ob da ein Wallbox-Tag dabei ist, das zum Abschalten der Wallbox-Abfragen führt...  Dann müsste ich rscp2mqtt diesbzgl. entschärfen.
Gruß
Thomas

pvtom, 2024-01-09:
..oder einfacher: Das Programm im Verbose-Mode starten mit rscp2mqtt -v und dann schauen, ob der "Error: Wallbox not available" erscheint :-)

docolli, 2024-01-09:
Kommst du auch ohne Wallbox im Display auf "Wallbox -> Ladepriorisierung"?
[image]
Das ist der Wert "Bis SOC".

docolli, 2024-01-09:
Aha, dein Hinweis auf das .config File war gut!
Ich denke ich sollte hier die Wallbox-Werte hinzufügen, die ich aktualisisert sehen will, oder? 

```
#-Forced Topics
#--will be published with every cycle
FORCE_PUB=e3dc/[a-z]+/power
FORCE_PUB=e3dc/battery/rsoc
FORCE_PUB=e3dc/pvi/.*/string_[1,2]
```

pvtom, 2024-01-09:
Prima, dass es klappt!
Das Default-Verhalten ist, nur bei Änderungen MQTT zu triggern.
Will man Daten in jedem Cycle pushen, muss der Topic mit FORCE_PUB konfiguriert werden.

docolli, 2024-01-09:
Okay, dann sollte ich aber eine Änderung des Wertes übers Display per MQTT mitbekommen, auch wenn FORCE_PUB nicht gesetzt ist? Hmm, ich muss in den Keller und das testen...

docolli, 2024-01-09:
Ich habe das Gefühl, dass es klappt, wenn ich per MQTT `e3dc/set/wallbox/battery_discharge_until` den Wert ändere, dann ändert er sich im E3DC Display und ich bekomme auch den neuen Wert per MQTT.
ABer es klappt nicht, wenn ich den Wert über das E3DC Display ändere. Teste ich aber systematisch erst morgen. Melde mich nochmals.
Danke bis dahin!

docolli, 2024-01-11:
Manchmal hilft 1-2 Tage warten und es nochmals probieren. Jetzt geht es wie erwartet. Auch wenn ich den Wert übers E3DC Display ändere bekomme ich nun es per MQTT mit.

docolli, 2024-01-28:
Muss mich wieder melden. Wollte heute mal einen Wert ändern, das klappt auch (sehe ich im Display vom HKW), aber der geänderte Wert kommt nicht mehr per MQTT zurück.
Folgende Einträge bekomme ich nach dem Setzen des MQTT health flags:

```
grep rscp syslog
Jan 28 19:54:41 fhem-heizung rscp2mqtt[12563]: [2024-01-28 19:54:41] pid=12563 ppid=1 RscpMqttMain.cpp(423) Topic >e3dc/system/software< marked for logging
Jan 28 19:54:41 fhem-heizung rscp2mqtt[12563]: [2024-01-28 19:54:41] pid=12563 ppid=1 RscpMqttMain.cpp(421) Topic >e3dc/solar/power< forced
Jan 28 19:54:41 fhem-heizung rscp2mqtt[12563]: [2024-01-28 19:54:41] pid=12563 ppid=1 RscpMqttMain.cpp(421) Topic >e3dc/battery/power< forced
Jan 28 19:54:41 fhem-heizung rscp2mqtt[12563]: [2024-01-28 19:54:41] pid=12563 ppid=1 RscpMqttMain.cpp(421) Topic >e3dc/home/power< forced
Jan 28 19:54:41 fhem-heizung rscp2mqtt[12563]: [2024-01-28 19:54:41] pid=12563 ppid=1 RscpMqttMain.cpp(421) Topic >e3dc/grid/power< forced
Jan 28 19:54:41 fhem-heizung rscp2mqtt[12563]: [2024-01-28 19:54:41] pid=12563 ppid=1 RscpMqttMain.cpp(421) Topic >e3dc/addon/power< forced
Jan 28 19:54:41 fhem-heizung rscp2mqtt[12563]: [2024-01-28 19:54:41] pid=12563 ppid=1 RscpMqttMain.cpp(423) Topic >e3dc/history/2022/battery/energy/charge< marked for logging
Jan 28 19:54:41 fhem-heizung rscp2mqtt[12563]: [2024-01-28 19:54:41] pid=12563 ppid=1 RscpMqttMain.cpp(423) Topic >e3dc/history/2022/battery/energy/discharge< marked for logging
Jan 28 19:54:41 fhem-heizung rscp2mqtt[12563]: [2024-01-28 19:54:41] pid=12563 ppid=1 RscpMqttMain.cpp(423) Topic >e3dc/history/2022/solar/energy< marked for logging
Jan 28 19:54:41 fhem-heizung rscp2mqtt[12563]: [2024-01-28 19:54:41] pid=12563 ppid=1 RscpMqttMain.cpp(423) Topic >e3dc/history/2022/grid/energy/in< marked for logging
Jan 28 19:54:41 fhem-heizung rscp2mqtt[12563]: [2024-01-28 19:54:41] pid=12563 ppid=1 RscpMqttMain.cpp(423) Topic >e3dc/history/2022/grid/energy/out< marked for logging
Jan 28 19:54:41 fhem-heizung rscp2mqtt[12563]: [2024-01-28 19:54:41] pid=12563 ppid=1 RscpMqttMain.cpp(423) Topic >e3dc/history/2022/home/energy< marked for logging
Jan 28 19:54:41 fhem-heizung rscp2mqtt[12563]: [2024-01-28 19:54:41] pid=12563 ppid=1 RscpMqttMain.cpp(423) Topic >e3dc/history/2022/consumed< marked for logging
Jan 28 19:54:41 fhem-heizung rscp2mqtt[12563]: [2024-01-28 19:54:41] pid=12563 ppid=1 RscpMqttMain.cpp(423) Topic >e3dc/history/2022/autarky< marked for logging
Jan 28 19:54:41 fhem-heizung rscp2mqtt[12563]: [2024-01-28 19:54:41] pid=12563 ppid=1 RscpMqttMain.cpp(423) Topic >e3dc/history/2023/battery/energy/charge< marked for logging
Jan 28 19:54:41 fhem-heizung rscp2mqtt[12563]: [2024-01-28 19:54:41] pid=12563 ppid=1 RscpMqttMain.cpp(423) Topic >e3dc/history/2023/battery/energy/discharge< marked for logging
Jan 28 19:54:41 fhem-heizung rscp2mqtt[12563]: [2024-01-28 19:54:41] pid=12563 ppid=1 RscpMqttMain.cpp(423) Topic >e3dc/history/2023/solar/energy< marked for logging
Jan 28 19:54:41 fhem-heizung rscp2mqtt[12563]: [2024-01-28 19:54:41] pid=12563 ppid=1 RscpMqttMain.cpp(423) Topic >e3dc/history/2023/grid/energy/in< marked for logging
Jan 28 19:54:41 fhem-heizung rscp2mqtt[12563]: [2024-01-28 19:54:41] pid=12563 ppid=1 RscpMqttMain.cpp(423) Topic >e3dc/history/2023/grid/energy/out< marked for logging
Jan 28 19:54:41 fhem-heizung rscp2mqtt[12563]: [2024-01-28 19:54:41] pid=12563 ppid=1 RscpMqttMain.cpp(423) Topic >e3dc/history/2023/home/energy< marked for logging
Jan 28 19:54:41 fhem-heizung rscp2mqtt[12563]: [2024-01-28 19:54:41] pid=12563 ppid=1 RscpMqttMain.cpp(423) Topic >e3dc/history/2023/consumed< marked for logging
Jan 28 19:54:41 fhem-heizung rscp2mqtt[12563]: [2024-01-28 19:54:41] pid=12563 ppid=1 RscpMqttMain.cpp(423) Topic >e3dc/history/2023/autarky< marked for logging
Jan 28 19:54:41 fhem-heizung rscp2mqtt[12563]: [2024-01-28 19:54:41] pid=12563 ppid=1 RscpMqttMain.cpp(421) Topic >e3dc/battery/rsoc< forced
Jan 28 19:54:41 fhem-heizung rscp2mqtt[12563]: [2024-01-28 19:54:41] pid=12563 ppid=1 RscpMqttMain.cpp(423) Topic >e3dc/battery/dcb/1/soh< marked for logging
Jan 28 19:54:41 fhem-heizung rscp2mqtt[12563]: [2024-01-28 19:54:41] pid=12563 ppid=1 RscpMqttMain.cpp(423) Topic >e3dc/battery/dcb/2/soh< marked for logging
Jan 28 19:54:41 fhem-heizung rscp2mqtt[12563]: [2024-01-28 19:54:41] pid=12563 ppid=1 RscpMqttMain.cpp(423) Topic >e3dc/battery/dcb/3/soh< marked for logging
Jan 28 19:54:41 fhem-heizung rscp2mqtt[12563]: [2024-01-28 19:54:41] pid=12563 ppid=1 RscpMqttMain.cpp(423) Topic >e3dc/battery/dcb/4/soh< marked for logging
Jan 28 19:54:41 fhem-heizung rscp2mqtt[12563]: [2024-01-28 19:54:41] pid=12563 ppid=1 RscpMqttMain.cpp(421) Topic >e3dc/pvi/power/string_1< forced
Jan 28 19:54:41 fhem-heizung rscp2mqtt[12563]: [2024-01-28 19:54:41] pid=12563 ppid=1 RscpMqttMain.cpp(421) Topic >e3dc/pvi/power/string_2< forced
Jan 28 19:54:41 fhem-heizung rscp2mqtt[12563]: [2024-01-28 19:54:41] pid=12563 ppid=1 RscpMqttMain.cpp(421) Topic >e3dc/pvi/voltage/string_1< forced
Jan 28 19:54:41 fhem-heizung rscp2mqtt[12563]: [2024-01-28 19:54:41] pid=12563 ppid=1 RscpMqttMain.cpp(421) Topic >e3dc/pvi/voltage/string_2< forced
Jan 28 19:54:41 fhem-heizung rscp2mqtt[12563]: [2024-01-28 19:54:41] pid=12563 ppid=1 RscpMqttMain.cpp(421) Topic >e3dc/pvi/current/string_1< forced
Jan 28 19:54:41 fhem-heizung rscp2mqtt[12563]: [2024-01-28 19:54:41] pid=12563 ppid=1 RscpMqttMain.cpp(421) Topic >e3dc/pvi/current/string_2< forced
[... cut ...]
Jan 28 19:54:41 fhem-heizung rscp2mqtt[12563]: [2024-01-28 19:54:41] pid=12563 ppid=1 RscpMqttMain.cpp(434) Container 0x06800400 Tag 0x06800006 in temporary cache.
Jan 28 19:54:41 fhem-heizung rscp2mqtt[12563]: [2024-01-28 19:54:41] pid=12563 ppid=1 RscpMqttMain.cpp(434) Container 0x06800400 Tag 0x06800007 in temporary cache.
Jan 28 19:54:41 fhem-heizung rscp2mqtt[12563]: [2024-01-28 19:54:41] pid=12563 ppid=1 RscpMqttMain.cpp(434) Container 0x06800400 Tag 0x0680000C in temporary cache.
Jan 28 19:54:41 fhem-heizung rscp2mqtt[12563]: [2024-01-28 19:54:41] pid=12563 ppid=1 RscpMqttMain.cpp(434) Container 0x06800400 Tag 0x0680000D in temporary cache.

```

docolli, 2024-01-28:
Zusatzinfos:

Nach Neustart von rscp2mqtt bekomme ich bis auf einen Tag "wallbox_power" neue Nachrichten:
[image]

Das Log zeigt nach Neustart folgende Einträge:
```
Jan 28 20:03:54 fhem-heizung rscp2mqtt[22308]: [2024-01-28 20:03:54] pid=22308 ppid=1 RscpMqttMain.cpp(423) Topic >e3dc/system/software< marked for logging
Jan 28 20:03:54 fhem-heizung rscp2mqtt[22308]: [2024-01-28 20:03:54] pid=22308 ppid=1 RscpMqttMain.cpp(421) Topic >e3dc/solar/power< forced
Jan 28 20:03:54 fhem-heizung rscp2mqtt[22308]: [2024-01-28 20:03:54] pid=22308 ppid=1 RscpMqttMain.cpp(421) Topic >e3dc/battery/power< forced
Jan 28 20:03:54 fhem-heizung rscp2mqtt[22308]: [2024-01-28 20:03:54] pid=22308 ppid=1 RscpMqttMain.cpp(421) Topic >e3dc/home/power< forced
Jan 28 20:03:54 fhem-heizung rscp2mqtt[22308]: [2024-01-28 20:03:54] pid=22308 ppid=1 RscpMqttMain.cpp(421) Topic >e3dc/grid/power< forced
Jan 28 20:03:54 fhem-heizung rscp2mqtt[22308]: [2024-01-28 20:03:54] pid=22308 ppid=1 RscpMqttMain.cpp(421) Topic >e3dc/addon/power< forced
Jan 28 20:03:54 fhem-heizung rscp2mqtt[22308]: [2024-01-28 20:03:54] pid=22308 ppid=1 RscpMqttMain.cpp(423) Topic >e3dc/history/2022/battery/energy/charge< marked for logging
Jan 28 20:03:54 fhem-heizung rscp2mqtt[22308]: [2024-01-28 20:03:54] pid=22308 ppid=1 RscpMqttMain.cpp(423) Topic >e3dc/history/2022/battery/energy/discharge< marked for logging
Jan 28 20:03:54 fhem-heizung rscp2mqtt[22308]: [2024-01-28 20:03:54] pid=22308 ppid=1 RscpMqttMain.cpp(423) Topic >e3dc/history/2022/solar/energy< marked for logging
Jan 28 20:03:54 fhem-heizung rscp2mqtt[22308]: [2024-01-28 20:03:54] pid=22308 ppid=1 RscpMqttMain.cpp(423) Topic >e3dc/history/2022/grid/energy/in< marked for logging
Jan 28 20:03:54 fhem-heizung rscp2mqtt[22308]: [2024-01-28 20:03:54] pid=22308 ppid=1 RscpMqttMain.cpp(423) Topic >e3dc/history/2022/grid/energy/out< marked for logging
Jan 28 20:03:54 fhem-heizung rscp2mqtt[22308]: [2024-01-28 20:03:54] pid=22308 ppid=1 RscpMqttMain.cpp(423) Topic >e3dc/history/2022/home/energy< marked for logging
Jan 28 20:03:54 fhem-heizung rscp2mqtt[22308]: [2024-01-28 20:03:54] pid=22308 ppid=1 RscpMqttMain.cpp(423) Topic >e3dc/history/2022/consumed< marked for logging
Jan 28 20:03:54 fhem-heizung rscp2mqtt[22308]: [2024-01-28 20:03:54] pid=22308 ppid=1 RscpMqttMain.cpp(423) Topic >e3dc/history/2022/autarky< marked for logging
Jan 28 20:03:54 fhem-heizung rscp2mqtt[22308]: [2024-01-28 20:03:54] pid=22308 ppid=1 RscpMqttMain.cpp(423) Topic >e3dc/history/2023/battery/energy/charge< marked for logging
Jan 28 20:03:54 fhem-heizung rscp2mqtt[22308]: [2024-01-28 20:03:54] pid=22308 ppid=1 RscpMqttMain.cpp(423) Topic >e3dc/history/2023/battery/energy/discharge< marked for logging
Jan 28 20:03:54 fhem-heizung rscp2mqtt[22308]: [2024-01-28 20:03:54] pid=22308 ppid=1 RscpMqttMain.cpp(423) Topic >e3dc/history/2023/solar/energy< marked for logging
Jan 28 20:03:54 fhem-heizung rscp2mqtt[22308]: [2024-01-28 20:03:54] pid=22308 ppid=1 RscpMqttMain.cpp(423) Topic >e3dc/history/2023/grid/energy/in< marked for logging
Jan 28 20:03:54 fhem-heizung rscp2mqtt[22308]: [2024-01-28 20:03:54] pid=22308 ppid=1 RscpMqttMain.cpp(423) Topic >e3dc/history/2023/grid/energy/out< marked for logging
Jan 28 20:03:54 fhem-heizung rscp2mqtt[22308]: [2024-01-28 20:03:54] pid=22308 ppid=1 RscpMqttMain.cpp(423) Topic >e3dc/history/2023/home/energy< marked for logging
Jan 28 20:03:54 fhem-heizung rscp2mqtt[22308]: [2024-01-28 20:03:54] pid=22308 ppid=1 RscpMqttMain.cpp(423) Topic >e3dc/history/2023/consumed< marked for logging
Jan 28 20:03:54 fhem-heizung rscp2mqtt[22308]: [2024-01-28 20:03:54] pid=22308 ppid=1 RscpMqttMain.cpp(423) Topic >e3dc/history/2023/autarky< marked for logging
Jan 28 20:03:54 fhem-heizung rscp2mqtt[22308]: [2024-01-28 20:03:54] pid=22308 ppid=1 RscpMqttMain.cpp(421) Topic >e3dc/battery/rsoc< forced
Jan 28 20:03:54 fhem-heizung rscp2mqtt[22308]: [2024-01-28 20:03:54] pid=22308 ppid=1 RscpMqttMain.cpp(423) Topic >e3dc/battery/dcb/1/soh< marked for logging
Jan 28 20:03:54 fhem-heizung rscp2mqtt[22308]: [2024-01-28 20:03:54] pid=22308 ppid=1 RscpMqttMain.cpp(423) Topic >e3dc/battery/dcb/2/soh< marked for logging
Jan 28 20:03:54 fhem-heizung rscp2mqtt[22308]: [2024-01-28 20:03:54] pid=22308 ppid=1 RscpMqttMain.cpp(423) Topic >e3dc/battery/dcb/3/soh< marked for logging
Jan 28 20:03:54 fhem-heizung rscp2mqtt[22308]: [2024-01-28 20:03:54] pid=22308 ppid=1 RscpMqttMain.cpp(423) Topic >e3dc/battery/dcb/4/soh< marked for logging
Jan 28 20:03:54 fhem-heizung rscp2mqtt[22308]: [2024-01-28 20:03:54] pid=22308 ppid=1 RscpMqttMain.cpp(421) Topic >e3dc/pvi/power/string_1< forced
Jan 28 20:03:54 fhem-heizung rscp2mqtt[22308]: [2024-01-28 20:03:54] pid=22308 ppid=1 RscpMqttMain.cpp(421) Topic >e3dc/pvi/power/string_2< forced
Jan 28 20:03:54 fhem-heizung rscp2mqtt[22308]: [2024-01-28 20:03:54] pid=22308 ppid=1 RscpMqttMain.cpp(421) Topic >e3dc/pvi/voltage/string_1< forced
Jan 28 20:03:54 fhem-heizung rscp2mqtt[22308]: [2024-01-28 20:03:54] pid=22308 ppid=1 RscpMqttMain.cpp(421) Topic >e3dc/pvi/voltage/string_2< forced
Jan 28 20:03:54 fhem-heizung rscp2mqtt[22308]: [2024-01-28 20:03:54] pid=22308 ppid=1 RscpMqttMain.cpp(421) Topic >e3dc/pvi/current/string_1< forced
Jan 28 20:03:54 fhem-heizung rscp2mqtt[22308]: [2024-01-28 20:03:54] pid=22308 ppid=1 RscpMqttMain.cpp(421) Topic >e3dc/pvi/current/string_2< forced
[... cut ...]
Jan 28 20:03:54 fhem-heizung rscp2mqtt[22308]: [2024-01-28 20:03:54] pid=22308 ppid=1 RscpMqttMain.cpp(434) Container 0x06800400 Tag 0x06800006 in temporary cache.
Jan 28 20:03:54 fhem-heizung rscp2mqtt[22308]: [2024-01-28 20:03:54] pid=22308 ppid=1 RscpMqttMain.cpp(434) Container 0x06800400 Tag 0x06800007 in temporary cache.
Jan 28 20:03:54 fhem-heizung rscp2mqtt[22308]: [2024-01-28 20:03:54] pid=22308 ppid=1 RscpMqttMain.cpp(434) Container 0x06800400 Tag 0x0680000C in temporary cache.
Jan 28 20:03:54 fhem-heizung rscp2mqtt[22308]: [2024-01-28 20:03:54] pid=22308 ppid=1 RscpMqttMain.cpp(434) Container 0x06800400 Tag 0x0680000D in temporary cache.

```

pvtom, 2024-01-28:
Guten Abend,

rscp2mqtt sendet nur diese Topics:
- "e3dc/wallbox/total/power" und
- "e3dc/wallbox/solar/power".

"e3dc/wallbox/power" gibt es nicht. Was mappst Du auf wallbox_power?

Gruß Thomas

docolli, 2024-01-28:
Danke für die rasche Rückmeldung!

Hast Recht, das Topic wallbox_power ist eine verwaistes Mapping (in meiner FHEM Installation), das muss ich löschen. Mein Fehler...

Dennoch haben sich "unsupported Tags" im Laufe der Zeit akkumuliert. Nach einem Neustart sind es definitiv weniger, wie du durch einen Vergleich der beiden Logfiles siehst.

pvtom, 2024-01-28:
Dass einige Tags vom HKW nicht beantwortet werden, kommt bei mir auch vor. Das ist meist aber nur temporär, z.B. kommt es vor, dass kurz nach Mitternacht die HISTORY-Tags nicht beantwortet werden. Später geht es dann wieder. Daher kann man die meisten in der Liste der "unsupported tags" wohl ignorieren.
Du kannst die Tags in der RscpTags.h nachschauen. Wenn diese Tags aber doch eigentlich funktionieren, kannst Du die Meldungen ignorieren.

pvtom, 2024-01-29:
Hallo,
ein Tipp noch. Aufgrund der Zeilennummern, die in Deinen Logs vorkommen, und einem Abgleich mit meinen Sourcen, gehe ich davon aus, dass Du die Version v3.10 im Einsatz hast. Stimmt das?
Wenn ja, bitte auf eine neuere Version upgraden... ab der v3.11 habe ich das Handling bei der Rückmeldung von Fehlern bei Wallbox-Tags geändert. Wenn es einen Fehler gibt, setzt die v3.10 (und auch ältere Versionen) die Wallbox-Abfragen ganz aus... das kann zu dem Verhalten führen, dass Du keine Wallbox-Daten mehr bekommst.
Gruß
Thomas