pvtom/rscp2mqtt #43: Automatic detection of the number of PVI trackers not detection 2nd PVI tracker

genezig, 2024-01-21:
Hi there,

I'm using rscp2mqtt with an S10E, latest firmware (Q4/2023). Until rscp2mqtt v3.9 everything works fine. But from 3.10 the second PVI tracker is no longer delivering values other than zero - 0 voltage, 0 current and 0 power on the 2nd string (which is forced to be published by "FORCE_PUB=e3dc/pvi/.*/string_[1,2]" in .config).

Since PVI_TRACKER in .config does not seem to work any longer, I'm missing the second string with >=v3.10. You probably aim to get the autodetection to work - but may I suggest: If PVI_TRACKER is not defined in .config, rscp2mqtt tries auto detection, but if it's defined, it overrides auto detection?

Your great work is very much appreciated!

pvtom, 2024-01-21:
Hello,
can you check which value is output for the topic "e3dc/pvi/used_string_count"?
If you have installed two strings, the value should be "2".
What do you get for "e3dc/pvi/energy_all/string_2"?
Best regards
Thomas

genezig, 2024-01-21:
Hi again,
> Hello, can you check which value is output for the topic "e3dc/pvi/used_string_count"?
Sure: the value is 1 (even though I have 2 strings)
> What do you get for "e3dc/pvi/energy_all/string_2"?
Only the topic "e3dc/pvi/energy_all/string_1" is published, with a plausible value.

pvtom, 2024-01-21:
Okay, then the auto detection is not working for you. The suggestion to override the detection is good. I will implement it.

genezig, 2024-01-21:
Thank you in advance. If I can do any research/testing to help solving the auto detection issue I would be glad to do so.

Quaksalber, 2024-01-21:
Good evening

 

Dear Thomas. I have the same problem. I was already on the roof the check the cabling 😊

 

e3dc/pvi/used_string_count = 1

 

and therefore, I only see one string:

 

e3dc/pvi/energy_all/string_1

 

Best regards from Switzerland

pvtom, 2024-01-21:
hopefully solved by the new release v3.12.

Quaksalber, 2024-01-21:
Lieber Thomas

 

Up and running! Ganz herzlichen Dank.

 

Grüsse

 

Matthias

genezig, 2024-01-22:
Installed v3.12 - setting PVI_TRACKER manually works perfectly. Just fyi: Now found the following error lines in the log:

`Jan 22 09:17:56 smarthome rscp2mqtt[4433]: [2024-01-22 09:17:56] pid=4433 ppid=1 RscpMqttMain.cpp(1501) Error: Container 0x02840000 Tag 0x02000060 received error code 6.`
`Jan 22 09:17:56 smarthome rscp2mqtt[4433]: [2024-01-22 09:17:56] pid=4433 ppid=1 RscpMqttMain.cpp(1501) Error: Container 0x02840000 Tag 0x02000070 received error code 6.`
`Jan 22 09:17:56 smarthome rscp2mqtt[4433]: [2024-01-22 09:17:56] pid=4433 ppid=1 RscpMqttMain.cpp(1501) Error: Container 0x02840000 Tag 0x02000080 received error code 6.`

They seem to appear only once on every start of rscp2mqtt (using systemd service). No further impact on the service discovered.

Many thanks and best regards!

pvtom, 2024-01-22:
Hi,
the messages mean that the requested tags TAG_PVI_REQ_COS_PHI, TAG_PVI_REQ_VOLTAGE_MONITORING and TAG_PVI_REQ_FREQUENCY_UNDER_OVER cannot be answered by your S10. This also occurs with my device. Apparently there are other devices or software versions that can do this. The messages only appear once at startup and can be ignored. rscp2mqtt then no longer queries these tags.
Best regards
Thomas

closed 2024-01-26 by pvtom