pvtom/rscp2mqtt #45: Add a Parameter for ID of external Powermeter

maaque, 2024-02-02:
Great software and exactly, what I needed. Many thanks.

The ID of the external PM  in my E3DC S10 is 4 and not 6. Therefore I had to change RscpMqttMain.cpp the line 
`protocol.appendValue(&PMContainer, TAG_PM_INDEX, (uint8_t)6);` to
`protocol.appendValue(&PMContainer, TAG_PM_INDEX, (uint8_t)4);`

It would be probably a good idea, to add a parameter e.g. PM_ID to the config file.

pvtom, 2024-02-02:
Hi maaque,
thank you for your feedback. 
It is exciting to see the differences between the S10 systems.
I will implement this as a configuration option in the next version.

Best regards
Thomas

maaque, 2024-02-02:
Perfekt

Viele Grüße 
Matthias

pvtom, 2024-02-10:
Fixed by release v3.14.

Please add the line to your .config:
PM_INDEX=4
and remove:
PM_EXTERN=true

closed 2024-02-10 by pvtom

maaque, 2024-02-10:
Works perfectly. Thanks