pvtom/rscp2mqtt #46: Multiple powermeters support

cwihne, 2024-02-06:
Apart from root consumption power meter (PM #0, ROOT), I run a second PM to explicitly measure heat pump consumption (PM #1, ADDITIONAL_CONSUMPTION).
Topic "pm" reports full PM #0 values, but PM #1 values are not exposed. Changing RscpMqttMain.cpp from
`protocol.appendValue(&PMContainer, TAG_PM_INDEX, (uint8_t)0);` to
`protocol.appendValue(&PMContainer, TAG_PM_INDEX, (uint8_t)1);`
shows PM#1 in pm/* topics.
Maybe a list of indexes in config file would allow cycling through all available powermeters, or those that are wanted to be queried?
Assuming pm_0 and pm_1 relate to the  two powermeters in my system, "pm_0/energy" and "pm_1/energy" topics are available, but no more. Both those energy values are 0.00 at all times.

pvtom, 2024-02-06:
Hello cwhine,
yes, good idea. I will implement this in the next release.
Best regards
Thomas

pvtom, 2024-02-10:
Fixed by release v3.14.

Please add the following lines to your .config file:
PM_INDEX=0
PM_INDEX=1

rs2cpmqtt will created topics like:
e3dc/pm/1/* for power meter 0
e3dc/pm/2/* for power meter 1

But maybe no changes regarding  "pm_0/energy" and "pm_1/energy" topics... I have no influence on that...

Best regards
Thomas

closed 2024-02-10 by pvtom

cwihne, 2024-02-17:
The applied extension works well with powermeters #0 and #1, and surely for more in the case of need. Thank you very much.