pvtom/rscp2mqtt #49: add wallbox historical data

kasi0815, 2024-02-16:
Hi, a great software which I use since some time.   :-)

Is it possible to add historical data of the wallbox like:  e3dc/yesterday/wallbox/energy

Maybe it's not just me who would really appreciate it. Many thanks

pvtom, 2024-02-17:
Hi kasi0815,

I have added the following topics with the new release v3.15 today:

Wallbox Energy All  e3dc/wallbox/total/energy [kWh]
Wallbox Energy Solar  e3dc/wallbox/solar/energy [kWh]
Wallbox Energy L1 	e3dc/wallbox/energy/L1 [kWh]
Wallbox Energy L2  3dc/wallbox/energy/L2 [kWh]
Wallbox Energy L3  e3dc/wallbox/energy/L3 [kWh]

Maybe that helps. I can't test the wallbox topics because I don't have an E3/DC wallbox installed :-(

I don't know for which time periods the determined energy values apply (last charging process?).
Perhaps you can share your experiences.

Best regards
Thomas

kasi0815, 2024-02-17:
What a great and super fast support! This weekend is for my family and I will test and feedback early next week.

Have a nice Sunday.

docolli, 2024-02-18:
Thanks for adding more wallbox data! Since I have a E3DC wallbox I can share my experience here:

```
wallbox_active_phases | 7 | 2024-02-17 19:42:34
wallbox_battery_before_car | 0 | 2024-02-17 19:42:34
wallbox_battery_discharge_until | 75 | 2024-02-17 19:42:34
wallbox_battery_to_car | 1 | 2024-02-17 19:42:34
wallbox_canceled | false | 2024-02-17 19:42:34
wallbox_charging | true | 2024-02-18 10:29:30
wallbox_disable_battery_at_mix_mode | true | 2024-02-17 19:42:34
wallbox_energy_L1 | 5 | 2024-02-18 11:37:00
wallbox_energy_L2 | 0 | 2024-02-17 19:42:34
wallbox_energy_L3 | 0 | 2024-02-17 19:42:34
wallbox_index | 0 | 2024-02-17 19:42:30
wallbox_key_state | 0 | 2024-02-17 19:42:34
wallbox_locked | false | 2024-02-17 19:42:34
wallbox_max_current | 20 | 2024-02-17 19:42:34
wallbox_number_phases | 1 | 2024-02-17 19:42:34
wallbox_number_used_phases | 1 | 2024-02-17 19:42:34
wallbox_plugged | true | 2024-02-18 10:29:19
wallbox_power_L1 | 4297 | 2024-02-18 11:45:16
wallbox_power_L2 | 0 | 2024-02-18 11:45:16
wallbox_power_L3 | 0 | 2024-02-18 11:45:16
wallbox_soc | 0 | 2024-02-17 19:42:34
wallbox_solar_energy | 34905 | 2024-02-18 11:45:16
wallbox_solar_power | 4269 | 2024-02-18 11:45:16
wallbox_sun_mode | false | 2024-02-17 19:42:34
wallbox_total_energy | 133667 | 2024-02-18 11:45:16
wallbox_total_power | 4297 | 2024-02-18 11:45:16
```

Currently I am loading and it seems that the `wallbox_energy_Lx` values are the energy in kWh (without decimals) for each phase for the actual loading session. 
[image]

However, I cannot match the values `wallbox_solar_energy` and `wallbox_total_energy` to any values in my E3DC portal.
[image]

docolli, 2024-02-18:
A little note: I know it can be remapped, be I would prefer the following MQTT topic names for solar and total energy/power to get better value grouping:

```
wallbox_energy_L1 | 8 | 2024-02-18 12:25:14
wallbox_energy_L2 | 0 | 2024-02-17 19:42:34
wallbox_energy_L3 | 0 | 2024-02-17 19:42:34
wallbox_energy_solar | 38141 | 2024-02-18 12:44:51
wallbox_energy_total | 136986 | 2024-02-18 12:44:51
...
wallbox_power_L1 | 2721 | 2024-02-18 12:46:44
wallbox_power_L2 | 8 | 2024-02-18 12:46:44
wallbox_power_L3 | 8 | 2024-02-18 12:46:44
wallbox_power_solar | 2736 | 2024-02-18 12:46:44
wallbox_power_total | 2734 | 2024-02-18 12:46:44
```

docolli, 2024-02-18:
Compiled the dev branch and tested it:
```
wallbox_active_phases | 7 | 2024-02-18 16:37:11
wallbox_battery_before_car | 0 | 2024-02-18 16:37:11
wallbox_battery_discharge_until | 75 | 2024-02-18 16:37:11
wallbox_battery_to_car | 1 | 2024-02-18 16:37:11
wallbox_canceled | false | 2024-02-18 16:37:13
wallbox_charging | false | 2024-02-18 16:37:12
wallbox_disable_battery_at_mix_mode | true | 2024-02-18 16:37:11
wallbox_energy_L1 | 10272.41 | 2024-02-18 16:37:12
wallbox_energy_L2 | 7.69 | 2024-02-18 16:37:12
wallbox_energy_L3 | 7.69 | 2024-02-18 16:37:12
wallbox_energy_solar | 39859 | 2024-02-18 14:06:16
wallbox_energy_total | 138808 | 2024-02-18 14:06:16
wallbox_index | 0 | 2024-02-18 16:37:08
wallbox_key_state | 0 | 2024-02-18 16:37:11
wallbox_locked | false | 2024-02-18 16:37:12
wallbox_max_current | 20 | 2024-02-18 16:37:12
wallbox_number_phases | 1 | 2024-02-18 16:37:11
wallbox_number_used_phases | 1 | 2024-02-18 16:37:12
wallbox_plugged | true | 2024-02-18 16:37:12
wallbox_power_L1 | 0 | 2024-02-18 16:37:12
wallbox_power_L2 | 0 | 2024-02-18 16:37:12
wallbox_power_L3 | 0 | 2024-02-18 16:37:12
wallbox_power_solar | 0 | 2024-02-18 14:06:16
wallbox_power_total | 0 | 2024-02-18 14:06:16
wallbox_soc | 0 | 2024-02-18 16:37:11
wallbox_sun_mode | false | 2024-02-18 16:37:13
```

Energy in Wh has now much more resolution. And it makes sense, my parallel evcc installation has recorded 10283Wh for this loading session. Also also in E3DC portal there is 10,29kWh, which I get when I add energy of all all phases L1+L2+L3: 10272.4+7.7+7.7.
[image]

I am still not sure about the energy values for total and solar. If I compare the values between yesterday and today for total I get 133667 / 138808, the difference is 5141. If I compare solar energy I get 34905 / 39859, the difference is 4954. That is not the ~10000Wh, that have been loaded.

pvtom, 2024-02-18:
Cool, that the energy/Lx values fits now.
I did some other changes to clearify the power/Lx and the total and solar energy topics.
rscp2mqtt will print log to stdout starting with "TAG_WB_DATA"?
Could you please post the output to this issue?

docolli, 2024-02-18:
I am getting only the follwing output, when I start rscp2mqtt:

```
rscp2mqtt [v3.16.dev]
E3DC system >192.168.1.121:5033< user: >***<
MQTT broker >127.0.0.1:1883< qos = >0< retain = >false< client id >rscp2mqtt< prefix >e3dc<
Fetching data every 10 seconds.
Requesting PVI ✓ | PM (0) | DCB ✓ (1 battery string) | Wallbox (0) ✓ | Autorefresh ✗
Log level = 0
Stdout to terminal

[2024-02-18 19:35:13] pid=25558 ppid=25549 RscpMqttMain.cpp(2476) Connecting to server 192.168.1.121:5033
[2024-02-18 19:35:13] pid=25558 ppid=25549 RscpMqttMain.cpp(2483) Success: E3DC connected.
[2024-02-18 19:35:13] pid=25558 ppid=25549 RscpMqttMain.cpp(1501) RSCP authentication level 10
[2024-02-18 19:35:13] pid=25558 ppid=25549 RscpMqttMain.cpp(2031) Connecting to broker 127.0.0.1:1883
[2024-02-18 19:35:13] pid=25558 ppid=25549 RscpMqttMain.cpp(2040) Success: MQTT broker connected.
[2024-02-18 19:35:13] pid=25558 ppid=25549 RscpMqttMain.cpp(313) MQTT: starting listener loop
```

Must I change something in config? Or use a parameter when I start rscp2mqtt manually?

docolli, 2024-02-18:
Ah, I found the Log Level in config and changed it to 1. However I am getting this, but no messages starting with "TAG_WB_DATA":
```
rscp2mqtt [v3.16.dev]
E3DC system >192.168.1.121:5033< user: >***<
MQTT broker >127.0.0.1:1883< qos = >0< retain = >false< client id >rscp2mqtt< prefix >e3dc<
Fetching data every 10 seconds.
Requesting PVI ✓ | PM (0) | DCB ✓ (1 battery string) | Wallbox (0) ✓ | Autorefresh ✗
Log level = 1
Stdout to terminal

[2024-02-18 19:39:43] pid=25565 ppid=25549 RscpMqttMain.cpp(2476) Connecting to server 192.168.1.121:5033
[2024-02-18 19:39:43] pid=25565 ppid=25549 RscpMqttMain.cpp(2483) Success: E3DC connected.
[2024-02-18 19:39:43] pid=25565 ppid=25549 RscpMqttMain.cpp(1501) RSCP authentication level 10
[2024-02-18 19:39:43] pid=25565 ppid=25549 RscpMqttMain.cpp(2031) Connecting to broker 127.0.0.1:1883
[2024-02-18 19:39:43] pid=25565 ppid=25549 RscpMqttMain.cpp(2040) Success: MQTT broker connected.
[2024-02-18 19:39:43] pid=25565 ppid=25549 RscpMqttMain.cpp(313) MQTT: starting listener loop
[2024-02-18 19:39:44] pid=25565 ppid=25549 RscpMqttMain.cpp(1530) Error: Container 0x05840000 Tag 0x05860000 received error code 6.
[2024-02-18 19:39:44] pid=25565 ppid=25549 RscpMqttMain.cpp(1619) Error: Container 0x02840000 Tag 0x02000060 received error code 6.
[2024-02-18 19:39:44] pid=25565 ppid=25549 RscpMqttMain.cpp(1619) Error: Container 0x02840000 Tag 0x02000070 received error code 6.
[2024-02-18 19:39:44] pid=25565 ppid=25549 RscpMqttMain.cpp(1619) Error: Container 0x02840000 Tag 0x02000080 received error code 6.
[2024-02-18 19:39:46] pid=25565 ppid=25549 RscpMqttMain.cpp(1530) Error: Container 0x05840000 Tag 0x05860000 received error code 6.
[2024-02-18 19:39:58] pid=25565 ppid=25549 RscpMqttMain.cpp(1530) Error: Container 0x05840000 Tag 0x05860000 received error code 6.
[2024-02-18 19:40:09] pid=25565 ppid=25549 RscpMqttMain.cpp(1530) Error: Container 0x05840000 Tag 0x05860000 received error code 6.
[2024-02-18 19:40:20] pid=25565 ppid=25549 RscpMqttMain.cpp(1530) Error: Container 0x05840000 Tag 0x05860000 received error code 6.
[2024-02-18 19:40:32] pid=25565 ppid=25549 RscpMqttMain.cpp(1530) Error: Container 0x05840000 Tag 0x05860000 received error code 6.
[2024-02-18 19:40:43] pid=25565 ppid=25549 RscpMqttMain.cpp(1530) Error: Container 0x05840000 Tag 0x05860000 received error code 6.
[2024-02-18 19:40:54] pid=25565 ppid=25549 RscpMqttMain.cpp(1530) Error: Container 0x05840000 Tag 0x05860000 received error code 6.
[2024-02-18 19:41:05] pid=25565 ppid=25549 RscpMqttMain.cpp(1530) Error: Container 0x05840000 Tag 0x05860000 received error code 6.
[2024-02-18 19:41:17] pid=25565 ppid=25549 RscpMqttMain.cpp(1530) Error: Container 0x05840000 Tag 0x05860000 received error code 6.
[2024-02-18 19:41:28] pid=25565 ppid=25549 RscpMqttMain.cpp(1530) Error: Container 0x05840000 Tag 0x05860000 received error code 6.
[2024-02-18 19:41:39] pid=25565 ppid=25549 RscpMqttMain.cpp(1530) Error: Container 0x05840000 Tag 0x05860000 received error code 6.

```

pvtom, 2024-02-18:
Have you reloaded and recompiled the changes from Branch dev again?

docolli, 2024-02-18:
No, sorry. I misunderstood your post and thought that the "other changes" were related to your first publish of dev.
Here it is:
```
######
TAG_WB_DATA: topic >e3dc/wallbox/status< tag >0x0E860000< datatype = 14 format = 0 divisor = 1 bit_to_bool = 0 unit ><
TAG_WB_DATA: topic >e3dc/wallbox/status< payload_float >0.00<
TAG_WB_DATA: topic >e3dc/wallbox/status< payload_double >0.00<
TAG_WB_DATA: topic >e3dc/wallbox/status< payload_int32 >243662849<
TAG_WB_DATA: topic >e3dc/wallbox/status< payload_uint32 >243662849<
######
TAG_WB_DATA: topic >e3dc/wallbox/soc< tag >0x0E800003< datatype = 5 format = 0 divisor = 1 bit_to_bool = 0 unit >%<
TAG_WB_DATA: topic >e3dc/wallbox/soc< payload_float >0.00<
TAG_WB_DATA: topic >e3dc/wallbox/soc< payload_double >0.00<
TAG_WB_DATA: topic >e3dc/wallbox/soc< payload_int32 >0<
TAG_WB_DATA: topic >e3dc/wallbox/soc< payload_uint32 >0<
######
TAG_WB_DATA: topic >e3dc/wallbox/active_phases< tag >0x0E80000F< datatype = 3 format = 0 divisor = 1 bit_to_bool = 0 unit ><
TAG_WB_DATA: topic >e3dc/wallbox/active_phases< payload_float >0.00<
TAG_WB_DATA: topic >e3dc/wallbox/active_phases< payload_double >0.00<
TAG_WB_DATA: topic >e3dc/wallbox/active_phases< payload_int32 >7<
TAG_WB_DATA: topic >e3dc/wallbox/active_phases< payload_uint32 >7<
######
TAG_WB_DATA: topic >e3dc/wallbox/number_phases< tag >0x0E84103B< datatype = 3 format = 0 divisor = 1 bit_to_bool = 0 unit ><
TAG_WB_DATA: topic >e3dc/wallbox/number_phases< payload_float >0.00<
TAG_WB_DATA: topic >e3dc/wallbox/number_phases< payload_double >0.00<
TAG_WB_DATA: topic >e3dc/wallbox/number_phases< payload_int32 >1<
TAG_WB_DATA: topic >e3dc/wallbox/number_phases< payload_uint32 >1<
######
TAG_WB_DATA: topic >e3dc/wallbox/key_state< tag >0x0E841050< datatype = 3 format = 0 divisor = 1 bit_to_bool = 0 unit ><
TAG_WB_DATA: topic >e3dc/wallbox/key_state< payload_float >0.00<
TAG_WB_DATA: topic >e3dc/wallbox/key_state< payload_double >0.00<
TAG_WB_DATA: topic >e3dc/wallbox/key_state< payload_int32 >0<
TAG_WB_DATA: topic >e3dc/wallbox/key_state< payload_uint32 >0<
######
TAG_WB_DATA: topic >e3dc/wallbox/power/L1< tag >0x0E80000C< datatype = 11 format = 0 divisor = 1 bit_to_bool = 0 unit >W<
TAG_WB_DATA: topic >e3dc/wallbox/power/L1< payload_float >0.00<
TAG_WB_DATA: topic >e3dc/wallbox/power/L1< payload_double >0.00<
TAG_WB_DATA: topic >e3dc/wallbox/power/L1< payload_int32 >0<
TAG_WB_DATA: topic >e3dc/wallbox/power/L1< payload_uint32 >0<
######
TAG_WB_DATA: topic >e3dc/wallbox/power/L2< tag >0x0E80000D< datatype = 11 format = 0 divisor = 1 bit_to_bool = 0 unit >W<
TAG_WB_DATA: topic >e3dc/wallbox/power/L2< payload_float >0.00<
TAG_WB_DATA: topic >e3dc/wallbox/power/L2< payload_double >0.00<
TAG_WB_DATA: topic >e3dc/wallbox/power/L2< payload_int32 >0<
TAG_WB_DATA: topic >e3dc/wallbox/power/L2< payload_uint32 >0<
######
TAG_WB_DATA: topic >e3dc/wallbox/power/L3< tag >0x0E80000E< datatype = 11 format = 0 divisor = 1 bit_to_bool = 0 unit >W<
TAG_WB_DATA: topic >e3dc/wallbox/power/L3< payload_float >0.00<
TAG_WB_DATA: topic >e3dc/wallbox/power/L3< payload_double >0.00<
TAG_WB_DATA: topic >e3dc/wallbox/power/L3< payload_int32 >0<
TAG_WB_DATA: topic >e3dc/wallbox/power/L3< payload_uint32 >0<
######
TAG_WB_DATA: topic >e3dc/wallbox/energy/total< tag >0x0E800001< datatype = 7 format = 3 divisor = 1000 bit_to_bool = 0 unit >kWh<
TAG_WB_DATA: topic >e3dc/wallbox/energy/total< payload_float >0.00<
TAG_WB_DATA: topic >e3dc/wallbox/energy/total< payload_double >0.00<
TAG_WB_DATA: topic >e3dc/wallbox/energy/total< payload_int32 >139653<
TAG_WB_DATA: topic >e3dc/wallbox/energy/total< payload_uint32 >139653<
######
TAG_WB_DATA: topic >e3dc/wallbox/energy/solar< tag >0x0E800002< datatype = 7 format = 3 divisor = 1000 bit_to_bool = 0 unit >kWh<
TAG_WB_DATA: topic >e3dc/wallbox/energy/solar< payload_float >0.00<
TAG_WB_DATA: topic >e3dc/wallbox/energy/solar< payload_double >0.00<
TAG_WB_DATA: topic >e3dc/wallbox/energy/solar< payload_int32 >40429<
TAG_WB_DATA: topic >e3dc/wallbox/energy/solar< payload_uint32 >40429<
######
TAG_WB_DATA: topic >e3dc/wallbox/energy/L1< tag >0x0E800012< datatype = 11 format = 3 divisor = 1 bit_to_bool = 0 unit >Wh<
TAG_WB_DATA: topic >e3dc/wallbox/energy/L1< payload_float >-581196830177376145228573966336.00<
TAG_WB_DATA: topic >e3dc/wallbox/energy/L1< payload_double >11116.60<
TAG_WB_DATA: topic >e3dc/wallbox/energy/L1< payload_int32 >-253051304<
TAG_WB_DATA: topic >e3dc/wallbox/energy/L1< payload_uint32 >4041915992<
######
[... 221 lines cut ...]
######
TAG_WB_DATA: topic >e3dc/wallbox/energy/solar< tag >0x0E800002< datatype = 7 format = 3 divisor = 1000 bit_to_bool = 0 unit >kWh<
TAG_WB_DATA: topic >e3dc/wallbox/energy/solar< payload_float >0.00<
TAG_WB_DATA: topic >e3dc/wallbox/energy/solar< payload_double >0.00<
TAG_WB_DATA: topic >e3dc/wallbox/energy/solar< payload_int32 >40429<
TAG_WB_DATA: topic >e3dc/wallbox/energy/solar< payload_uint32 >40429<
######
TAG_WB_DATA: topic >e3dc/wallbox/energy/L1< tag >0x0E800012< datatype = 11 format = 3 divisor = 1 bit_to_bool = 0 unit >Wh<
TAG_WB_DATA: topic >e3dc/wallbox/energy/L1< payload_float >-581196830177376145228573966336.00<
TAG_WB_DATA: topic >e3dc/wallbox/energy/L1< payload_double >11116.60<
TAG_WB_DATA: topic >e3dc/wallbox/energy/L1< payload_int32 >-253051304<
TAG_WB_DATA: topic >e3dc/wallbox/energy/L1< payload_uint32 >4041915992<
######
TAG_WB_DATA: topic >e3dc/wallbox/energy/L2< tag >0x0E800013< datatype = 11 format = 3 divisor = 1 bit_to_bool = 0 unit >Wh<
[... cut ...]
TAG_WB_DATA: topic >e3dc/wallbox/energy/L2< payload_uint32 >1749552749<
######
TAG_WB_DATA: topic >e3dc/wallbox/energy/L3< tag >0x0E800014< datatype = 11 format = 3 divisor = 1 bit_to_bool = 0 unit >Wh<
TAG_WB_DATA: topic >e3dc/wallbox/energy/L3< payload_float >3778957621074844442427392.00<
TAG_WB_DATA: topic >e3dc/wallbox/energy/L3< payload_double >7.70<
TAG_WB_DATA: topic >e3dc/wallbox/energy/L3< payload_int32 >1749552749<
TAG_WB_DATA: topic >e3dc/wallbox/energy/L3< payload_uint32 >1749552749<
```

docolli, 2024-02-18:
I started loading with 1,4kW:
```
######
TAG_WB_DATA: topic >e3dc/wallbox/power/L1< tag >0x0E80000C< datatype = 11 format = 0 divisor = 1 bit_to_bool = 0 unit >W<
TAG_WB_DATA: topic >e3dc/wallbox/power/L1< payload_float >0.00<
TAG_WB_DATA: topic >e3dc/wallbox/power/L1< payload_double >1368.00<
TAG_WB_DATA: topic >e3dc/wallbox/power/L1< payload_int32 >0<
TAG_WB_DATA: topic >e3dc/wallbox/power/L1< payload_uint32 >0<
######
TAG_WB_DATA: topic >e3dc/wallbox/power/L2< tag >0x0E80000D< datatype = 11 format = 0 divisor = 1 bit_to_bool = 0 unit >W<
TAG_WB_DATA: topic >e3dc/wallbox/power/L2< payload_float >0.00<
TAG_WB_DATA: topic >e3dc/wallbox/power/L2< payload_double >0.00<
TAG_WB_DATA: topic >e3dc/wallbox/power/L2< payload_int32 >0<
TAG_WB_DATA: topic >e3dc/wallbox/power/L2< payload_uint32 >0<
######
TAG_WB_DATA: topic >e3dc/wallbox/power/L3< tag >0x0E80000E< datatype = 11 format = 0 divisor = 1 bit_to_bool = 0 unit >W<
TAG_WB_DATA: topic >e3dc/wallbox/power/L3< payload_float >0.00<
TAG_WB_DATA: topic >e3dc/wallbox/power/L3< payload_double >0.00<
TAG_WB_DATA: topic >e3dc/wallbox/power/L3< payload_int32 >0<
TAG_WB_DATA: topic >e3dc/wallbox/power/L3< payload_uint32 >0<
######
```

pvtom, 2024-02-18:
Ich sehe folgendes im Output:
1. wallbox/status ist ein Container, d.h. da gibt es eine untergeordnete Struktur mit weiteren Informationen. Mal schauen, ob ich da etwas finde.
2. wallbox/soc ist 0. Ich dachte, das könnte der Ladestand der Autobatterie sein. Ist er scheinbar nicht. Oder er wird für die vorliegende Wallbox nicht ausgegeben.
3. wallbox/power/Lx funktioniert. Beim Laden wird ein Wert von ca 1400 Watt an L1 für einphasiges Laden ausgegeben
4. wallbox/energy/solar und wallbox/energy/total sind Integer-Werte. Der Divisor von 1000 zieht hier nicht. D.h. es sind voraussichtlich Werte in Wh (nicht kWh)... die Frage bleibt, warum man diese Werte im Portal so nicht findet.

Ich habe den Branch dev nochmal aktualisiert:
- Log Output wieder entfernt
- zu 4. Einheit auf Wh geändert

docolli, 2024-02-18:
1. Wenn ich da helfen kann, sag Bescheid. Dann kompiliere ich morgen Abend eine neue dev Version und wir suchen weiter.
2. Das dachte ich auch. Kann aber sein, dass mein E-Auto (Nissan Leaf ZE-1) nicht mit der Wallbox kommuniziert. Kann aber gut sein, dass man hier mit anderen Autos den Soc sieht.
3. Ja, mit double werden die Werte korrekt interpretiert.
4. Ich denke auch, dass Wh als uint32 korrekt ist. Ab welchen Zeitpunkt diese Werte gelten und warum die bei der aktuellen Ladung vom Differenzwert her nicht zur geladenen Energie passen, ist noch fraglich.

docolli, 2024-02-18:
Jetzt passt es wieder. Zwei Nachkommastellen sind aber hier eigentlich nicht nötig.
```
wallbox_power_L1 | 1372.00 | 2024-02-18 21:14:40
wallbox_power_L2 | 0.00 | 2024-02-18 21:08:49
wallbox_power_L3 | 0.00 | 2024-02-18 21:08:49
wallbox_power_solar | 0 | 2024-02-18 21:14:04
wallbox_power_total | 1372 | 2024-02-18 21:14:40
```

pvtom, 2024-02-18:
Die Nachkommastellen habe ich entfernt.
Neuer Stand ist unter dev eingecheckt.

docolli, 2024-02-19:
Hi Maik,

sorry that we temporary have switched to german as issue language.

Thomas has changed the decimals for the `e3dc/wallbox/power/Lx` values in his last dev release. The energy values still have the 2 decimals. 

I have an Easy Connect Wallbox from E3DC. And a car that can load only from 1 phase. And in my system, the energy values for each phase L1/L2/L3 make sense, when they are seen as "Wh"
```
wallbox_energy_L1 | 11674.36 | 2024-02-19 17:51:36
wallbox_energy_L2 | 7.70 | 2024-02-19 17:51:36
wallbox_energy_L3 | 7.70 | 2024-02-19 17:51:36
wallbox_energy_solar | 40436 | 2024-02-19 17:51:36
wallbox_energy_total | 140210 | 2024-02-19 17:51:36
```

In the "my.e3dc.com" portal at E-Mobility I can see my last loading with the following values:
[image]
That matches the sum of the energy values for each phase.

However, how the two values for total and solar should be interpreted, is unclear to me, even when we think about shifting the comma. For sure they are not directly related to the last loading, since they increase with each loading.
We can take a note of both of them before and after a loading to see how they both have changed compared to the actual values for this loading. Maybe then we can find a relationship and a meaning / conversion formula for these values.

kasi0815, 2024-02-19:
Hi docolli,

German is fine for me as well but may be not for all.  ;-)

Here are some values taken during charging my car today:
`````
e3dc/wallbox/energy/total 101167
e3dc/wallbox/energy/solar 4022
e3dc/wallbox/energy/L1 33952.38
e3dc/wallbox/energy/L2 33613.80
e3dc/wallbox/energy/L3 33600.86
`````
I agree that the energy counters (first two rows) have the unit Wh. The increase of first row today was exactly the charging amount. But I still need to test for the second one (solar) as we had no sun today.

However, the total amount of rows 1 and 2 cannot be assigned to any total value shown in my E3DC app. So I must have triggered a reset in between (weeks ago) to set the values to zero. In next days I will run more tests to see whether switching off the wallbox (circuit breaker) resets those values to zero. 

Now to the Lx values where I see my mistake. The Lx energy values are counters (for every single phase) as well and during charging the Lx power values show the actual charging power. So the decimal places are ok.   :-)

docolli, 2024-02-20:
Good job! Now my previous values are also making sense with the info, that total power is grid power.

>I am still not sure about the energy values for total and solar. If I compare the values between yesterday and today for total I get 133667 / 138808, the difference is 5141. If I compare solar energy I get 34905 / 39859, the difference is 4954. That is not the ~10000Wh, that have been loaded.

5141 kWh (grid) +4954 kWh (solar) = 10148 kWh😃

I can confirm that we should rename `e3dc/wallbox/energy/total` to `e3dc/wallbox/energy/grid`.

However, as far as I can see it from the values, `e3dc/wallbox/power/total` is really the total power of L1+L2+L3 and NOT the grid power.

kasi0815, 2024-02-20:
Hmmm. Now some time later the E3DC app shows a different value. So the update inside my app for the "Wallbox Gesamtladeleistung" was not correct when I wrote my last post. Sadly, my car doesn't tell me the total charging power, so I only have the app as a reference.

Please wait with renaming. I thing it needs some more tests!

docolli, 2024-02-21:
My values from today:

[image]
[image]
Both my E3DC portal and my evcc software have recorded the same 7,12kWh total energy for this charging.

Now the values from rscp2mqtt:
```
wallbox_energy_L1 | 7108.55
wallbox_energy_L2 | 7.12
wallbox_energy_L3 | 7.12
```
So these values are related to the LAST CHARGING.

Now the change of the values for wallbox_energy_total and wallbox_energy_solar:
```
wallbox_energy_total 147959 -> 155082 => Difference 7123
wallbox_energy_solar 48080 -> 55165 => Difference 7085
```
I my case the total value is really the total energy for this charging session. The solar value is really the value for solar energy. If I calculate to solar proportion 7085/7123 = 99,5%, which is the same as evcc has calculated it.

So, from my point of view and the data of today, we should NOT rename `wallbox_energy_total`.

kasi0815, 2024-02-23:
Still no sun. The weekend should get better...

kasi0815, 2024-02-24:
Sonne!   :-)

Here is the result of today's test. Before charging:
```
e3dc/wallbox/energy/total 102567
e3dc/wallbox/energy/solar 4588
```
After charging:
```
e3dc/wallbox/energy/total 107977
e3dc/wallbox/energy/solar 8851
```
This results in:
```
Total: 5410Wh
Solar: 4263Wh
```

Those results above are reflected in the E3DC app (new and old):
```
Netz: 1.15kWh
Solar: 4.26kWh
Gesamt: 5.41kWh
```
So there is NO renaming necessary. Sorry for confusion and thank you for implementation!

pvtom, 2024-02-24:
Hallo,
unsere Zusammenarbeit hat ja prima funktioniert :-)

Zwei Fragen hätte ich noch:
1. Die Power-Werte werden in Watt und ohne Nachkommastellen ausgeben, die Energie-Werte in Wh mit zwei Nachkommastellen. Soll ich die Nachkommastellen bei den Energie-Werten bestehen lassen oder wegnehmen?
2.  Ich könnte die Differenz für e3dc/wallbox/energy/total vor und nach dem Ladevorgang (auch für e3dc/wallbox/energy/solar) berechnen und zusätzlich ausgeben, z.B. als e3dc/wallbox/energy/last_charging/total und e3dc/wallbox/energy/last_charging/solar, damit man die Werte für eine Ladung nicht erst ausrechnen muss. Dazu müsste ich die Startwerte speichern und von den Werten, die während der Ladung entstehen, abziehen. Nachts oder wenn man feststellt, dass das Laden beendet wurde, würde man sich die letzten Werte wieder als Startwerte merken. Macht das Sinn? Und wie könnte man das Ladeende feststellen? Wenn der Wert wallbox/plugged von true auf false springt? wallbox/plugged hört sich an, wie Auto angestöpselt oder nicht. Passt das? Alternativ könnte ich die Werte von wallbox/energy/L1 bis wallbox/energy/L3 addieren, das ergibt, so habe ich verstanden, den Total-Wert für die letzte Ladung... das wäre einfacher, man hätte dann allerdings den Solar-Wert nicht. Was meint Ihr?

Gruß Thomas

pvtom, 2024-02-25:
Ich habe die Änderungen zu 1. "Nachkommastellen" und 2. "last charging" umgesetzt und im Branch "dev" veröffentlicht.
Viel Spaß beim Testen ;-)

kasi0815, 2024-02-25:
Hallo, ich war noch einen Test wegen dem Reset der Energy-Counter-Werte `e3dc/wallbox/energy/total ` und `e3dc/wallbox/energy/solar` schuldig. Die angezeigten Werte sind bei mir wesentlich kleiner als das, was mit dieser Wallbox geladen wurde. Hier ist der Grund: Ein 'Ausschalten' der Wallbox ändert NICHTS an den Werten. Aber ein Reset des E3DC-Hauskraftwerkes setzt diese Werte auf Null!
```
e3dc/wallbox/index 0
e3dc/wallbox/power/total 0
e3dc/wallbox/power/solar 0
e3dc/wallbox/battery_to_car 1
e3dc/wallbox/battery_before_car 0
e3dc/wallbox/battery_discharge_until 50
e3dc/wallbox/disable_battery_at_mix_mode true
e3dc/wallbox/active_phases/L1 true
e3dc/wallbox/active_phases/L2 true
e3dc/wallbox/active_phases/L3 true
e3dc/wallbox/number_phases 3
e3dc/wallbox/soc 0
e3dc/wallbox/key_state 0
e3dc/wallbox/energy/total 0
e3dc/wallbox/energy/solar 0
e3dc/wallbox/energy/L1 0.00
e3dc/wallbox/energy/L2 0.00
e3dc/wallbox/energy/L3 0.00
e3dc/wallbox/power/L1 0
e3dc/wallbox/power/L2 0
e3dc/wallbox/power/L3 0
e3dc/wallbox/number_used_phases 3
e3dc/wallbox/max_current 16
e3dc/wallbox/plugged false
e3dc/wallbox/locked false
e3dc/wallbox/charging false
e3dc/wallbox/canceled false
e3dc/wallbox/sun_mode false
```

kasi0815, 2024-02-25:
Das neue "last charging" werde ich in der kommenden Woche testen. Zwei meiner Anwendungsfälle sind:
A) Ich aktualisiere um Mitternacht eine Historie, in der der Hausverbrauch, Verbrauch der Heizung, Summe der Energie der Solarerzeugung, ... , Summe der Energie der Wallbox abgespeichert sind. Dafür nutze ich jetzt `e3dc/wallbox/energy/total` als "Wallbox-Stromzähler" und muss Ausnahmen, wie z. B. das Anlagen-Reset behandeln (siehe oben).
B) Ich möchte das Laden des Autos protokollieren. Ein Ladezyklus kann bei mir auch mehrere Tage dauern, da ich oft nur mit Überschuss lade. Also, wenn das Auto in der Garage steht, es als Speicher für den Solarstrom verwende, wenn das Hauskraftwerk bereits zu 100% geladen ist und ich den Solar-Überschuß einspeißen müsste. Ich denke, dass `e3dc/wallbox/plugged` ein guter Indikator für den Ladezyklus ist und werde es nächste Woche, so die Sonne will, testen.   :-)

docolli, 2024-02-27:
Habe eben erst aktualisiert. Werde berichten wie es läuft.

>Hier ist der Grund: Ein 'Ausschalten' der Wallbox ändert NICHTS an den Werten. Aber ein Reset des E3DC-Hauskraftwerkes setzt diese Werte auf Null!

Vermutlich würde auch ein Löschen der Wallbox im HKW und wieder Einbinden reichen. Ich möchte aber kurz auf dein "Reset des E3DC-HKW" kommen: Du hast dein HKW bereits mal "resettet"? Ich habe im Herbst mal einen Fehler gemacht und beim Einbinden der Wallbox den falschen Typ ausgewählt. Jetzt ist diese IP Adresse im HKW "verbrannt", es sagt diese würde bereits benutzt werden. Um meine Wallbox wieder anmelden zu können, musste ich auch eine andere IP ausweichen. Du hast also Erfahrung mit einem Reset des HKW? Ich habe mich das nie getraut, weil ich Sorge habe, dass ich dann Werte neu setzen muss, die ich mir vorher nicht gemerkt habe bzw. die nur mein Solarteur setzen kann und ich nicht die notwendigen Rechte habe. Wäre prima, wenn Du mir schildern könntest, wie man den Reset macht und hinterher alles wieder einrichtet.

kasi0815, 2024-03-01:
I checked the new feature "last charging" the last days keeping the car plugged over night and continued charging on the next day, etc. Everything is fine for me! Good job.   :-)

docolli, 2024-03-01:
I also checked it and found no problems. Thank you and let's release it. 👍

pvtom, 2024-03-01:
Das Release ist jetzt raus. Hat Spaß gemacht! Schönes Wochenende!

closed 2024-03-01 by pvtom

docolli, 2024-05-08:
@jomifa Ich spekuliere mal: Könnte es sein, dass ein Strompfad über einen Wechselrichter bereits "gesättigt" ist, also gar nicht mehr Strom darüber fließen kann und deine Wallbox aber mehr Strom benötigt und darum dann der Rest aus den Netz kommt? Dann wäre dies der Flaschenhals und könnte nicht über eine Parametrierung beeinflusst werden? Wie gesagt, dies ist nur Spekulation und diese Vermutung kann komplett falsch sein!

Kannst du mal bitte die absoluten Werte für den Stromfluss hier posten? Unter "das Fahrzeug mit ca. 50-70% geladen" kann ich mir z.B. wenig vorstellen, worauf beziehen sich die Prozent? Auf die maximale Wallboxleistung von 11kW?

Also Werte für PV Erzeugung, Akku Beladung/Entladung, Netzbezug/einspeisung, Hausverbrauch und Wallboxleistung. 

Welches S10 hast du (E / SE / X oder gar eines der Pro Serie)? Die haben mit 8 bzw. 12kW durchaus unterschiedliche Ausgangsleistungen.

jomifa, 2024-05-08:
Ich habe dir drei Fotos gemacht, die das von mir beschrieben näher beleuchten. Ich habe ein HKW E3/DC S10 E Pro AI13, einen integrierten Akku (1 String) mit 13kWh, einem Notstromversorgung 3-phasig, eine PV-Anlage mit 15,75 kWp, 2 MPP Tracker, einer E3/DC Wallbox Multi Connect mit 22kWh 3-phasig angeschlossen. Das Auto unterstützt leider nur 11kWh. 

Ein Foto für die Nutzung des Netzstromes beim Laden des Autos reiche ich gerne nach. Das passiert nicht so oft.

[image]
[image]
[image]

docolli, 2024-05-09:
Danke, ich sehe schon, was Du meinst. Die PV erzeugt 4000W und die Wallbox lädt nur mit 1800W (1-phasig) und es gehen 1600W ins Netz. Die 1600W könnten ja auch ins Auto geladen werden.

Man muss dazu wissen, dass es auch stark von den Möglichkeiten des Autos abhängt, was die Wallbox laden kann. Das musste ich auch erst lernen. So dachte ich zunächst auch, dass mein Leaf ZE1 mit max. 6kW prima zu meiner 11kW Wallbox passt, beim ersten Laden hat es mir aber immer die 3x16A Absicherung der Wallbox im Sicherungsschrank raus. Bis ich verstanden habe, dass der Leaf ZE1 nur 1-phasig lädt, die 11kW Wallbox aber 3-phasig ausgelegt ist, also 1-phasig nur 3,7kW (=16A) bereitstellen kann.

Auch ist es unterschiedlich, welche Mindestladung ein Auto 1-phasig (meist 6A) benötigt, um überhaupt Laden zu können. Auch 3-phasig gibt es da andere Mindeststromgrenzen. Jetzt muss das System entscheiden, was es tun soll, wenn (z.B. bei einer Wolke) die PV Leistung für die Wallbox unter 6A fällt. Soll dann mit Netzstrom auf 6A ergänzt werden, soll die Hausbatterie (bis zu welchem SOC) aushelfen, oder soll das Laden dann (nach einiger Zeit) unterbrochen werden.

Du hast die Ladestrategie deaktiviert. Weshalb? Ich habe bei mir sofort (mit keinen weiteren Einträgen) gewählt.

Wie sieht bei Dir die Ladepriorisierung aus? Hier meine Einstellungen:
[image]

Und dann sehe ich, dass du die Phasenumschaltung auf "Auto" hast. Ich habe eine Easy Connect, da kann ich nur 1/2/3 wählen und bei mir steht die auf "1" (mehr kann mein Leaf nicht). Hast du da schon mal probiert auf 1 oder 3 zu stellen?

Welches Auto hast du angesteckt? Immer daran denken, dass das Auto letztendlich steuert, wieviel Strom es aus der Wallbox bezieht. Die Wallbox teilt dem Auto nur mit, wieviel Leistung es aktuell maximal beziehen könnte.
https://ladefragen.de/27/

Bist du schon mal auf die Webseite deiner Wallbox gegangen und hast Dir dort die Werte angeschaut (nix umstellen!) ? Gib mal die IP deiner Wallbox in einen Browser ein, kommt da so eine Seite mit "Compleo" und der Frage nach einem Passwort? Google "compleo wallbox passwort"😉
[image]
Ich lade gerade mit 11 bzw. 12A (Status C).
[image]

jomifa, 2024-05-09:
Hallo,
der screenshot für die 1-phasige Ladung war nur ein Beispiel. Ist mehr Sonnenenergie vorhanden lädt die Wallbox natürlich auch 3-phasig.

Was die Ladestrategie und Priorisierung angeht habe ich mich für folgende Regelung entschieden:

  1.  Immer zuerst die Batterie laden
  2.  Wenn voll, dann Wallbox im Sonnenmodus
  3.  Nur wenn der Akku des PKWs zu niedrig ist und gebraucht wird, dann umschalten der Wallbox auf Netzstrom (Schlüsselschalter auf Sperrmodus) demnächst über das Dashboard.

Sollte ich das Ziel auch anders/besser erreichen können, dann bin ich für Vorschläge offen. 😉

Du siehst wir haben hier ganz unterschiedliche Nutzungsweisen gesetzt, da das Auto nicht sehr oft genutzt wird (home office).
Das Auto, das geladen wird, ist ein Hyundai Ionic 5 mit einem 11 kW AC Charger. Wenn ich diesen Wagen an einer stinknormalen AC Wallbox auflade, bekommt er von einer AC Wallbox ca. 11kW geladen. Bei einer DC Wallbox entsprechend mehr. Irgendwann regelt mein Wagen den Strom entsprechend seiner Charger Kapazität/Strategie ab. Warum nicht bei der E3DC Multi Connect? Ich denke, dass das Verhalten von E3DC hier kundenschädlich ist. Billigen Strom ins Netz laden und eventuell später den teuren Strom aus dem Netz ziehen zu müssen, wenn keine Sonnenenergie mehr da ist. Wie blöd ist das denn?

Was die IP-Adresse meiner Wallbox angeht, kann ich sie nicht über die Eingabe dieser in einem Browser erreichen. Vielleicht gilt das nur für die Easy Connect.

Gruß
Joachim

grafik.png (view on web)<https://github.com/pvtom/rscp2mqtt/assets/78646938/c418433b-5096-4326-91d1-3288cf0ac2fb>

https://ladefragen.de/27/

grafik.png (view on web)<https://github.com/pvtom/rscp2mqtt/assets/78646938/155c5a8b-9f1e-4b32-8f2f-66e972efdb2b>
grafik.png (view on web)<https://github.com/pvtom/rscp2mqtt/assets/78646938/34c303c5-47f7-43fb-adfd-a6157c494ff1>

docolli, 2024-05-09:
@jomifa Bitte editiere deinen Post, da hängt jetzt die komplette Mail von mir noch mit dran!

Okay, die Umschaltung 1p/3p funktioniert also. Hast du die Wallbox schon mal aus deinem Hauskraftwerk gelöscht und wieder hinzugefügt (unbedingt MultiConnect auswählen!)? Ich weiß, dass dann das HKW einige Initialisierungsbefehle per Modbus schickt. Vielleicht hilft das die Wallbox wieder auf Kurs zu bekommen?

Ist die WB per LAN ans Hausnetz angeschlossen,w o auch das E3DC HKW drin ist? Kannst du die WB und HKW anpingen? Nicht dass du irgendwelche komischen Netzwerkprobleme hast, welche die Kommunikation stören und somit die Wallbox nicht sauber auf die Vorgaben vom HKW reagiert.