pvtom/rscp2mqtt #50: Powermeter values float precision

cwihne, 2024-02-17:
Reading and interpreting powermeter values, I came across the wish to have kWh with float precision like 7.54 kWh. So far, TAG_PM_ENERGY_Lx deliver int kWh. I found my experience isn't good enough to change them to float, so I changed in `RscpMqttMapping.h`:
from `{ TAG_PM_DATA, TAG_PM_ENERGY_L1, 0, "%s/energy/L1", "", F_FLOAT_D, UNIT_KWH, 1000, 0, false, false, false },`
to `{ TAG_PM_DATA, TAG_PM_ENERGY_L1, 0, "%s/energy/L1", "", F_FLOAT_2, UNIT_WH, 1, 0, false, false, false },`
...
(changing `F_FLOAT_D` to `F_FLOAT_2` didn't help, it's still integer)
Depending on your preference, is there a chance to change this to float from your end, or help by pointing to the correct changes for me to apply?

pvtom, 2024-02-18:
Hello,
I have changed the code.
If you like you can check out the branch "dev" (not "main" as usual) with the changed files RscpMqttMain.cpp and RscpMqttMapping.h
Best regards 
Thomas

pvtom, 2024-02-18:
will be fixed with release v3.16

cwihne, 2024-02-18:
Checked the dev branch. The change works just great. Thank you.

closed 2024-02-18 by cwihne