pvtom/rscp2mqtt #51: battery_discharge_until

kasi0815, 2024-03-06:
Hallo,

ich habe leider keine weiteren Informationen zum Parameter `battery_discharge_until` für die Wallbox gefunden. Meine Tests ergaben, dass beim Erreichen des Batterielevels das Laden des Autos unterbrochen wird. Wenn man mit Überschuss lädt, dann wird sich die Batterie wieder aufladen.

Was passiert beim Erreichen der eingestellten Schwelle? Wird das Laden beendet und muss dann durch eine gesonderte Aktion wieder getriggert werden?

Vielen Dank!

pvtom, 2024-03-06:
Guten Abend,
das kann ich leider nicht beantworten, da ich keine Wallbox von E3/DC habe und es somit nicht ausprobieren kann.
rscp2mqtt liest den Wert aus und kann ihn schreiben. Wie sich die Wallbox verhält, weiß ich leider nicht.
Evtl. kann @docolli weiterhelfen?
Gruß Thomas

docolli, 2024-03-06:
Das ist der Wert den man auch im Hauskraftwerk unter "Batterieentladung durch Wallbox - bis SOC xx%" einstellen kann.

Bis zu diesem Wert liefert das Hauskraftwerk volle Leistung um die Hausleistung und die Wallboxleistung zu bedienen. Darunter liefert das Hauskraftwerk aus der Batterie nur noch was das Haus braucht, die Wallboxleistung kommt aus dem Stromnetz.

docolli, 2024-03-07:
Ergänzung:
[image]

Mit diesen Einstellung verhält sich bei mir das HKW so, dass es die Wallboxladung bis zur angegebenen SOC mit voller Leistung (4,5kW) unterstützt.

Ich denke es finden sich auf diesem Bild auch die Werte für folgende MQTT Topics (in Klammern meine Werte):

- battery_before_car (0)
- battery_to_car (1)
- disable_battery_at_mix_mode (true)
- sun_mode (false)

Ich kann nachher mal die Einstellungen ändern und schauen, welche Topics dahinter stecken. @pvtom: Kann sein, dass wir für manche Topics besser true/false nahmen sollten.

docolli, 2024-03-07:
Folgende Erkenntnisse durch Eingabe am HKW Display und Beobachtung der MQTT Topics:

1. Ladepriorität
Dies ist das Topic `battery_before_car`.
0 = erst Wallbox
1 = erst Batterie

Wird dieses auf 1 gesetzt, dann werden automatisch die Topics `battery_to_car = 0` und `disable_battery_at_mix_mode = 0` gesetzt.
Nur wenn dieses Topic 0 ist, dann kann man die folgenden Topics benutzen.

2. Im Sonnenmodus
Dies ist das Topic `battery_to_car`.
0 = unterbunden
1= erlaubt
Wenn dieses auf 1 ist, dann kann man `battery_discharge_until` verwenden

3. Bis SOC
Dies ist das Topic `battery_discharge_until`.
xx = Prozentwert

4. Im Mischmodus und während Haltezeit und Zeitfunktion
Dies ist das Topic `disable_battery_at_mix_mode`.
true = unterbunden
false = erlaubt.

pvtom, 2024-03-09:
Hallo,
ich habe die Änderungsvorschläge (Umbenennungen der Topics und beim Payload true/false statt 1/0) im Branch "dev" zum Ausprobieren umgesetzt. Ich hoffe, es funktioniert, sonst muss ich im Code noch etwas anpassen.
Eine Frage: Der Topic e3dc/wallbox/max_current ist in WALLBOX.md auch mit true/false beschrieben. Ist das tatsächlich passend? Oder wird da die Stromstärke in Ampere zurückgegeben? Je nachdem würde ich statt 1/0 dann auch hier true/false setzen oder eben die Einheit auf UNIT_A ändern.
Gruß Thomas

docolli, 2024-03-09:
Vielen Dank für die Umsetzung, teste ich mal die nächsten Tage, wenn es bei mir passt.

Den `max_current` bekomme ich mit dem Wert "20" zurück und das korreliert bei mir nur mit dem Wert für die Schieflastbegrenzung auf der internen Webseite der Wallbox.
[image]

https://www.e-mobileo.de/schieflast-begrenzung-deutschland-oesterreich-schweiz/
Hier noch ein Hinweis aus einem anderen Forum:
[image]

[image]

Müsste man fast in `max_current_asymmetrical` umbenennen.

docolli, 2024-03-10:
Teste gerade den aktuelle dev-Zwieg.

1. Danke fürs korrektere Verwenden von "charge" statt "load" im Topic-Namen! Macht sprachlich echt mehr Sinn.
2. `charge_battery_before_car`: hier bekomme ich immer noch "0/1" anstatt "false/true" zurück. Ist das Absicht, oder hast du das übersehen?
3. Das mit den Werten für die Phasen ist aus meiner Sicht noch nicht intuitiv zu verstehen. Mein Auto lädt nur 1phasig, die WB ist aber 3phasig angeschlossen. Für `wallbox_active_phases` bekommen ich "7", also alle 3 Phasen aktiv zurück. 
Zusätzlich gibt es `wallbox_number_phases`und `wallbox_number_used_phases`, welche bei mir beide "1" zurück geben. 
Sind die "active Phases" die an der Wallbox angeschlossenen und aktuell stromführenden Phasen? Dann wäre statt "active" eher sowas wie "connected" sprachlich besser verständlich.
"Active" könnte dann vermutlich besser zu `wallbox_number_phases`und `wallbox_number_used_phases` dazu, das sind wohl eher die Zahl der tatsächlich aktuell für die Ladung genutzten Phasen. Die WB kann ja zwischen 3p und 1p Ladung wechseln (wenn das Auto mitmacht), um noch Überschuss ins Auto laden zu können, wenn der zuwenig ist, um 3p mit dem Mindeststrom (meist 6A) laden zu können. Vielleicht hat @kasi0815 eine 3p Auto und kann die Werte mal protokollieren, wenn der Überschuss geringer wird und die WB von 3p auf 1p umschaltet.

PS: hier sehe ich noch einige weitere RSCP Tags für die Wallbox https://github.com/Eba-M/E3DC-Control/blob/master/RscpTags.h Können wir gerne mal diskutieren, welche noch fehlen und welche Sinn machen, hier ins Projekt einzubauen. Gerne als extra Issue!

pvtom, 2024-03-10:
zu 2) habe ich gerade eine Anpassung der Datei RscpMqttMapping.h im dev-Branch vorgenommen. Da hatte ich wohl etwas übersehen.
zu 3) sollten wir abwarten, was ein "3 Phasen-Nutzer" zu wallbox_number_phases und wallbox_number_used_phases herausfindet
zum P.S.) ja, ich habe auch schon gesehen, dass es weitere Tags gibt. Eine gute Quelle für Tags ist auch die Datei rscpLibV0.9.3.min.js, die das E3/DC-Portal nutzt. Wir können gerne versuchen, neue Tags einzubinden.
Gruß Thomas

kasi0815, 2024-03-11:
Sorry, wir waren am Wochenende nicht da.  Ich teste in den kommenden Tagen mit, auch 3phasig.   :-)

kasi0815, 2024-03-11:
zu 3) Wie kann ich helfen?

Hier der output zu `mosquitto_pub -h localhost -p 1883 -t "e3dc/set/force" -m 1`:

```
e3dc/wallbox/index 0
e3dc/wallbox/energy/last_charging/total 0
e3dc/wallbox/energy/last_charging/solar 0
e3dc/wallbox/power/total 0
e3dc/wallbox/power/solar 0
e3dc/wallbox/discharge_battery_to_car true
e3dc/wallbox/charge_battery_before_car false
e3dc/wallbox/discharge_battery_until 5
e3dc/wallbox/disable_battery_at_mix_mode true
e3dc/wallbox/active_phases/L1 true
e3dc/wallbox/active_phases/L2 true
e3dc/wallbox/active_phases/L3 true
e3dc/wallbox/number_phases 3
e3dc/wallbox/soc 0
e3dc/wallbox/key_state false
e3dc/wallbox/energy/total 107083
e3dc/wallbox/energy/solar 68720
e3dc/wallbox/energy/L1 38865
e3dc/wallbox/energy/L2 33215
e3dc/wallbox/energy/L3 35003
e3dc/wallbox/power/L1 0
e3dc/wallbox/power/L2 0
e3dc/wallbox/power/L3 0
e3dc/wallbox/number_used_phases 3
e3dc/wallbox/max_current 16
e3dc/wallbox/status 200
e3dc/wallbox/plugged true
e3dc/wallbox/locked false
e3dc/wallbox/charging false
e3dc/wallbox/canceled true
e3dc/wallbox/sun_mode true
```

Wenn ich auf 1phasig umschalte, dann ändern sich: `e3dc/wallbox/number_used_phases 1` und `e3dc/wallbox/number_phases 1`

kasi0815, 2024-03-11:
zu 3) Die Sonne ist gnädig und das Wetter wechselhaft. Hier ein Mitschnitt, in dem zuerst 1phasig geladen wird, dann das System auf 3phasig wechselt und wieder zurück auf 1 phasig wechselt (innerhalb sehr kurzer Zeit):

```
e3dc/wallbox/energy/last_charging/total 127
e3dc/wallbox/energy/last_charging/solar 123
e3dc/wallbox/power/total 3565
e3dc/wallbox/power/solar 3462
e3dc/wallbox/energy/total 107210
e3dc/wallbox/energy/solar 68843
e3dc/wallbox/energy/L1 38992
e3dc/wallbox/power/L1 3565
e3dc/wallbox/energy/last_charging/total 138
e3dc/wallbox/energy/last_charging/solar 133
e3dc/wallbox/power/total 3634
e3dc/wallbox/power/solar 3633
e3dc/wallbox/energy/total 107221
e3dc/wallbox/energy/solar 68853
e3dc/wallbox/energy/L1 39003
e3dc/wallbox/power/L1 3634
e3dc/wallbox/energy/last_charging/total 148
e3dc/wallbox/energy/last_charging/solar 144
e3dc/wallbox/power/total 3611
e3dc/wallbox/power/solar 3588
e3dc/wallbox/energy/total 107231
e3dc/wallbox/energy/solar 68864
e3dc/wallbox/energy/L1 39014
e3dc/wallbox/power/L1 3611
e3dc/wallbox/energy/last_charging/total 157
e3dc/wallbox/energy/last_charging/solar 153
e3dc/wallbox/power/total 0
e3dc/wallbox/power/solar 0
e3dc/wallbox/number_phases 3
e3dc/wallbox/energy/total 107240
e3dc/wallbox/energy/solar 68873
e3dc/wallbox/energy/L1 39022
e3dc/wallbox/power/L1 0
e3dc/wallbox/number_used_phases 3
e3dc/wallbox/status 136
e3dc/wallbox/charging false
e3dc/wallbox/energy/last_charging/total 164
e3dc/wallbox/energy/last_charging/solar 157
e3dc/wallbox/power/total 6026
e3dc/wallbox/power/solar 4125
e3dc/wallbox/energy/total 107247
e3dc/wallbox/energy/solar 68877
e3dc/wallbox/energy/L1 39025
e3dc/wallbox/energy/L2 33217
e3dc/wallbox/energy/L3 35005
e3dc/wallbox/power/L1 2001
e3dc/wallbox/power/L2 2024
e3dc/wallbox/power/L3 2001
e3dc/wallbox/status 168
e3dc/wallbox/charging true
e3dc/wallbox/energy/last_charging/total 178
e3dc/wallbox/energy/last_charging/solar 168
e3dc/wallbox/power/total 4140
e3dc/wallbox/power/solar 2895
e3dc/wallbox/energy/total 107261
e3dc/wallbox/energy/solar 68888
e3dc/wallbox/energy/L1 39029
e3dc/wallbox/energy/L2 33222
e3dc/wallbox/energy/L3 35010
e3dc/wallbox/power/L1 1380
e3dc/wallbox/power/L2 1380
e3dc/wallbox/power/L3 1380
e3dc/wallbox/energy/last_charging/total 190
e3dc/wallbox/energy/last_charging/solar 176
e3dc/wallbox/power/solar 2349
e3dc/wallbox/energy/total 107273
e3dc/wallbox/energy/solar 68896
e3dc/wallbox/energy/L1 39033
e3dc/wallbox/energy/L2 33226
e3dc/wallbox/energy/L3 35014
e3dc/wallbox/power/L3 1403
e3dc/wallbox/energy/last_charging/total 203
e3dc/wallbox/energy/last_charging/solar 182
e3dc/wallbox/power/solar 2232
e3dc/wallbox/energy/total 107286
e3dc/wallbox/energy/solar 68902
e3dc/wallbox/energy/L1 39037
e3dc/wallbox/energy/L2 33230
e3dc/wallbox/energy/L3 35018
e3dc/wallbox/power/L1 1357
e3dc/wallbox/power/L2 1403
e3dc/wallbox/power/L3 1380
e3dc/wallbox/energy/last_charging/total 215
e3dc/wallbox/energy/last_charging/solar 189
e3dc/wallbox/power/solar 2133
e3dc/wallbox/energy/total 107298
e3dc/wallbox/energy/solar 68909
e3dc/wallbox/energy/L1 39041
e3dc/wallbox/energy/L2 33234
e3dc/wallbox/energy/L3 35022
e3dc/wallbox/power/L2 1380
e3dc/wallbox/power/L3 1403
e3dc/wallbox/energy/last_charging/total 227
e3dc/wallbox/energy/last_charging/solar 195
e3dc/wallbox/power/total 4117
e3dc/wallbox/power/solar 2097
e3dc/wallbox/energy/total 107310
e3dc/wallbox/energy/solar 68915
e3dc/wallbox/energy/L1 39045
e3dc/wallbox/energy/L2 33238
e3dc/wallbox/energy/L3 35027
e3dc/wallbox/power/L3 1380
e3dc/wallbox/energy/last_charging/total 239
e3dc/wallbox/energy/last_charging/solar 202
e3dc/wallbox/power/total 0
e3dc/wallbox/power/solar 0
e3dc/wallbox/number_phases 1
e3dc/wallbox/energy/total 107322
e3dc/wallbox/energy/solar 68922
e3dc/wallbox/energy/L1 39049
e3dc/wallbox/energy/L2 33242
e3dc/wallbox/energy/L3 35031
e3dc/wallbox/power/L1 0
e3dc/wallbox/power/L2 0
e3dc/wallbox/power/L3 0
e3dc/wallbox/number_used_phases 1
e3dc/wallbox/status 136
e3dc/wallbox/charging false
e3dc/wallbox/energy/last_charging/total 240
e3dc/wallbox/energy/last_charging/solar 204
e3dc/wallbox/power/total 1518
e3dc/wallbox/power/solar 1518
e3dc/wallbox/energy/total 107323
e3dc/wallbox/energy/solar 68924
e3dc/wallbox/energy/L1 39051
e3dc/wallbox/power/L1 1518
e3dc/wallbox/status 168
e3dc/wallbox/charging true
e3dc/wallbox/energy/last_charging/total 245
e3dc/wallbox/energy/last_charging/solar 209
e3dc/wallbox/power/total 1748
e3dc/wallbox/power/solar 1749
e3dc/wallbox/energy/total 107328
e3dc/wallbox/energy/solar 68929
e3dc/wallbox/energy/L1 39056
e3dc/wallbox/power/L1 1748
```

docolli, 2024-03-11:
Also `number_phases`und `number_used_phases` laufen bei Dir synchron?
Wenn ja, dann macht das keinen Sinn, die Info als zwei Topics herauszugeben.

kasi0815, 2024-03-11:
`number_phases` und `number_used_phases` laufen bei mir synchron. `number_used_phases` würde aus meiner Sicht reichen.

`number_active_phases` finde ich als Zahl bei mir nicht. Ist diese notwendig?

Umbenennung in `e3dc/wallbox/phases/L1 true` ist ok.

docolli, 2024-03-11:
>number_active_phases finde ich als Zahl bei mir nicht. Ist diese notwendig?

Ah, moment. Ich glaube die ist inzwischen raus als Topic und wird in L1/L2/L3 aufgelöst. 
Muss ich noch in meiner Hausautomation löschen.

pvtom, 2024-03-11:
Ich habe number_used_phases rausgenommen und
e3dc/wallbox/active_phases/L1 true
e3dc/wallbox/active_phases/L2 true
e3dc/wallbox/active_phases/L3 true
in
e3dc/wallbox/phases/L1 true
e3dc/wallbox/phases/L2 true
e3dc/wallbox/phases/L3 true
umbenannt.
number_phases habe ich so gelassen, da es eine korrespondierende set - Funktion mit gleichem Namen gibt.
Siehe dev-branch!

kasi0815, 2024-03-15:
Eine Bemerkung: Ich sehe `wallbox/energy/last_charging/total` immer 2x hintereinander. Ist das so gewollt?
```
e3dc/wallbox/power/L2 1748
e3dc/wallbox/power/L3 1748
e3dc/wallbox/energy/last_charging/total 42
e3dc/wallbox/energy/last_charging/solar 41
e3dc/wallbox/power/total 4669
e3dc/wallbox/power/solar 4620
e3dc/wallbox/energy/total 155050
e3dc/wallbox/energy/solar 90978
e3dc/wallbox/energy/L1 57313
e3dc/wallbox/energy/L2 47963
e3dc/wallbox/energy/L3 49774
e3dc/wallbox/power/L1 1541
e3dc/wallbox/power/L2 1564
e3dc/wallbox/power/L3 1564
e3dc/wallbox/energy/last_charging/total 56
e3dc/wallbox/energy/last_charging/solar 55
e3dc/wallbox/power/total 5244
e3dc/wallbox/power/solar 4716
e3dc/wallbox/energy/total 155064
e3dc/wallbox/energy/solar 90992
e3dc/wallbox/energy/L1 57318
e3dc/wallbox/energy/L2 47968
e3dc/wallbox/energy/L3 49779
e3dc/wallbox/power/L1 1748
e3dc/wallbox/power/L2 1748
e3dc/wallbox/power/L3` 1748
```

pvtom, 2024-03-15:
Ich sehe ein Deinem Listing e3dc/wallbox/energy/last_charging/**total** und e3dc/wallbox/energy/last_charging/**solar** hintereinander.
Bei der nächsten Ausgabe der beiden Werte sieht man, dass jeweils 14 Wh dazugekommen sind. Das passt zu den Werten zu e3dc/wallbox/energy/total und e3dc/wallbox/energy/solar.
Aus meiner Sicht ist das richtig, oder?

kasi0815, 2024-03-15:
> Ich sehe in Deinem Listing e3dc/wallbox/energy/last_charging/**total** und e3dc/wallbox/energy/last_charging/**solar** hintereinander. Bei der nächsten Ausgabe der beiden Werte sieht man, dass jeweils 14 Wh dazugekommen sind. Das passt zu den Werten zu e3dc/wallbox/energy/total und e3dc/wallbox/energy/solar. Aus meiner Sicht ist das richtig, oder?

Oh je, ich bin Blind. Stimmt natürlich. Ein schönes Wochenende!

closed 2024-07-11 by pvtom