pvtom/rscp2mqtt #52: "e3dc/set/interval" not considered

cwihne, 2024-03-11:
It seems that "e3dc/set/interval" is not considered, while the INTERVAL value of config file is taken into account on startup.

When starting rscp2mqtt, `INTERVAL` setting in .config file sets the initial cadence of writing data to MQTT broker. When `INTERVAL` is changed and rscp2mqtt got a restart, new frequency is applied.
README.md reports that the cadence can be changed by e.g. `mosquitto_pub -h localhost -p 1883 -t "e3dc/set/interval" -m 2`, to change the interval. MQTT Explorer reflects the change in correct topic, but rscp2mqtt's update frequency doesn't change.
Commenting INTERVAL in config file results in default 1s cadence, but also here, newly published interval doesn't change this.
Btw.-1 "set/\*" topics are not created by rscp2mqtt on startup, which might or might not be expected, e.g. to avoid feedback loops.:check:
Btw.-2 `mosquitto_pub -h localhost -p 1883 -t "e3dc/set/force" -m 1` does result in a full topics refresh (except e3dc/set/*) :+1:

pvtom, 2024-03-11:
yes, it's a bug. I can reproduce it. Will be fixed by the next release.

pvtom, 2024-03-16:
fixed by v3.17 today

closed 2024-03-16 by pvtom