pvtom/rscp2mqtt #53: Feature Request: power meter summaries

cwihne, 2024-03-11:
Recent version of rscp2mqtt offers great support for multiple power meters, with subsequent values for power, voltage, energy and active phases. Those provide individual values for phases L1, L2, L3.
Yet, at least power and energy readings make sense to be provided as a sum of L1+L2+L3, e.g. as value for title topic:
`e3dc/pm/1/power` = 1092 (W) or `e3dc/pm/2/energy` = 535.836 (kWh). 
This releases subscribed applications from such highly repetetive calculations.
Maybe similar summaries also for topics in other paths, aside of power meters?

Btw. topics `e3dc/pm_0/energy` and `e3dc/pm_1/energy` are constantly 0.00. Same for those under historical topics.

pvtom, 2024-03-17:
Good idea. I have realised the sum topics for the power meter values, currently I'm testing. Delivery with the next release v3.18.
The pm_0 and pm_1 values make no sense, they are always 0.0 for me too, so I will remove the topics.

pvtom, 2024-03-23:
Published with v3.18.1

closed 2024-03-23 by pvtom