pvtom/rscp2mqtt #57: ENERGY per String

MyGitIT, 2024-04-03:
Hi,

short question, is it possible to select the e3dc/pvi/energy/string_x (today) instead off e3dc/pvi/energy_all/string_x (all time)?

This would be great!

Thx

pvtom, 2024-04-06:
Hi,
the S10 device only provides the total energy values via the interface. Daily values can be calculated by memorizing the values at the change of day. I like the idea and will implement it soon.
Best regards
Thomas

pvtom, 2024-04-14:
implemented with release v3.19.
New topic: e3dc/pvi/energy/string_x for the daily values.

MyGitIT, 2024-04-14:
Hi Thomas,

vielen Dank, wird das Topic auch in INFLUX geschrieben?

pvtom, 2024-04-15:
Hi Sven,
Deine Frage hat mich stutzig gemacht.
Das Topic soll natürlich auch in die InfluxDB geschrieben werden.
Nur leider passiert das nicht. Grrrh. Ein Problem mit der Reihenfolge: Einrichten des Topics, Setzen des InfluxDB-Flags... es sind auch noch andere Topics, die dynamisch angelegt werden, betroffen.
Das wird mit dem nächsten Release dann gefixt. Also Danke für Deine Nachfrage :-)
Gruß Thomas

pvtom, 2024-04-15:
Reopen, da Topic nicht in die InfluxDB geschrieben wird.

pvtom, 2024-04-20:
Die Lösung steht jetzt mit v3.20 zur Verfügung.

MyGitIT, 2024-04-20:
Hi Thomas,

die Topics sind in der Datenbank, danke dafür. Aber kann es sein, dass die Werte nur auf Stundenbasis und nicht auf Tag aggregiert werden?

Gruß Sven

pvtom, 2024-04-20:
Hallo Sven,
die Werte werden für den aktuellen Tag aggregiert. Dazu wird der Wert um Mitternacht als Startwert gespeichert und dann zur Berechnung des Tageswerts benutzt (aktueller Wert minus Startwert).
Das klappt natürlich nach Neustart des Programms erst richtig für den nächsten Tag. Evtl. kommen Dir die Werte gerade daher nicht richtig vor.
Mit RETAIN_FOR_SETUP=true in der .config wird der Startwert im MQTT als retained Topic gespeichert und nach dem Neustart eingelesen. So überlebt der Startwert einen Neustart von rscp2mqtt. Mit v3.20 hat sich allerdings das Format ein wenig geändert ggü. v3.19, von daher zieht ein vorhandener Startwert bei Wechsel auf v3.20 erstmal nicht. Auch das regelt die nächste Mitternacht.
Ich hoffe, das passt so?
Gruß Thomas

MyGitIT, 2024-04-21:
Hallo Thomas,

der neue Stringwert passt:

InfluxDB:
[image]

E3DC-Portal:
[image]

Perfekt, vielen, vielen Dank!

Gruß Sven

closed 2024-04-21 by MyGitIT