pvtom/rscp2mqtt #58: INFLUXDB_TOPIC=e3dc/system/software wird nicht in InfluxDB geschrieben

MyGitIT, 2024-04-05:
Hallo,

das Topic e3dc/system/software wird nicht in die InfluxDB geschrieben, ebenso die Topics

e3dc/system/inverter_count,
e3dc/system/serial_number,
e3dc/system/production_date,
e3dc/system/software

Ist das so gewollt?

Gruß Sven

pvtom, 2024-04-06:
Hallo Sven,
die Seriennummer und das Produktionsdatum ändern sich nicht. Von daher macht es aus meiner Sicht keinen Sinn, die Werte in einer Zeitreihen-Datenbank zu speichern.
Die Software-Version ändert sich zwar ab und zu, ist aber eine Zeichenkette. Die InfluxDB hat Strings bei meinen Versuchen immer mit einer Fehlermeldung abgelehnt.
Der Inverter_Count (Anzahl der Wechselrichter?) ändert sich ja auch nur selten.
Von daher... ja, es ist so gewollt :-)
Gruß
Thomas

MyGitIT, 2024-04-06:
Hallo Thomas,

alles klar, dann hole ich mir das Topic per Telegraf.

Danke, Gruß Sven

pvtom, 2024-04-07:
Hi Sven,
ich werde es so anpassen, dass auch Strings geschrieben werden.
Ein Bug ist, dass Werte, die keine Einheit haben, derzeit gar nicht geschrieben werden, wie z.B. der inverter_count.
Gruß Thomas

MyGitIT, 2024-04-07:
Hallo Thomas,

sehr gut, dann habe ich alles aus einer Hand. Bisher habe ich Solaranzeige, RSCPGui, E3DC-Control,... gebraucht, jetzt kann ich das mit rscp2mqtt, Mosquito, Telegraf und Co. machen. Nur eine Datenquelle nur einmal Daten abfragen, perfekt.

Super Sache, danke dir.

pvtom, 2024-04-14:
fixed by v3.19

closed 2024-04-14 by pvtom

pvtom, 2024-04-14:
Strings werden in einem separaten Measurement (default: e3dc_meta) gespeichert

MyGitIT, 2024-04-14:
Hallo Thomas!

Perfekt, danke für deine Arbeit!