pvtom/rscp2mqtt #61: Timouts...

timmost, 2024-04-14:
Hi Tom,

probiere mal wieder... bin jetzt erstmal ohne Docker unterwegs. Funktioniert soweit, ich bekomme nur ab und an Timeouts.. kannst Du sagen, was den Timeout erzeugt (e3dc, influx etc?)

pi@sturmserver:/opt/rscp2mqtt $ ./rscp2mqtt
rscp2mqtt [v3.19]
E3DC system >192.168.178.58:5033< user: >Timmo.Sturm<
MQTT broker >localhost:1883< qos = >0< retain = >false< client id >✗< prefix >e3dc<
Fetching data every 10 seconds.
Requesting PVI ✓ | PM (0) | DCB ✓ (1 battery string) | Wallbox (0) ✓ | Autorefresh ✗
Log level = 1
Stdout to terminal

[2024-04-14 13:12:23] pid=1067620 ppid=982363 RscpMqttMain.cpp(2717) Connecting to server 192.168.178.58:5033
[2024-04-14 13:12:23] pid=1067620 ppid=982363 RscpMqttMain.cpp(2724) Success: E3DC connected.
[2024-04-14 13:12:23] pid=1067620 ppid=982363 RscpMqttMain.cpp(1688) RSCP authentication level 10
[2024-04-14 13:12:23] pid=1067620 ppid=982363 RscpMqttMain.cpp(2221) Connecting to broker localhost:1883
[2024-04-14 13:12:23] pid=1067620 ppid=982363 RscpMqttMain.cpp(2233) Success: MQTT broker connected.
[2024-04-14 13:12:23] pid=1067620 ppid=982363 RscpMqttMain.cpp(345) MQTT: starting listener loop
[2024-04-14 13:12:24] pid=1067620 ppid=982363 RscpMqttMain.cpp(1806) Error: Container 0x02840000 Tag 0x02000060 received error code 6.
[2024-04-14 13:12:24] pid=1067620 ppid=982363 RscpMqttMain.cpp(1806) Error: Container 0x02840000 Tag 0x02000070 received error code 6.
[2024-04-14 13:12:24] pid=1067620 ppid=982363 RscpMqttMain.cpp(1806) Error: Container 0x02840000 Tag 0x02000080 received error code 6.
[2024-04-14 13:12:28] pid=1067620 ppid=982363 RscpMqttMain.cpp(2100) Error: Response receive timeout (retry)
[2024-04-14 13:12:32] pid=1067620 ppid=982363 RscpMqttMain.cpp(2100) Error: Response receive timeout (retry)

meine config:

#-E3DC Access
#--IP address and port of your home power station
E3DC_IP=192.168.178.58
E3DC_PORT=5033

#--your E3/DC web account and the RSCP/AES password configured in the device (Issue #31)
E3DC_USER={private}
E3DC_PASSWORD={private}
E3DC_AES_PASSWORD={private}

#-MQTT Broker
#--Name or IP address and port of your MQTT broker
MQTT_HOST=localhost
MQTT_PORT=1883
#--MQTT parameters
MQTT_CLIENT_ID=
MQTT_QOS=0
MQTT_RETAIN=false
#--MQTT authentication if necessary set MQTT_AUTH=true and enter user and password
MQTT_AUTH=false
MQTT_USER=
MQTT_PASSWORD=

#-System
#--cycle interval in seconds (max 300 seconds)
INTERVAL=10
#--log level, set "1" to get detailled logging
LOG_LEVEL=1
#--log files
LOGFILE=/tmp/rscp2mqtt.log
TOPIC_LOG_FILE=/tmp/rscp2mqtt.history
#--dry run
DISABLE_MQTT_PUBLISH=false
#--set power mode
AUTO_REFRESH=false
#--Start of Operation
HISTORY_START_YEAR=2017
#--Request daily historical values
DAILY_VALUES=true
#--Publish statistic values
STATISTIC_VALUES=true

#-MQTT Topics
#--Topic prefix (max 24 characters), default is e3dc
PREFIX=e3dc

#-Power Meter
#--PM_INDEX can be set multiple times if several power meters exist, default is 0
#--PM_EXTERN=true has the same effect as PM_INDEX=6 (6 for S10 Mini, compatibilty with previous releases)
PM_EXTERN=false
PM_INDEX=0
PM_REQUESTS=true

#-Photovoltaic Inverter
PVI_REQUESTS=true
#--Number of PVI strings/trackers (disables the auto detection)
PVI_TRACKER=2

#-Battery Settings
#--Number of Battery Strings, default is 1
BATTERY_STRINGS=1
#--Battery Modules
DCB_REQUESTS=true

#-Battery SOC limiter
#SOC_LIMITER=true
#LIMIT_CHARGE_SOC=0
#LIMIT_DISCHARGE_SOC=0
#LIMIT_CHARGE_DURABLE=false
#LIMIT_DISCHARGE_DURABLE=false
#LIMIT_DISCHARGE_BY_HOME_POWER=0

#-Wallbox Support for the wallbox with the specified index
#--only one WB_INDEX entry is supported, default is 0
WALLBOX=true
WB_INDEX=0

#-Forced Topics
#--will be published with every cycle
FORCE_PUB=e3dc/[a-z]+/power
#FORCE_PUB=e3dc/pvi/.*/string_[1,2]
FORCE_PUB=e3dc/solar/energy
FORCE_PUB=e3dc/battery/soc
FORCE_PUB=e3dc/battery/energy/discharge
FORCE_PUB=e3dc/battery/energy/charge
FORCE_PUB=e3dc/home/energy
FORCE_PUB=e3dc/grid/energy/in
FORCE_PUB=e3dc/grid/energy/out

#-Topic Logging
#--logging to extra log file configured by TOPIC_LOG_FILE
#TOPIC_LOG=e3dc/history/.*
#TOPIC_LOG=e3dc/battery/dcb/[1-4]/soh
#TOPIC_LOG=e3dc/system/software

timmost, 2024-04-14:
Hi Tom
habe gerade gesehen, dass Du Timeouts in 3.19 vor ein paar Std adressiert hattest.. allerdings habe ich genau mit der noch issues... (s.o.) VG timmo

pvtom, 2024-04-14:
Hallo Timmo,
die Timeouts kommen beim Zugriff auf das Hauskraftwerk.
Mit v3.19 habe ich eine Änderung gebracht, die bei "Max. buffer size exceeded" dazu führt, dass das Programm danach weiterläuft... allerdings ohne es testen zu können.
Sonnige Grüße
Thomas

timmost, 2024-04-14:
Hi tom,

Weiterlaufen tut es… ich beobachte es mal.

Viele Grüße - Timmo

Timmo Sturm
Wittelsbacherstr. 8
85622 Feldkirchen
+49 151 402 71 711

pvtom, 2024-04-15:
Hallo Timmo,
ich habe noch ein wenig geforscht. Das Lesen vom Netzwerk-Socket beim Empfang der Daten vom Hauskraftwerk ist nicht korrekt umgesetzt. Der Code ist noch eins zu eins aus dem Beispielprogramm, das E3/DC zur Verfügung stellt.
Ich habe mir andere Beispiele diesbgzl. angeschaut und auch mal ChatGPT nach einem Code-Schnipsel gefragt und das Programm dann angepasst.
Läuft bei mir schon. Da ich den Retry-Fehler nicht bekomme bzw. in Vergangenheit nur ab und zu hatte, kann ich nicht sagen, ob die Anpassung hilft. Wenn Du Lust hast, kann ich Dir den Code (eine Datei) im Branch namens DEV zum Ausprobieren zur Verfügung stellen. Ich will nicht gleich alle anderen Nutzer damit behelligen...
Gruß Thomas

timmost, 2024-04-15:
Hi Tom, können wir gerne machen, allerdings dann erstmal nativ ohne docker…

Viele Grüße - Timmo

Timmo Sturm
Wittelsbacherstr. 8
85622 Feldkirchen
+49 151 402 71 711

pvtom, 2024-04-15:
Genau, ganz ohne Docker... mit Kompilieren und so :-)
Kannst Dir die geänderte Datei z.B. mit curl holen:

curl https://raw.githubusercontent.com/pvtom/rscp2mqtt/dev/RscpMqttMain.cpp

timmost, 2024-04-15:
Hi Tom,
compiliert sauber durch. und die Timeouts sind weg ;-)
Hier mal einfach der Dump der ersten 30 Sekunden...  ich habe nichts Schreckliches gesehen, aber vielleicht siehst Du da noch was. Nächster Schritt wäre das mal ein paar Tage laufen zu lassen, und dann wieder in Docker zu gehen (das war schon gut mit den Docker Containers, ich habe meinen Raspi vier mal neu aufgesetzt, bis ich die USB Sticks in die Tonne getreten habe...  mit Docker ist das wirklich fix, und die SSD ist super schnell). Scheinbar werden auch hochwertige USBs als Boot Partition hoffnungslos überfordert (werden richtig heiß und verabschieden sich)
Vielen Dank - Timmo

pi@sturmserver:/opt/rscp2mqtt $ ./rscp2mqtt -v
rscp2mqtt [v3.20.dev]
E3DC system >192.168.178.58:5033< user: >Timmo.Sturm<
MQTT broker >localhost:1883< qos = >0< retain = >false< client id >✗< prefix >e3dc<
History year range from 2017 to 2024
Fetching data every 10 seconds.
Requesting PVI ✓ | PM (0) | DCB ✓ (1 battery string) | Wallbox (0) ✓ | Autorefresh ✗
Log level = 1
Stdout to terminal

[2024-04-15 21:53:27] pid=1144717 ppid=1135969 RscpMqttMain.cpp(540) Topic >e3dc/battery/state< unused? Payload is empty
[2024-04-15 21:53:27] pid=1144717 ppid=1135969 RscpMqttMain.cpp(540) Topic >e3dc/grid/state< unused? Payload is empty
[2024-04-15 21:53:27] pid=1144717 ppid=1135969 RscpMqttMain.cpp(540) Topic >e3dc/solar/power_max< unused? Payload is empty
[2024-04-15 21:53:27] pid=1144717 ppid=1135969 RscpMqttMain.cpp(540) Topic >e3dc/home/power_min< unused? Payload is empty
[2024-04-15 21:53:27] pid=1144717 ppid=1135969 RscpMqttMain.cpp(540) Topic >e3dc/home/power_max< unused? Payload is empty
[2024-04-15 21:53:27] pid=1144717 ppid=1135969 RscpMqttMain.cpp(540) Topic >e3dc/grid/power_min< unused? Payload is empty
[2024-04-15 21:53:27] pid=1144717 ppid=1135969 RscpMqttMain.cpp(540) Topic >e3dc/grid/power_max< unused? Payload is empty
[2024-04-15 21:53:27] pid=1144717 ppid=1135969 RscpMqttMain.cpp(540) Topic >e3dc/battery/soc_min< unused? Payload is empty
[2024-04-15 21:53:27] pid=1144717 ppid=1135969 RscpMqttMain.cpp(540) Topic >e3dc/battery/soc_max< unused? Payload is empty
[2024-04-15 21:53:27] pid=1144717 ppid=1135969 RscpMqttMain.cpp(540) Topic >e3dc/grid_in_limit_duration< unused? Payload is empty
[2024-04-15 21:53:27] pid=1144717 ppid=1135969 RscpMqttMain.cpp(540) Topic >e3dc/sunshine_duration< unused? Payload is empty
[2024-04-15 21:53:27] pid=1144717 ppid=1135969 RscpMqttMain.cpp(540) Topic >e3dc/wallbox/index< unused? Payload is empty
[2024-04-15 21:53:27] pid=1144717 ppid=1135969 RscpMqttMain.cpp(540) Topic >e3dc/wallbox/energy/last_charging/total< unused? Payload is empty
[2024-04-15 21:53:27] pid=1144717 ppid=1135969 RscpMqttMain.cpp(540) Topic >e3dc/wallbox/energy/last_charging/solar< unused? Payload is empty
[2024-04-15 21:53:27] pid=1144717 ppid=1135969 RscpMqttMain.cpp(537) Topic >e3dc/pm/power< forced
[2024-04-15 21:53:27] pid=1144717 ppid=1135969 RscpMqttMain.cpp(540) Topic >e3dc/pm/power< unused? Payload is empty
[2024-04-15 21:53:27] pid=1144717 ppid=1135969 RscpMqttMain.cpp(540) Topic >e3dc/pm/energy< unused? Payload is empty
[2024-04-15 21:53:27] pid=1144717 ppid=1135969 RscpMqttMain.cpp(540) Topic >e3dc/pvi/energy/string_1< unused? Payload is empty
[2024-04-15 21:53:27] pid=1144717 ppid=1135969 RscpMqttMain.cpp(540) Topic >e3dc/pvi/energy/string_2< unused? Payload is empty
[2024-04-15 21:53:27] pid=1144717 ppid=1135969 RscpMqttMain.cpp(540) Topic >e3dc/pvi/energy_start/string_1< unused? Payload is empty
[2024-04-15 21:53:27] pid=1144717 ppid=1135969 RscpMqttMain.cpp(540) Topic >e3dc/pvi/energy_start/string_2< unused? Payload is empty
[2024-04-15 21:53:27] pid=1144717 ppid=1135969 RscpMqttMain.cpp(537) Topic >e3dc/solar/power< forced
[2024-04-15 21:53:27] pid=1144717 ppid=1135969 RscpMqttMain.cpp(540) Topic >e3dc/solar/power< unused? Payload is empty
[2024-04-15 21:53:27] pid=1144717 ppid=1135969 RscpMqttMain.cpp(537) Topic >e3dc/battery/power< forced
[2024-04-15 21:53:27] pid=1144717 ppid=1135969 RscpMqttMain.cpp(540) Topic >e3dc/battery/power< unused? Payload is empty
[2024-04-15 21:53:27] pid=1144717 ppid=1135969 RscpMqttMain.cpp(537) Topic >e3dc/home/power< forced
[2024-04-15 21:53:27] pid=1144717 ppid=1135969 RscpMqttMain.cpp(540) Topic >e3dc/home/power< unused? Payload is empty
[2024-04-15 21:53:27] pid=1144717 ppid=1135969 RscpMqttMain.cpp(537) Topic >e3dc/grid/power< forced
[2024-04-15 21:53:27] pid=1144717 ppid=1135969 RscpMqttMain.cpp(540) Topic >e3dc/grid/power< unused? Payload is empty
[2024-04-15 21:53:27] pid=1144717 ppid=1135969 RscpMqttMain.cpp(537) Topic >e3dc/addon/power< forced
[2024-04-15 21:53:27] pid=1144717 ppid=1135969 RscpMqttMain.cpp(540) Topic >e3dc/addon/power< unused? Payload is empty
[2024-04-15 21:53:27] pid=1144717 ppid=1135969 RscpMqttMain.cpp(537) Topic >e3dc/battery/soc< forced
[2024-04-15 21:53:27] pid=1144717 ppid=1135969 RscpMqttMain.cpp(540) Topic >e3dc/battery/soc< unused? Payload is empty
[2024-04-15 21:53:27] pid=1144717 ppid=1135969 RscpMqttMain.cpp(540) Topic >e3dc/coupling/mode< unused? Payload is empty
[2024-04-15 21:53:27] pid=1144717 ppid=1135969 RscpMqttMain.cpp(540) Topic >e3dc/mode< unused? Payload is empty
[2024-04-15 21:53:27] pid=1144717 ppid=1135969 RscpMqttMain.cpp(540) Topic >e3dc/ems/balanced_phases/L3< unused? Payload is empty
[2024-04-15 21:53:27] pid=1144717 ppid=1135969 RscpMqttMain.cpp(540) Topic >e3dc/ems/balanced_phases/L2< unused? Payload is empty
[... cut ...]
MQTT publish topic >e3dc/pm/voltage/L2< payload >225.43< unit >V<
MQTT publish topic >e3dc/pm/voltage/L3< payload >227.06< unit >V<
MQTT publish topic >e3dc/battery/energy/charge< payload >7.79< unit >kWh<
MQTT publish topic >e3dc/battery/energy/discharge< payload >7.53< unit >kWh<
MQTT publish topic >e3dc/solar/energy< payload >18.27< unit >kWh<
MQTT publish topic >e3dc/grid/energy/in< payload >4.22< unit >kWh<
MQTT publish topic >e3dc/grid/energy/out< payload >1.25< unit >kWh<
MQTT publish topic >e3dc/home/energy< payload >13.02< unit >kWh<
^C

pvtom, 2024-04-15:
Hallo Timmo,
das sieht doch gut aus.
Da haben wir hoffentlich eine Lösung gefunden.
Ja, ich lasse die Software bei mir jetzt auch ein paar Tage laufen.
Offizielles Release dann am Wochenende.
Gruß Thomas

timmost, 2024-04-15:
cool freut mich!
Vg timmo

pvtom, 2024-04-20:
Die Lösung steht jetzt mit v3.20 zur Verfügung.

closed 2024-04-21 by pvtom