pvtom/rscp2mqtt #62: Wunsch: Wallbox Energy per day

MyGitIT, 2024-04-21:
Hallo Thomas,

deine Applikation ist echt super und dein Support perfekt.
Einen Wunsch hätte ich noch, kannst du das gleiche wie bei den Solarstrings auch für die Wallbox machen, energy_total_day und energy_solar_day?
Da gibt es ja schon Topics, die einen enthalten aber nur die letzte Ladung und die anderen die Gesamtladung.

Gruß Sven

pvtom, 2024-04-22:
Hallo Sven,
gute Idee. Habe ich gestern Abend umgesetzt, kommt mit dem nächsten Release v3.21.
Gruß Thomas

pvtom, 2024-04-28:
v3.21 veröffentlicht

MyGitIT, 2024-05-02:
Passt, perfekt.

Vielen Dank Thomas

closed 2024-05-02 by MyGitIT

jomifa, 2024-05-08:
Hallo Thomas,
habe heute festgestellt, dass die Werte für:

e3dc/wallbox/energy/day/total -107644
e3dc/wallbox/energy/day/solar -93269

negativ sind. In den Tagen davor waren sie positiv.

Gruß,
Joachim

pvtom, 2024-05-08:
Hallo Joachim,
die day-Werte liefert das Hauskraftwerk nicht, rscp2mqtt berechnet die Werte so:
Um 0:00 Uhr werden die Werte zu e3dc/wallbox/energy/total und e3dc/wallbox/energy/solar gespeichert und tagsüber werden die dann aktuellen Werte von den 0:00 Uhr - Werten abgezogen und als e3dc/wallbox/energy/day/total und e3dc/wallbox/energy/day/solar ausgegeben.
e3dc/wallbox/energy/total und e3dc/wallbox/energy/solar sind die Werte seit dem letzten Restart/Reset der Wallbox. D.h. wenn es tagsüber einen Reset gibt, sind die Werte am Tag kleiner als die um 0:00 Uhr gemerkten Werte, dann werden die day-Werte negativ.
Das wäre meine Erlärung für Deine Beobachtung. Kannst Du bestätigen, dass die Werte zu e3dc/wallbox/energy/total und e3dc/wallbox/energy/solar zurückgesetzt wurden?
Ich passe das Tool jetzt so an, dass dieser Fall erkannt wird und in diesem Fall die gemerkten Werte aktualisiert werden.

Zur Abregelung der Netzeinspeisung gibt es diese Topics:
e3dc/grid_in_limit wird true, wenn die Einspeisung bei 70% der Leistung abgeregelt wird.
e3dc/grid_in_limit_duration zeigt die Anzahl der Minuten der Dauer der Drosselung am aktuellen Tag. Der Wert wird um Mitternacht wieder auf 0 gesetzt.
Die durch Deinen Solarteur eingestellten Grenzwerte kannst Du hier sehen:
e3dc/system/derate_at_percent_value steht auf 0.7 (es sei denn Deine Anlage ist nicht auf 70% gedrosselt)
e3dc/system/derate_at_power_value zeigt die Leistung an, bei der gedrosselt wird, also den 70% entspricht.

Gruß Thomas

jomifa, 2024-05-08:
Hallo Thomas,
danke dir für deine Ausführungen. Die Wallbox wurde von mir nicht wissentlich zurückgesetzt. Allerdings habe ich versucht das Laden der Wallbox mit den Befehlen:

mosquitto_pub -h 192.168.178.46 -p 1883 -t "e3dc/set/wallbox/suspended" -m true
mosquitto_pub -h 192.168.178.46 -p 1883 -t "e3dc/set/wallbox/toggle" -m true

zeitweise zu unterbinden. Möglicherweise führte das zu dem Problem.

Grüße,
Joachim

pvtom, 2024-05-09:
Hallo Joachim,

wann genau die Werte zu e3dc/wallbox/energy/total und e3dc/wallbox/energy/solar zurückgesetzt werden, ist mir nicht klar.
Wie geschrieben kann ich das auch nicht herausfinden, da ich keine E3/DC-Wallbox besitze.
Es wäre super, wenn Du die Werte zu e3dc/wallbox/energy/total und e3dc/wallbox/energy/solar beobachten könntest, insb. wenn man die suspended/toggle-Befehle einsetzt.
Vielleicht auch interessant sind diese Topics: e3dc/wallbox/energy_start/total und e3dc/wallbox/energy_start/start. Hier werden die nächtlichen Werte gespeichert, die dann am Tag zur Berechnung der day-Werte herangezogen werden (Differenzbildung).
Mein Ehrgeiz, da eine saubere Lösung zu entwickeln, ist ungebrochen ;-)

Einen schönen Feiertag!
Gruß Thomas

jomifa, 2024-05-09:
Hallo Thomas,
dein Ehrgeiz erfreut die E3DC Gemeinde sicherlich sehr. 😉

Heute sind die Werte

e3dc/wallbox/energy/total und e3dc/wallbox/energy/solar

wieder positiv und irgendwann zurückgesetzt worden. Da ich nicht genau weiß, was sie bedeuten, binde ich diese Topics nicht in das Dashboard bzw. in die Influxdb2 ein.

Die von dir erwähnten Topics:

e3dc/wallbox/energy_start/total und e3dc/wallbox/energy_start/start

sind mir noch nicht untergekommen.

Meintest du mit dem Topic:

e3dc/wallbox/energy_start/start

das Topic:

e3dc/wallbox/energy_start/solar?

Eine Frage zum log. Ich habe folgendes setting in der .config:

#--log level, set "1" to get detailled logging
LOG_LEVEL=1
#--log files
LOGFILE=/tmp/rscp2mqtt.log
TOPIC_LOG_FILE=/tmp/rscp2mqtt.history

Allerdings wird kein Logfile geschrieben. Ich sehe nur das history Log.

Was die Drosselung der Anlage angeht, habe ich für das Topic:

e3dc/system/derate_at_percent_value 1.0

gefunden. Das bedeutet, dass meine Anlage nicht gedrosselt wird.

Ich wünsche dir noch einen schönen Vatertag 😊

Grüße,
Joachim

jomifa, 2024-05-16:
Hallo Thomas,
ich habe noch eine Bitte. Könntest du für die Wallbox auch jährliche Werte für Total und Solar berücksichtigen?

Vielen Dank. 😉

Mit freundlichen Grüßen

Joachim Fach

Buchenweg 20
56479 Oberroßbach

Telefon   +49 (0) 2667 83 23 201
Fax          +49 (0) 2667 83 23 203
Mobile   +49 (0) 171 699 88 41
eMail     ***@***.******@***.***>

pvtom, 2024-05-18:
Hallo Joachim,
leider muss ich sagen, dass ich die Umsetzung der jährlichen Wallbox-Werte für nicht machbar halte.
Im Gegensatz zu den Energie-Werten, die das Hauskraftwerk tages-, wochen-, monats- und jahrweise selbst speichert und die man per RSCP abfragen kann, habe ich für die Wallbox keine Tags gefunden, die die Zeitbereiche abdecken.
Daher ist die Lösung für die daily Wallbox-Werte derzeit so umgesetzt, dass das Programm die Energiewerte zum Tageswechsel speichert und beim Laden, die neuen Werte vom Mitternachtswert abzieht und so die Differenz den Tageswert darstellt. (Probleme, dass die Werte falsch waren, weil der Bezugswert plötzlich zurückgesetzt wurde, hast Du ja gemeldet.) DIe Lösung könnte natürlich auch für das ganze Jahr machen, die Fehlerwahrscheinlichkeit wäre mir dann aber zu hoch.
Evtl. ergibt es sich ja, dass mir o.g. Tags doch noch unterkommen, dann kann man das umsetzen.
Gruß Thomas

jomifa, 2024-05-18:
Hallo Thomas,
schade, dass E3/DC keine Topics dafür vorgesehen hat. In der Cloud von E3/DC kann man über die Auswertungen auf beliebige Werte zugreifen. Gibt es eigentlich eine offizielle Beschreibung der Topics von E3/DC?

***@***.***

Grüße und entspannte Pfingsttage,
Joachim

pvtom, 2024-05-19:
Hallo Joachim,
wenn das Portal die Daten bietet, sollte es auch eine Möglichkeit geben, diese per RSCP zu lesen.
Ich habe gerade eine Anfrage an E3/DC gestellt, mit der Bitte, mir eine aktuelle Dokumentation bereitszustellen.
Mal schauen, ob da was kommt.
Ebenfalls frohe Pfingsten!
Danke und Gruß Thomas

jomifa, 2024-06-06:
Hallo Thomas,
hast du bei e3/DC etwas erreichen können?

Gruß Joachim

pvtom, 2024-06-07:
Hallo Joachim,
ich habe bei E3/DC am 19.05.2024 ein Ticket (Nummer ist ITSMSD-158706) erstellt, mit der Bitte, mir eine aktuelle Dokumentation zur RSCP-Schnittstelle bereitzustellen.
Bisher habe ich keine Antwort erhalten.
Gruß Thomas

jomifa, 2024-06-08:
Hallo Thomas,
warum verwundert es mich nicht, dass du die Doku noch nicht erhalten hast. E3/DC ist eines der teuersten Unternehmen, was HKWs angeht und eines der schlechtesten Unternehmen was den support angeht. Wartezeiten von mehr als 2 Monaten für geöffnete Tickets ist keine Seltenheit.

Ich habe folgende Topics gefunden:

e3dc/wallbox/energy/day/total 13563                               ok
e3dc/wallbox/energy/day/solar 13248                              ok
e3dc/wallbox/energy_start/total 218104                          unklar
e3dc/wallbox/energy_start/solar 207468                         unklar
e3dc/wallbox/energy/total 231667                                     unklar
e3dc/wallbox/energy/solar 220716                                     unklar

Wie sind die als unklar bezeichneten Werte zu beurteilen?

Die Jahreswerte können es nicht sein, da ich in diesem Zeitraum über 800kWh geladen habe. Auch die Monatswerte scheiden aus.

Grüße Joachim

pvtom, 2024-06-08:
Hallo Joachim,

ich bin auch mal gespannt, wann E3/DC sich meldet. Solche Anfragen haben wahrscheinlich nicht die höchste Priorität.

Zu den Topics:

e3dc/wallbox/energy/day/total 13563                               ok
e3dc/wallbox/energy/day/solar 13248                              ok
-> das sind die Tageswerte, die rscp2mqtt errechnet.

e3dc/wallbox/energy/total 231667                                     unklar
e3dc/wallbox/energy/solar 220716                                     unklar
-> das sind die Werte, die das Hauskraftwerk liefert. Das sind nicht die Jahreswerte. Die Werte werden immer mal wieder auf 0 zurückgesetzt. Wann? Keine Ahnung.

e3dc/wallbox/energy_start/total 218104                          unklar
e3dc/wallbox/energy_start/solar 207468                         unklar
-> das sind die Werte, die das Hauskraftwerk um 0:00 Uhr geliefert hat.
Die Tageswerte werden dann immer berechnet: "e3dc/wallbox/energy/day/total" = "e3dc/wallbox/energy/total" minus "e3dc/wallbox/energy_start/total". Analog die solar-Werte.

Es gab ja mal bei einem Anwender den Fall, dass die Tageswerte plötzlich negativ waren... weil das Hauskraftwerk oder die Wallbox die e3dc/wallbox/energy/total und e3dc/wallbox/energy/solar - Werte zurückgesetzt hat... da ergab die Berechnung dann komische Werte. Das fange ich mittlerweile ab.

Das erklärt vielleicht auch, warum ich die Rechnerei für Jahreswerte nicht machen möchte... da haben wir mehr Ärger und sicherlich nie genaue Werte...

Ich habe aber gesehen, dass es weitere Tags für die Wallbox im RSCP-Protokoll gibt. Man kann sicherlich auch Jahreswerte direkt abgreifen, wenn man diese bereits im Display bzw. in der App angeboten bekommt. Nur dazu wäre halt die neue Dokumentation hilfreich.

Gruß Thomas

jomifa, 2024-06-26:
Hallo Thomas,
eine kleine Verbesserung für die Darstellung der Graphen im Leistungsdiagramm von node red dashboard. Dadurch werden die Linien der Graphen dünner dargestellt  und verschmieren somit weniger.

var topic;
var payload;
var ui_control;

for (let i = 0; i < 6; i++) {
    switch (i) {
        case 0:
                topic = null;
                payload = []; // clear chart
                break;
        case 1: topic = 'PV [W]';
                payload = 0;
                break;
        case 2: topic = 'Hausverbrauch [W]';
                payload = 0;
                break;
        case 3: topic = 'Netz [W]';
                payload = 0;
                break;
        case 4: topic = 'Akku [W]';
                payload = 0;
                break;
        case 5: topic = 'Wallbox [W]';
                payload = 0;
                break;
        }

        ui_control = {
                "options": {
                        "elements": {
                                "line": {
                                        "borderWidth": 1,
                                        "fill": false
                                },
                                "point": {
                                        "radius": 0
                                }
                        }
                }
        }

    var newmsg = {payload:payload,topic:topic,ui_control:ui_control};
    node.send(newmsg);
}
return;

Grüße
Joachim

Mit freundlichen Grüßen

Joachim Fach

Buchenweg 20
56479 Oberroßbach

Telefon   +49 (0) 2667 83 23 201
Fax          +49 (0) 2667 83 23 203
Mobile   +49 (0) 171 699 88 41
eMail     ***@***.******@***.***>

pvtom, 2024-07-07:
Hallo Joachim,
ich habe die dünneren Linien bei mir lokal ausprobiert.
Stimmt, das sieht besser aus.
Ich werde die Änderung demnächst in das offizielle Paket übernehmen.
Danke Dir
und Gruß Thomas