pvtom/rscp2mqtt #63: e3dc/set/wallbox/control stop

schnidrig, 2024-04-25:
Hallo 

Hab heute ein wenig mit der Wallbox gespielt. Dabei ist mir folgendes aufgefallen:

1) "e3dc/set/wallbox/control stop" ist eigentlich ein Schalter. Man kann damit das laden aus aber auch wieder einschalten. Daher ist der Name vielleicht etwas unglücklich. Es ist eher "start_stop"

2) "e3dc/set/wallbox/control stop" verändert den maximalen Strom. Dies ist ein unangenehmer Nebeneffekt. Könnte man nicht den aktuellen max_current beibehalten?

pvtom, 2024-04-26:
Hallo Christian,
da ich keine Wallbox von E3/DC besitze, kann ich das Verhalten leider nicht selbst testen.
Ich kann im Programmcode aber nachvollziehen, dass der Stromwert und der Sonnen/Mix-Modus geändert wird.
Ich bin davon ausgegangen, dass "stop" das Laden beendet und z.B. "e3dc/set/wallbox/control" "solar:16" das Laden startet. Funktioniert das Einschalten so?
Wenn ja, sollte man überlegen, ob man "stop" umbenennt und die Werte rettet oder bei abgeschaltetem Laden den "stop"-Befehl einfach ignoriert.
Gruß
Thomas

schnidrig, 2024-04-26:
Ja kann ich gut verstehen. Die TAG Namen in RSCP sind nicht selbsterklärend. Aber hey! Vielen Dank für diese Projekt. Ich finde es super hilfreich.

Nein die Schalter funktionieren nicht wie du denkst.

- stop ist ein toggle: ein/aus
- solar:16 und mix:16 sind beide reine Configuration. Diese starten das Laden nicht.

Insofern ist der Name von "e3dc/wallbox/canceled" eigentlich auch falsch. Es wäre eher "e3dc/wallbox/suspended". Im E3DC Portal steht dann "Ladevorgang pausiert".

Persönlich finde ich es eher mühsam, dass du Strom und Modus Config zusammenfasst. Im UI sind die auch getrennt und ich musste erst mal nachschauen gehn, welche Stromstärke bei mir eigentlich konfiguriert ist.
Ich switche gelegentlich zwischen mix und solar, die Stromstärke verändere ich absolut nie. Wüsste nicht wozu.

Ich poste mal einen Screenshot des UIs:

[image]

schnidrig, 2024-04-26:
ich bin dabei dies in Openhab zu integrieren. 
Es wäre einfacher, wenn da einfach ein Schalter für Sonnenmodus ein/aus wäre wie im UI anstatt solar/mix.

pvtom, 2024-04-28:
Hallo Christian,
ich habe mit dem neuen Release v3.21 einige Deiner Ideen umgesetzt, siehe WALLBOX.md.
Ich hoffe, es funktioniert alles?!
Es gibt zur Steuerung jetzt diese Topics:
- e3dc/set/wallbox/sun_mode true/false
- e3dc/set/wallbox/max_current
- e3dc/set/wallbox/charge (true zum Einschalten des Ladens / false zum Abschalten)
- e3dc/set/wallbox/toggle (analog zu "stop")
Gruß Thomas

schnidrig, 2024-04-28:
Hallo Thomas

Vielen Dank für die Anpassungen

Bei meinen ersten Tests hab ich folgendes gefunden:

1) Ich hab jede Sekunde folgenden Fehler im log:

````
rscp2mqtt-1  | [2024-04-28 16:29:35] pid=1 ppid=0 RscpMqttMain.cpp(1871) Error: Container >TAG_WB_DATA< Tag >TAG_WB_REQ_AVAILABLE_SOLAR_POWER< received error code >UNKNOWN_TAG<.
````

2) Das Setzen von `e3dc/set/wallbox/charge`, `e3dc/set/wallbox/toggle`, `e3dc/set/wallbox/sun_mode`,  `e3dc/set/wallbox/max_current` generiert folgenden Fehler:

````
rscp2mqtt-1  | [2024-04-28 16:43:36] pid=1 ppid=0 RscpMqttMain.cpp(1871) Error: Container >TAG_WB_DATA< Tag >TAG_WB_EXTERN_DATA< received error code >UNKNOWN_TAG<.
`````

Dies scheint jedoch keinen Einfluss auf die Funktion zu haben. toggle, max_current und sun_mode funktionieren wie erwartet.

3) `e3dc/set/wallbox/charge` 
- setzen auf `true` funktioniert als toggle gleich wie `e3dc/set/wallbox/toggle`
- setzen auf `false` macht überhaupt nichts.

4) Doku in der Datei: `WALLBOX.md`
steht noch `e3dc/wallbox/canceled` anstelle von `e3dc/wallbox/suspended`

---
Noch ein Hinweis. Die Integration in Openhab ist ungleich schwerer, wenn das Kommando nicht dem Status entspricht:
- e3dc/set/wallbox/charge = true bewirkt e3dc/wallbox/suspended = false
- e3dc/set/wallbox/charge = false bewirkt e3dc/wallbox/suspended = true
man muss dann noch zusätzlich eine "transformation" für die Werte einbauen. Kein unlösbares Problem aber irgendwie unnötig. Weiss nicht wie andere Heimautomationsysteme damit umgehen.

Gruss & Danke
Christian

pvtom, 2024-04-28:
Hallo Christian,
danke für Deinen Test. Schade, dass es keine Punktladung war...
zu 1) ich habe den Tag TAG_WB_REQ_AVAILABLE_SOLAR_POWER neu aufgenommen. Frage wäre, ob der zugehörende Topic e3dc/wallbox/available_solar_power erzeugt wird? Wenn nein, nehme ich den Tag wieder raus.
zu 2) gut, dass die drei set-Topics grundsätzlich funktionieren. Mal schauen, ob ich ermitteln kann, warum die Fehlermeldung auftaucht... oder ich fange sie ab....
zu 3) e3dc/set/wallbox/charge habe ich so umgesetzt, dass der toggle-Befehl ausgeführt wird, wenn der mit charge gewünschte Status true/false nicht mit dem Ladestatus, den e3dc/wallbox/charging meldet, übereinstimmt. Evtl. habe ich da noch einen Denkfehler. Kannst Du mal schauen, ob e3dc/wallbox/charging ausgegeben wird und zum Zustand der Wallbox (true = Laden, false = nicht Laden) übereinstimmt?
zu 4) stimmt, die Doku sollte passend sein
zu der Anmerkung, dass set/wallbox/charge zu wallbox/suspended passen sollte... kann ich anpassen. 
Was ist inhaltlich der Unterschied zwischen e3dc/wallbox/charging und e3dc/wallbox/suspended? Vielleicht liegt hier der Fehler beim charge-Befehl.
Gruß Thomas

pvtom, 2024-04-28:
Übrigens kann man das nervige Loggen des Fehlers mit LOG_LEVEL=0 in der .config unterdrücken... ;-)

pvtom, 2024-04-28:
Hallo Christian,
nochmal zu 3) kann es sein, dass e3dc/wallbox/charging dann true ist, wenn tatsächlich geladen wird, d.h. das Hauskraftwerk es entschieden hat, weil z.B. Überschuss da ist, und e3dc/wallbox/suspended heißt, wenn es true ist, dass das Laden unterdrückt wird (wie der Name schon sagt), d.h. der Anwender kein Laden wünscht... dann müsste ich den suspend-Wert bei e3dc/set/wallbox/charge benutzen...
Gruß Thomas

schnidrig, 2024-04-28:
Richtig. e3dc/wallbox/charging ist true wenn tatsächlich geladen wird und
e3dc/wallbox/suspended ist true wenn der User das Laden unterdrückt.

schnidrig, 2024-04-28:
das Topic e3dc/wallbox/available_solar_power hab ich nie gesehn.

pvtom, 2024-04-29:
Hallo Christian,
danke für die Aufklärung.
Dann werde ich folgendes anpassen:
zu 1) der Tag wird entfernt
zu 3) aus "charge" wird "suspended". Dann passt der set-Befehl zum Status-Topic.
zu 4) die Doku wird angepasst
Ich werde versuchen, heute Abend ein neues Release zu veröffentlichen.
Gruß Thomas

pvtom, 2024-04-30:
Hallo Christian,
die Anpassungen sind jetzt mit Release v3.22 verfügbar.
Gruß Thomas

schnidrig, 2024-04-30:
Hallo Thomas

Nun funktioniert `e3dc/set/wallbox/suspended` wie erwartet.

Es wird allerdings immer noch folgender Fehler geloggt:
````
rscp2mqtt-1  | [2024-04-30 11:23:48] pid=1 ppid=0 RscpMqttMain.cpp(1870) Error: Container >TAG_WB_DATA< Tag >TAG_WB_EXTERN_DATA< received error code >UNKNOWN_TAG<.
````
Vielen Dank & Gruss
Christian

pvtom, 2024-04-30:
Hallo Christian,
dass der suspended-Befehl jetzt funktioniert, freut mich sehr.
Kommt die Fehlermeldung nur einmalig, nachdem ein e3dc/set/wallbox/xyz gesendet wurde - also als Antwort auf den set-Befehl? Oder kommt der Fehler alle Sekunde (bzw. im konfigurierten Intervall)?

Gruß Thomas

pvtom, 2024-04-30:
Okay, das ist ja dann nicht so dramatisch.
Ich schaue mir das trotzdem an, vielleicht finde ich etwas...

timmost, 2024-05-01:
Hallo Tom

zu 1. - sehe diese Entries im Log:
rscp2mqtt [v3.22]
(...)
[2024-05-01 07:48:14] pid=1 ppid=0 RscpMqttMain.cpp(382) MQTT: starting listener loop
[2024-05-01 07:48:15] pid=1 ppid=0 RscpMqttMain.cpp(1907) Error: Container >TAG_PVI_DATA< Tag >TAG_PVI_REQ_COS_PHI< received error code >NOT_AVAILABLE<.
[2024-05-01 07:48:15] pid=1 ppid=0 RscpMqttMain.cpp(1907) Error: Container >TAG_PVI_DATA< Tag >TAG_PVI_REQ_VOLTAGE_MONITORING< received error code >NOT_AVAILABLE<.
[2024-05-01 07:48:15] pid=1 ppid=0 RscpMqttMain.cpp(1907) Error: Container >TAG_PVI_DATA< Tag >TAG_PVI_REQ_FREQUENCY_UNDER_OVER< received error code >NOT_AVAILABLE<.

Wahrscheinlich ein Schönheitsfehler, hatte die WB heute wieder angestellt (ich stelle sie normalerweise an der Sicherung aus, da das E-Auto nicht vorhanden ist - und die WB ständig die Einfahrt mit der grünen Leuchte ausleuchtet). Könnte man  evtl durch eine Verfügbarkeitsabfrage klären.

VG timmo

pvtom, 2024-05-01:
Hallo Timmo,
die Meldungen kommen beim Programmstart, wenn die Messeinrichtung des Hauskraftwerks die Werte z.B. zum cos phi nicht ausgibt. Ist bei meinem Hauskraftwerk auch so. rscp2mqtt fragt die Werte dann nicht mehr ab, nur beim Start.
Das hat nichts mit der Wallbox zu tun. Die Meldungen wurden schon seit einigen Releases ausgegeben. Einfach ignorieren ;-)
Gruß Thomas

timmost, 2024-05-01:
Ok thx

Timmo Sturm
Wittelsbacherstr. 8
85622 Feldkirchen
+49 151 402 71 711

closed 2024-07-11 by pvtom