pvtom/rscp2mqtt #65: Wallbox shut off - log entires

timmost, 2024-05-04:
Hi Tom

noch ein Schönheitsfehler - ich habe zwar eine Wallbox, nehme aber meist die Sicherung raus, da das Elektroauto noch auf sich warten lässt (und die LED an der Wallbox in der Nacht so hell ist). 

Wenn die Wallbox vom Netz getrennt ist, wird regelmäßig dieser Eintrag ins Log geschrieben - vielleicht kann man das auch noch rauswerfen (keine Wallbox, dann keine Abfrage).

VG timmo

rscp2mqtt [v3.22]
E3DC system >192.168.178.58:5033< user: >n.n.<
MQTT broker >localhost:1883< qos = >0< retain = >false< client id >✗< prefix >e3dc<
Fetching data every 5 seconds.
Requesting PVI ✓ | PM (0) | DCB ✓ (1 battery string) | Wallbox (0) ✓ | Autorefresh ✗
Log level = 1
Stdout to terminal

[2024-05-01 07:48:14] pid=1 ppid=0 RscpMqttMain.cpp(2804) Connecting to server 192.168.178.58:5033
[2024-05-01 07:48:14] pid=1 ppid=0 RscpMqttMain.cpp(2811) Success: E3DC connected.
[2024-05-01 07:48:14] pid=1 ppid=0 RscpMqttMain.cpp(1790) RSCP authentication level 10
[2024-05-01 07:48:14] pid=1 ppid=0 RscpMqttMain.cpp(2324) Connecting to broker localhost:1883
[2024-05-01 07:48:14] pid=1 ppid=0 RscpMqttMain.cpp(2336) Success: MQTT broker connected.
[2024-05-01 07:48:14] pid=1 ppid=0 RscpMqttMain.cpp(382) MQTT: starting listener loop
[2024-05-01 07:48:15] pid=1 ppid=0 RscpMqttMain.cpp(1907) Error: Container >TAG_PVI_DATA< Tag >TAG_PVI_REQ_COS_PHI< received error code >NOT_AVAILABLE<.
[2024-05-01 07:48:15] pid=1 ppid=0 RscpMqttMain.cpp(1907) Error: Container >TAG_PVI_DATA< Tag >TAG_PVI_REQ_VOLTAGE_MONITORING< received error code >NOT_AVAILABLE<.
[2024-05-01 07:48:15] pid=1 ppid=0 RscpMqttMain.cpp(1907) Error: Container >TAG_PVI_DATA< Tag >TAG_PVI_REQ_FREQUENCY_UNDER_OVER< received error code >NOT_AVAILABLE<.
[2024-05-02 00:00:12] pid=1 ppid=0 RscpMqttMain.cpp(1770) Error: Container/Tag >TAG_DB_HISTORY_DATA_DAY< received error code >(null)<.
[2024-05-03 00:00:22] pid=1 ppid=0 RscpMqttMain.cpp(1770) Error: Container/Tag >TAG_DB_HISTORY_DATA_DAY< received error code >(null)<.
[2024-05-03 00:00:37] pid=1 ppid=0 RscpMqttMain.cpp(1770) Error: Container/Tag >TAG_DB_HISTORY_DATA_DAY< received error code >(null)<.
[2024-05-03 00:00:51] pid=1 ppid=0 RscpMqttMain.cpp(1770) Error: Container/Tag >TAG_DB_HISTORY_DATA_DAY< received error code >(null)<.
[2024-05-03 00:01:06] pid=1 ppid=0 RscpMqttMain.cpp(1770) Error: Container/Tag >TAG_DB_HISTORY_DATA_DAY< received error code >(null)<.

pvtom, 2024-05-04:
Hallo Timmo,
die Meldung "Error: Container/Tag >TAG_DB_HISTORY_DATA_DAY< received error code >(null)<" kommt bei mir kurz nach 0:00 Uhr ebenfalls mehrfach.
Das Hauskraftwerk kann die Daten für den gerade angebrochenen Tag noch nicht liefern. Die History-Tags werden erfahrungsgemäß alle 15 Minuten erzeugt. Der Fehler kommt wahrscheinlich so lange bis die 15 Minuten wieder um sind.
Das hat also nichts mit Deiner abgeschalteten Wallbox zu tun.
Gruß Thomas

pvtom, 2024-05-07:
Mit dem kommenden Release v3.23 kann man das Logging so konfigurieren, dass Fehlermeldungen (=Meldungen, die das Hauskraftwerk als Antwort einer Anfrage zu einem Tag ausgibt) gesammelt werden und gleiche Fehlermeldungen mit einem Zähler für die Anzahl des Auftretens versehen werden. Die Übersicht der Fehler wird dann gebündelt um Mitternacht, zum Programmende oder durch Abfrage mit e3dc/set/log/errors ausgegeben.
Das reduziert die Fehlermeldungen.
Man kann das Loggen von solchen Meldungen auch ganz abschalten. Das geht heute schon mit LOG_LEVEL=0
Ich schließe damit dieses Issue.
Gruß Thomas

closed 2024-05-07 by pvtom

timmost, 2024-05-07:
Super danke!
Irgendwann solltest Du das mal an E3DC lizenzieren…das ist super mächtig,
was Du hier baust!

Viele Grüße - Timmo

Timmo Sturm
Wittelsbacherstr. 8
85622 Feldkirchen
+49 151 402 71 711