pvtom/rscp2mqtt #66: RSCP authentication level 0

dejmf, 2024-05-05:
Hello Tom,
after I have installed the program rscp2mqtt according to the instructions, I get a message when I start the program. 

pi@Pi4JMF:/opt/rscp2mqtt $ ./rscp2mqtt
rscp2mqtt [v3.22.influxdb]
E3DC system >192.168.178.31:5033< user: >joachim.fach@jmf.de    <
MQTT broker >192.168.178.46:1883< qos = >0< retain = >false< client id >✗< prefix >e3dc<
INFLUXDB v2 >192.168.178.90:8086< orga = >jmf< bucket = >e3dc< measurements = >e3dc< and >e3dc_meta<
Fetching data every second.
Requesting PVI ✓ | PM (0) | DCB ✓ (1 battery string) | Wallbox (0) ✓ | Autorefresh ✗
Log level = 0
Stdout to terminal

[2024-05-05 18:42:40] pid=48814 ppid=46736 RscpMqttMain.cpp(2804) Connecting to server 192.168.178.31:5033
[2024-05-05 18:42:40] pid=48814 ppid=46736 RscpMqttMain.cpp(2811) Success: E3DC connected.
[2024-05-05 18:42:41] pid=48814 ppid=46736 RscpMqttMain.cpp(1790) RSCP authentication level 0
[2024-05-05 18:42:41] pid=48814 ppid=46736 RscpMqttMain.cpp(2324) Connecting to broker 192.168.178.46:1883
[2024-05-05 18:42:41] pid=48814 ppid=46736 RscpMqttMain.cpp(2336) Success: MQTT broker connected.
[2024-05-05 18:42:41] pid=48814 ppid=46736 RscpMqttMain.cpp(382) MQTT: starting listener loop
[2024-05-05 18:42:43] pid=48814 ppid=46736 RscpMqttMain.cpp(1790) RSCP authentication level 0
[2024-05-05 18:42:45] pid=48814 ppid=46736 RscpMqttMain.cpp(1790) RSCP authentication level 0
[2024-05-05 18:42:45] pid=48814 ppid=46736 RscpMqttMain.cpp(529) Setup topic >e3dc/set/up/0x00_0x00_0x0E< payload >0< date >20240505<
[2024-05-05 18:42:45] pid=48814 ppid=46736 RscpMqttMain.cpp(529) Setup topic >e3dc/set/up/0x00_0x00_0x0F< payload >0< date >20240505<
[2024-05-05 18:42:47] pid=48814 ppid=46736 RscpMqttMain.cpp(1790) **RSCP authentication level 0**
[2024-05-05 18:42:49] pid=48814 ppid=46736 RscpMqttMain.cpp(1790) RSCP authentication level 0
[2024-05-05 18:42:49] pid=48814 ppid=46736 RscpMqttMain.cpp(1790) RSCP authentication level 0
[2024-05-05 18:42:49] pid=48814 ppid=46736 RscpMqttMain.cpp(1790) RSCP authentication level 0
[2024-05-05 18:42:50] pid=48814 ppid=46736 RscpMqttMain.cpp(1790) RSCP authentication level 0

The program seems to loop at this point. I have installed the latest version from 29.03.2024 on a Raspberry 5 under Debian 12. Do you have any idea what the problem could be?

Kind regards,
Joachim

pvtom, 2024-05-05:
Hallo Joachim,

"RSCP authentication level 0" bedeutet, dass das Einloggen ins Hauskraftwerk nicht funktioniert hat.

Sind die Passwörter in der .config korrekt gesetzt?

E3DC_PASSWORD=(Portal-Passwort)
E3DC_AES_PASSWORD=(neu zuvergebenes Passwort)

Ist RSCP im Hauskraftwerk freigeschaltet, d.h. das RSCP-Passwort (gleich dem Wert von E3DC_AES_PASSWORD) gesetzt?
Siehe dazu auch die Issues #31 (https://github.com/pvtom/rscp2mqtt/issues/31) und #33.

Ich hoffe, das hilft schon weiter?

Gruß Thomas

dejmf, 2024-05-05:
Hallo Thomas,
vielen Dank für deine schnelle Reaktion.
Ich habe das Portal- und AES-Passwort, welches in der S10 RSCP-Passwort genannt wird, sowohl in der .config als auch in der S10 gesetzt.

Ich habe nochmals das RSCP-Passwort in der S10 gesetzt (7 Zeichen) und in die .config eingetragen und das RSCP2MQTT Programm ausgeführt. Immer mit demselben Ergebnis(RSCP authentication level 0) .

Was mir noch aufgefallen ist, ist, wenn ich das RSCP-Passwort in der S10 ändere und übernehme, dass danach immer nur 4 * angezeigt werden.

Ergo muss irgendetwas beim Setzen des RSCP-Passwortes nicht korrekt laufen. Ich habe aber keine Idee was.

Grüße
Joachim

pvtom, 2024-05-06:
Hallo Joachim,

wenn ich oben in den Chat schaue, fällt mir auf, dass hinter Deinem User, also der EMail-Adresse ein Leerzeichen steht:

E3DC system >192.168.178.31:5033< user: >joachim.fach@jmf.de <

Bitte in der .config in der Zeile mit E3DC_USER hinten das Leerzeichen entfernen.

Dass nur vier Sternchen angezeigt werden, ist bei mir auch so.

Gruß Thomas

dejmf, 2024-05-06:
Guten Morgen Thomas,
vielen Dank für deinen Hinweis. Das war des Rätsels Lösung. 😉

Ich wünsche dir einen guten Start in die Woche.

Gruß,
Joachim

pvtom, 2024-05-06:
Das freut mich! Viel Spaß mit dem Tool!
Gruß Thomas

closed 2024-05-06 by pvtom

dejmf, 2024-05-07:
Hallo Thomas,
ich habe dein Dashboard um die Wallbox erweitert. Dabei ist mir aufgefallen, dass die tägliche Energie:

e3dc/wallbox/energy/day/total 5183
e3dc/wallbox/energy/day/solar 4797

möglicherweise falsch dargestellt wird.

Das Dashboard von E3DC zeigt für denselben Zeitpunkt andere Werte an:

***@***.***

Außerdem verstehe ich die Werte für:

e3dc/wallbox/energy/total 96949
e3dc/wallbox/energy/solar 86609

nicht. Welcher Zeitraum dient hier als Grundlage? Aus meiner Sicht ist es weder der Jahres- noch der Monatszeitraum.

Wie bekomme ich die Werte für:

e3dc/ems/emergency_power_available
e3dc/ems/emergency_power_status

Diese Werte werden offensichtlich nicht zyklisch gesendet.

Grüße,
Joachim

pvtom, 2024-05-07:
Hallo Joachim,
das freut mich, dass Du das Dashboard erweitern willst.

Ich habe keine E3/DC-Wallbox, daher bin ich bei Fragen dazu auf Unterstützung anderer Anwender angewiesen.
Daher verweise ich hier auf folgende Issues.

Unter Issue #49 findet man einen Chat zu den Wallbox total und solar - Werten.
e3dc/wallbox/energy/total  und (solar) sind die Energiewerte in Wh seit dem letzten Reset/Restart (?) der Wallbox.

Mit Issue #62 sind die day - Werte dazu gekommen, die die Wh ab 0:00 Uhr darstellen, d.h. alle Ladungen an einem Tag.
Wenn da falsche Werte ausgegeben werden und Unsicherheiten dazu da sind, bitte einfach #62 kommentieren und um Hilfe bitten... ich kann ja leider nicht so richtig helfen ;-)

Daneben gibt es noch last_charging - Werte für die letzte Ladung (auch wenn sie am Vortag gestartet wurde).
Die Werte werden beim Abstöpseln bzw. beim erneuten Anstöpseln des E-Autos zurückgesetzt.

e3dc/ems/emergency_power_available und e3dc/ems/emergency_power_status werden bei jedem Abfragezyklus vom Hauskraftwerk abgefragt. Ausgegeben werden sie aber nur, wenn sich der Wert geändert hat.
Das ist das Standardverhalten von rscp2mqtt. D.h. nur geänderte Werte führen dazu, dass das Topic aufgegeben wird. Das Verhalten kann man für zu benennende Topics ändern, wenn man in der .config z.B.
FORCE_PUB=e3dc/ems/.*
einfügt. (Syntax sind hier "Reguläre Ausdrücke", so steht das .* für beliebige Zeichen).
Damit werden alle Topics, die mit e3dc/ems beginnen in jedem Zyklus (also z.B. alle 2 Sekunden, wenn INTERVAL=2 ist) ausgegeben.

Gruß Thomas