pvtom/rscp2mqtt #67: Absturz nach mehreren Stunden

lorbetzki, 2024-05-07:
Hi,

ich bekomme nach einigen Stunden folgenden Fehler
[image]

`RscpMqttMain.cpp(2205) Error: Socketreceive Error errno 4
`es wird zwar eine neue Verbindung aufgebaut, aber keine Daten empfangen. Danach breche ich mit Strg+c ab und bekomme 
`double free or corruption (out)`

Der Fehler tritt seit Version 3.19 auf, davor habe ich den Fehler nicht. Eine Idee?

pvtom, 2024-05-07:
Hi Kristian,
immer wieder spannend, was so auftaucht.
errno 4 ist der Output der recv-Funktion (Lesen vom Netzwerk-Socket) und steht für EINTR.
Der Status wird derzeit nicht korrekt behandelt.
Ich hoffe, Google hat hier recht und den richtigen Vorschlag gemacht.
Wenn Du möchtest, könntest Du Zeile 2204 in der Datei RscpMqttMain.cpp so abändern:

if ((errno == EAGAIN) || (errno == EWOULDBLOCK) || (errno == EINTR)) continue;

Ich hoffe, das hilft?!

Gruß Thomas

lorbetzki, 2024-05-07:
Alles klar, ich habe die Änderung vorgenommen, kompiliert und gestartet. Ich melde mich. Danke schonmal

lorbetzki, 2024-05-10:
alles ich würde sagen, der Fehler ist mit obiger Änderung behoben. Seither keine Abstürze mehr. Vielen Dank für die prompte Unterstützung

pvtom, 2024-05-10:
Das freut mich, prima. Danke fürs Testen!

pvtom, 2024-05-12:
Die Lösung ist jetzt in das neue Release v3.23 eingeflossen.

closed 2024-05-12 by pvtom