pvtom/rscp2mqtt #69: Hallo, hatte es irgendwie mal zum laufen, bin noch bei 3.12, kann mir jemand helfen wie ich auf eine neue Version updaten kann? apt update && apt upgrade halfen nichts, bin da doch recht neu unterwegs und dem Gebiet.

bmause, 2024-05-14:
Hallo, hatte es irgendwie mal zum laufen, bin noch bei 3.12, kann mir jemand helfen wie ich auf eine neue Version updaten kann? apt update && apt upgrade halfen nichts, bin da doch recht neu unterwegs und dem Gebiet. 
Läuft bei mir unter Proxmox in einen LXC Container.
Danke euch 

_Ursprünglich gepostet von @bmause in https://github.com/pvtom/rscp2mqtt/issues/64#issuecomment-2110891742_

pvtom, 2024-05-14:
Hallo, ich hoffe jemand aus der Community kann das beantworten.
Ich kenne Proxmox leider nicht. Wo hast Du denn den Container zu Version 3.12 her bekommen?
Gruß Thomas

pvtom, 2024-05-15:
Moin,
Du kannst die neue Version, der Anleitung in der README.md folgend, laden und kompilieren:

Cloning the Repository:
git clone https://github.com/pvtom/rscp2mqtt.git
cd rscp2mqtt

Compilation:
To build a program version without InfluxDB use:
make
To build a program version including InfluxDB support use:
make WITH_INFLUXDB=yes

Dann erhälst Du das ausführbare Programm rscp2mqtt.

Oder Du nutzt Docker.

Gruß Thomas

bmause, 2024-05-15:
Danke für den Tipp, werde ich ausprobieren 👍

bmause, 2024-05-17:
So nun klappte alles, es lief der dienst imm Hintergrund noch, und deswegen konnte ich das nicht richtig ausführen. :-)
Besten dank an euch