pvtom/rscp2mqtt #70: Error messages when starting RSCPM2QTT

jomifa, 2024-05-18:
Hello,
after reinstalling on another Raspbeery Pi, I get the following error messages when starting rscp2mqtt (version 3.33).

pi@Pi5JMF:/opt/rscp2mqtt $ ./rscp2mqtt
rscp2mqtt [3.23]
E3DC system >192.168.178.31:5033< user: >joachim.fach@jmf.de<
MQTT broker >192.168.178.90:1883< qos = >0< retain = >false< client id >✗< prefix >e3dc<
Fetching data every 30 seconds.
Requesting PVI ✓ | PM (0) | DCB ✓ (2 battery strings) | Wallbox (0) ✓ | Autorefresh ✗
Log Level = 1 (ON)
Stdout to terminal

[2024-05-18 15:37:03] pid=1748 ppid=922 RscpMqttMain.cpp(2952) Connecting to server 192.168.178.31:5033
[2024-05-18 15:37:03] pid=1748 ppid=922 RscpMqttMain.cpp(2959) Success: E3DC connected.
[2024-05-18 15:37:03] pid=1748 ppid=922 RscpMqttMain.cpp(1896) RSCP authentication level 10
[2024-05-18 15:37:03] pid=1748 ppid=922 RscpMqttMain.cpp(2428) Connecting to broker 192.168.178.90:1883
[2024-05-18 15:37:03] pid=1748 ppid=922 RscpMqttMain.cpp(2440) Success: MQTT broker connected.
[2024-05-18 15:37:03] pid=1748 ppid=922 RscpMqttMain.cpp(392) MQTT: starting listener loop
**[2024-05-18 15:37:05] pid=1748 ppid=922 RscpMqttMain.cpp(2011) Error: Container >TAG_PVI_DATA< Tag >TAG_PVI_REQ_COS_PHI< received >NOT_AVAILABLE< [1]
[2024-05-18 15:37:05] pid=1748 ppid=922 RscpMqttMain.cpp(2011) Error: Container >TAG_PVI_DATA< Tag >TAG_PVI_REQ_VOLTAGE_MONITORING< received >NOT_AVAILABLE< [1]
[2024-05-18 15:37:05] pid=1748 ppid=922 RscpMqttMain.cpp(2011) Error: Container >TAG_PVI_DATA< Tag >TAG_PVI_REQ_FREQUENCY_UNDER_OVER< received >NOT_AVAILABLE< [1]
[2024-05-18 15:37:07] pid=1748 ppid=922 RscpMqttMain.cpp(1876) Error: Container/Tag >TAG_DB_HISTORY_DATA_YEAR< received >UNKNOWN< [1]**
[2024-05-18 15:37:07] pid=1748 ppid=922 RscpMqttMain.cpp(1879) History start year has been corrected >2023<

pvtom, 2024-05-18:
Hello Joachim,
the first 3 messages appear when the program is started if the measuring device of the house power plant does not measure and output the values, e.g. for cos phi. As mentioned in issue #63, rscp2mqtt then no longer queries the values, only at startup. Simply ignore it.
Message 4: When did you install your device? The message “History start year has been corrected >2023<” indicates that the configuration of the value for HISTORY_START_YEAR (.config) was not set to the year of commissioning.
Could this be the case?
Best regards
Thomas

jomifa, 2024-05-18:
Hello Thomas,
thank you very much for your explanations. I was only interested in the clarification of the first 3 messages. The rest is clear.

Greetings, Joachim

closed 2024-06-01 by pvtom