pvtom/rscp2mqtt #71: Nulleinspeisung

rhofer1980, 2024-05-23:
Gibt es die Möglichkeit, die Einspeisung von PV Leistung auf null zu setzen? Wäre interessant für dynamische Stromtarife. 
Danke!

pvtom, 2024-05-23:
Hallo,
ich bin mir nicht sicher, ob ich die Frage richtig verstanden habe.
Die PV-Leistung soll heruntergeregelt werden, damit es nicht zum Einpeisen ins Netz kommt?
Gibt es eine entsprechende Funktion über das Display des Hauskraftwerks oder über das Portal?
Gruß Thomas

rhofer1980, 2024-05-24:
Die Funktion gibt es, ist über das Display einstellbar: "Vereinbarte Anschlusswirkleistung"
[image]

pvtom, 2024-05-24:
Das ist die Stelle, an der die berühmten 70% für das Einspeiselimit, wenn es für die Anlage gilt, eingestellt werden.
Das darf man aber nur mit speziellen Rechten. Der Solarteur hat das einmal eingestellt, das war's. Bei mir ist die Eingabe jetzt gesperrt. Kann sein, dass das auch über Schnittstelle geht... aber wer hat schon Solarteurs-Rechte?
Ich habe noch nicht verstanden, was der Sinn einer Reduzierung sein sollte. Warum soll man seine Anlage begrenzen, wenn die Sonne brennt und die Leistung ansteht?

rhofer1980, 2024-05-25:
Der Sinn besteht darin, wenn man zum Spotpreis einspeist kann der Einspeisetarif auch negativ werden, dann zahlt man fürs einspeisen. Das würde ich gerne verhindern.

pvtom, 2024-06-01:
@world: Hat hierzu jemand aus der Community eine Idee?