pvtom/rscp2mqtt #72: Make error 127

Frigo759, 2024-06-01:
Hallo,

ich versuche gerade rscp2mqtt in einem LXC Container (Debian 12) unter Proxmox 8.2.2 zu installieren.
Bei der Installation erhalte ich folgende Fehlermeldung.

`root@rscp2mqtt:~/rscp2mqtt# make WITH_INFLUXDB=yes`
`rm rscp2mqtt`
`rm: cannot remove 'rscp2mqtt': No such file or directory`
`make: [Makefile:14: clean] Error 1 (ignored)`
`g++ -O3 RscpMqttMain.cpp RscpProtocol.cpp AES.cpp SocketConnection.cpp -pthread -lmosquitto -lcurl -o rscp2mqtt -DINFLUXDB`
`make: g++: No such file or directory`
`make: *** [Makefile:8: rscp2mqtt] Error 127`

`root@rscp2mqtt:~/rscp2mqtt# ls`
`AES.cpp               MQTT_TLS.md        RscpProtocol.cpp      TOPICS.md
AES.h                 Makefile           RscpProtocol.h        WALLBOX.md
DOCKER.md             README.md          RscpTags.h            config.template
Dockerfile            RELEASE.md         RscpTagsOverview.h    request_days.sh
HA_MQTT_DISCOVERY.md  RscpMqttConfig.h   RscpTypes.h           rscp2mqtt.service
INFLUXDB.md           RscpMqttMain.cpp   SocketConnection.cpp
LICENSE               RscpMqttMapping.h  SocketConnection.h`

#3 habe ich mir auch schon angeguckt, hat mich aber irgendwie auch nicht weiter gebracht.

mit freundlichem Gruß
Christian

pvtom, 2024-06-01:
Hi,
die Fehlermeldung deutet daraufhin, dass die Entwicklungsumgebung (g++) nachinstalliert werden muss.
Bitte
apt install build-essential
ausführen.
Gruß Thomas

Frigo759, 2024-06-01:
Vielen Dank!
Das war das Problem.

closed 2024-06-01 by Frigo759