pvtom/rscp2mqtt #73: SOC nicht korrekt

hauseroliver, 2024-06-03:
Erst einmal vielen, vielen Dank für rscp2mqtt, ist wirklich TOP!
Ich versuche gerade alles perfekt in Home Assistant abzubilden und bin schon recht weit.

Einzig beim Wert für den Ladestand des Akkus habe ich grobe Probleme.
Beim Auslesen von /battery/soc  bekomme ich ich schwankende Werte. Nach Start passt es und ist identisch mit dem Wert der E3DC am Display und dem modbus Protokoll, das ich akutell noch produktiv in Home Assistant verwende.
Nach ein paar Minuten fällt der Wert aber um ca. 10% und ich habe keine Ahnung warum.

Kann es sein, dass hier ein falscher Wert ausgelesen wird?
Wenn ich direkt über Modbus gehe, bekomme ich den korrekten Wert über address: 40082

Hoffe es kann mir hier jemand helfen.
Grüße
Oliver

[image]

pvtom, 2024-06-03:
Hallo Oliver,
es gibt noch andere SOC-Topics, u.a. e3dc/battery/soc_float.
Passen die Werte hier?
In Issue #44 ist ein ähnliches Problem beschrieben, dass ich bei meiner Anlage allerdings nie nachvollziehen konnte.
Gruß Thomas

hauseroliver, 2024-06-04:
Hallo Thomas,
danke für deine rasche Reaktion!
Ich konnte das Problem nun nach längerer Einlesezeit fixen.
Es war kein Problem deiner Werte sondern von meiner MQTT/Home Assistant Config.

Ich hatte in der .config den PREFIX geändert, dadurch haben die FORCE_PUB nicht gegriffen und mein Home Assistant hat dann immer auf einen alten Wert von battery/soc zugegriffen.
Jetzt wird der soc jede Sekunde geliefert und alles ist gut und korrekt.

Einzig den battery/soc_float bekomme ich nicht immer mit, aber das ist mir eigentlich egal.

Falls jemand ähnliche Probleme haben sollten:
Mit folgenden Settings schaut es nun sehr gut aus und ich kann die verschiedenen Komponenten (Home Assistant, EMQX MQTT Server, LXC mit rscp2mqtt) beliebig restarten und alle Werte sind immer sofort und korrekt da:

MQTT_QOS=1
MQTT_RETAIN=false
RETAIN_FOR_SETUP=true
PREFIX=e3dc
FORCE_PUB=e3dc/[a-z]+/power
FORCE_PUB=e3dc/battery/soc
FORCE_PUB=e3dc/pvi/.*/string_[1,2]
FORCE_PUB=e3dc/addon/power
FORCE_PUB=e3dc/wallbox/charge_battery_before_car
FORCE_PUB=e3dc/wallbox/discharge_battery_to_car
FORCE_PUB=e3dc/wallbox/discharge_battery_until

Ich werde eventuell die nächsten Tage ein Package für Home Assistant machen, das die wichtigsten Werte schön zu Sensoren und Schalter macht. Wenn jemand Interesse hat, kann ich das dann gerne teilen.

Oliver

bmause, 2024-06-04:
Klingt gut 👌

closed 2024-07-11 by pvtom