pvtom/rscp2mqtt #74: Min. Charge/Discharge Limit

genezig, 2024-06-04:
Hi there,

I noticed that in RSCPgui it is possible to change the min. charge/discharge limit. It seems to control the minimum power for the battery to be discharged or charged with. Since 2.2 kW is the minimum recommended charge power for the S10E (I remember the value, but not where I read it) I would be glad to be able to control that value. Would be valuable on sunny days with occasional clouds passing by resulting in fluctuations of the solar power.

rscp2mqtt provides the mqtt topic ems/discharge_start/power, but it does not provide the corresponding set/discharge_start_power topic to change this value.

Would be great if it could be implemented.

Best regards

pvtom, 2024-06-04:
Hi,
it should be possible with "Power Limits"...
[image]

Best regards
Thomas

genezig, 2024-06-04:
Thank you for the quick reply. The power limit values mentioned in your screenshot set the _maximum_ charge/discharge power. The value reported by ems/discharge_start/power is a _minimum_ value. Charging as well as discharging only takes place if the charging/discharging power is _above_ this value. I believe the standard value is 65 W and it is meant to save the battery or other electronic units from switching from charge to discharge and vice versa too often. I would be glad to be able to set that value as mentioned in my opening post.

Best regards

pvtom, 2024-06-04:
Okay, sorry, I didn't read that correctly. I have already adapted the tool and it works.
I will bring the solution with the next release - probably next weekend.
Best regards
Thomas

genezig, 2024-06-04:
At the speed of light - thank you, I really appreciate it!

Best regards

pvtom, 2024-06-08:
Release v3.25 is available.

closed 2024-07-07 by pvtom