pvtom/rscp2mqtt #75: Using e3dc/set/wallbox/number_phases not working anymore?

Dirk070, 2024-06-06:
Hi Tom,

I used your great tool with FHEM, an switching the number of phases (1, 3) worked with e3dc/set/wallbox/number_phases.

After updating the phases can not be switched with this command. Any changes I missed?

Thanks in advance.

Regards,
Dirk

pvtom, 2024-06-06:
Hi Dirk,
the subject of the output is e3dc/wallbox/number_phases (without “used”).
I do not understand the problem. Is it a new problem? Has it worked in the past?
I can't check the wallbox functionality as I don't have an E3/DC wallbox :-(
Do you use the current version of rscp2mqtt (v3.24)?

Best regards
Thomas

Dirk070, 2024-06-08:
Hallo Thomas,

jetzt wieder auf deutsch, mich hatte ein Post irrititiert und ich hatte auch viel zu wenige Infos geliefert, sorry.

Mit der Docker Version 3.8 hatte ich die Phasen zwischen 1 und 3 umschalten können: e3dc/set/wallbox/number_phases
Da war ja mal meine Anfrage an Dich, im November 2023, die Du netterweise umgesetzt hattest.

Aktuell nutze ich die Docker Version 3.24

Der identische Befehl funktioniert nun nicht mehr.

Weiterhin funktioniert e3dc/set/wallbox/disable_battery_at_mix_mode

Die Readings aus der Wallbox kommen nicht komplett.

Diese funktionieren z.B. bei mir (Definition in FHEM):
mqtt2server:e3dc/wallbox/status:.* Wallbox_Status
mqtt2server:e3dc/wallbox/power/total:.* Wallbox_TotalPower
mqtt2server:e3dc/wallbox/sun_mode:.* Wallbox_SunMode
mqtt2server:e3dc/wallbox/power/solar:.* Wallbox_SolarPower
mqtt2server:e3dc/wallbox/plugged:.* Wallbox_Plugged
mqtt2server:e3dc/wallbox/max_current:.* Wallbox_MaxCurrent
mqtt2server:e3dc/wallbox/power/L1:.* Wallbox_L1
mqtt2server:e3dc/wallbox/power/L2:.* Wallbox_L2
mqtt2server:e3dc/wallbox/power/L3:.* Wallbox_L3
mqtt2server:e3dc/wallbox/locked:.* Wallbox_Locked
mqtt2server:e3dc/wallbox/key_state:.* Wallbox_KeyState

Nicht funktionieren aber diese:
mqtt2server:e3dc/wallbox/number_used_phases:.* Wallbox_UsedPhases
mqtt2server:e3dc/wallbox/number_phases:.* Wallbox_NumberPhases

Die Menge der Phasen lässt sich also nicht mehr ändern und ich bekomme auch kein Reading dazu.
Daher die Überlegung, ob sich etwas geändert hat.

Melde Dich gerne, wenn ich noch Infos liefern kann.

Danke vorab und schöne Grüße
Dirk

pvtom, 2024-06-08:
Hallo Dirk,

es gab schon Änderungen von 3.8 bis 3.24. Dazu gibt es diverse Issues, wo die Änderungen ausgiebig diskutiert und getestet wurden.
So ist z.B. e3dc/wallbox/number_used_phases weggefallen, weil es die gleiche Information wie e3dc/wallbox/number_phases lieferte.

e3dc/set/wallbox/number_phases allerdings hat sich nicht geändert.
Ich habe ja keine E3/DC-Wallbox. Wenn ich den Befehl sende bekomme ich von der S10 eine Fehlermeldung zurück.
Heißt aber, dass der Befehl an das Hauskraftwerk rausgeht.

Kannst Du schauen, ob Du Fehlermeldungen bekommst?

Gruß Thomas

Dirk070, 2024-06-09:
Schaue ich gerne nach, schreibst Du in der Docker Version ein Log?
In FHEM bekomme ich aus MQTT keine Fehlermeldungen.

pvtom, 2024-06-09:
Hallo Dirk,
dazu musst Du das Logging in der .config mit LOG_MODE=BUFFERED einschalten.
Wenn Du rscp2mqtt mit Docker im Vordergrund (also nicht "detached") startest, dann den Befehl zur Änderung der Phasenzahl ausführst und anschließend rscp2mqtt abbrichst (crtl-C), wird eine Übersciht der Errors/Messages rausgeschrieben.
Gruß Thomas

Dirk070, 2024-06-21:
Hallo Thomas,

hier fehlt mir offenbar schlicht Erfahrung.

Per Portainer habe ich den Docker Container auf der Synology wie folgt konfiguriert:
Command:
'/opt/rscp2mqtt/rscp2mqtt'
Volumes:
/opt/rscp2mqtt/.config auf /volume1/docker/rscp2mqtt/.config
/tmp/rscp2mqtt.log auf /volume1/docker/rscp2mqtt/rscp2mqtt.log
/tmp/rscp2mqtt.history auf /volume1/docker/rscp2mqtt/rscp2mqtt.history

Die Files für Log und History habe ich per Kopie der .config erzeugt dort die Inhalte gelöscht.

Beim Start des Containers wird "Stdout to terminal" und "Log Level = 2 (BUFFERED)" protokolliert.
Auch wenn ich den log Level auf ON setze, werden die Log Files nicht gefüllt und auch im Protokoll des Containers kommt keine Info. 

Kannst Du mir helfen, damit ich Dir die notwendigen Infos zur Verfügung stellen kann?

Danke Dir und schöne Grüße
Dirk

Dirk070, 2024-06-21:
Erledigt. Offenbar lag die Ursache in einer meiner FHEM-Definitionen, die sich sehr unglücklich auf nur genau einen Parameter ausgewirkt hat. Ich entschuldige mich für den unnötigen Aufwand.