pvtom/rscp2mqtt #76: [Feature Request] Mehrere Wallboxen ohne Index ansprechen

lorbetzki, 2024-06-16:
Hallo,

Derzeit ist es so, das wenn ich mehrere Wallboxen habe, ich diese per Index umschalten muss. Wäre es möglich dieses Verhalten zu ändern?

Dazu würde ich in der config 
`
WB_INDEX=0
`
in bspw
`
WB_QUANTITY=2
`
ändern und rscp2mqtt würde dann automatisiert WB_INDEX=0 und WB_INDEX=1 abfragen und entsprechende Topics ausgeben?

Dann kann man auch die Boxen zum einem direkt steuern ohne vorher mit WB_INDEX umschalten zu müssen und hat auch die Werte wie Solar-/Netzlademenge pro Wallbox.

Viele Grüße

P.S. Dir fehlt ein Spenden Button ;)

pvtom, 2024-06-16:
Hallo Kristian,
das ist auch ganz klar die naheliegende Lösung :-), bei Batterien und PVI habe ich das so auch schon realisiert.
Vom Aufwand her habe ich mich damals bei Wallboxen für das Umschalten entschieden.
Mal schauen, was passiert, wenn ich Zeit habe und Programmierlust diesbezüglich bekomme, ...
Bis dahin kann man als Workaround natürlich rscp2mqtt mit einer anderen Konfiguration (anderes PREFIX, anderes WB_INDEX) ein weiteres Mal starten (rscp2mqtt -c name_der_anderen_konfig_datei). Da man dann viele Werte doppelt abfragt, kann man z.B. PM_REQUESTS, PVI_REQUEST, DCB_REQUESTS, etc. auf false setzen, um die Flut abzumildern.
Vielleicht hilft das erstmal weiter?!
Wie viele Wallboxen hast Du denn im Einsatz?
Gruß Thomas

lorbetzki, 2024-06-16:
das würde wahrscheinlich helfen... wobei ich die e3dc nicht stressen möchte, wenn zwei oder mehr rscp2mqtt mit min 2 sek Abfragezeit die Anlage befeuern, kann ich mir vorstellen, das dies zu Problemen führt... Ist aber derzeit nicht soooo wichtig

ich selber habe "nur" eine, aber in meinem Umfeld kenne ich welche, die zwei haben.

lorbetzki, 2024-07-20:
Hi,

kleine Rückmeldung, es läuft soweit sehr gut. Danke schonmal.

Man muss jedoch ein wenig um die ecke denken.

Hat man nur eine Wallbox, bekommt man die topics für die Wallbox ohne eine ID. Das ist OK.

Hat man nun zwei Wallboxen, gibt man in der rscp2mqtt .config 
```
WB_INDEX=0
WB_INDEX=1
``` 

ein, auch OK.

Aber man bekommt nun insgesamt drei Topics. Eine topic ohne ID (darin sind die Werte für ALLE Wallboxen), ein Topic für die erste wallbox ID 1 und ein drittes für die zweite Wallbox mit ID 2

Also bspw
e3dc/wallbox/energy/L1 # für alle Wallboxen kumuliert
e3dc/wallbox/1/enegery/L1 #  für die erste Wallbox
e3dc/wallbox/2/enegery/L1 # für die zweite Wallbox.

Alles auch OK, habe ich aber nicht direkt erwartet, mich wunderte woher die ID 2 kam. Aus der README habe ich es nicht direkt rausgelesen. 

Dennoch, es läuft sehr gut. Vielen Dank.

closed 2024-10-26 by pvtom