pvtom/rscp2mqtt #77: eher eine Frage, Daten in Influxdb V2

bmause, 2024-06-16:
Hallo zusammen, 

ich möchte her noch ein paar mehr Daten in der Influxdb V2 haben, bekommen es aber nicht mit dieser Einstellung hin.
Ich hätte gerne noch die temperature und die Spannung in V. die kommen aber nicht an, komischer Weise aber die Frequenz in HZ.
hier mal meine Einstellung, habt ihr dazu einen Idee oder Tipp für mich ich mir die Daten holen kann?
auch dieses Topic klappte nicht "#INFLUXDB_TOPIC=e3dc/#"

#-InfluxDB
#--if compiled with InfluxDB option
ENABLE_INFLUXDB=true
INFLUXDB_HOST=192.168.2.83
INFLUXDB_PORT=8086
INFLUXDB_VERSION=2
INFLUXDB_MEASUREMENT=e3dc
INFLUXDB_MEASUREMENT_META=e3dc_meta
INFLUXDB_TOPIC=e3dc/#/voltage
INFLUXDB_TOPIC=e3dc/#/temperature
INFLUXDB_TOPIC=e3dc/[a-z]+/power
#INFLUXDB_TOPIC=e3dc/[a-z]+/current
INFLUXDB_TOPIC=e3dc/[a-z]+/frequency

pvtom, 2024-06-16:
Hallo,
es sind so genannte "reguläre Ausdrücke", die für die Filterung zum Einsatz kommen.
So steht z.B. ".*" für mehrere Zeichen in Folge, "[a-z]+" für Zeichenketten, die aus kleinen Buchstaben bestehen (wie bei dem frequency-Eintrag).
D.h. INFLUXDB_TOPIC=e3dc/.* steht für alle Topics, die mit "E3DC/" beginnen.
Ich hoffe, ich konnte Dir helfen?

Gruß Thomas

bmause, 2024-06-16:
> INFLUXDB_TOPIC=e3dc/.

Hallo Thomas, 

versuche es zu verstehen, habe jetzt mal all INFLUXDB_TOPIC aaskommentiert mit einer "#" und nur dieses 
INFLUXDB_TOPIC=e3dc/.
gesetzt, neu gestartet, jetzt sollten doch alle e3dc Topic nach influxdb kommen.
Leider kommt nichts an, unter den MQTT Explorer kommen alle topic mit Werten.
Hast du Idee was ich noch falsch haben könnte ?

Danke vorab :-)

pvtom, 2024-06-16:
Sorry, es muss INFLUXDB_TOPIC=e3dc/.* (Punkt und Sternchen) heißen. Punkt bedeutet, dass jedes Zeichen erlaubt ist, Sternchen, dass es mehr als ein Zeichen sein kann.

Gruß Thomas

bmause, 2024-06-16:
ja, jetzt klappt es danke dir.

Gruß Björn

closed 2024-06-17 by bmause