pvtom/rscp2mqtt #79: Limiting SOC

JensHauser, 2024-06-19:
With my S10X, I´d like to make use of limiting SOC (Set the maximum SOC to limit charging the battery).

In the docs I found the tags
* e3dc/limit/charge/soc
* e3dc/set/limit/charge/soc
 set this one, but it doesn´t work.

I read "it uses the Idle Periods described previously". So, do I need to just switch that on at the device or also define idle periods?

Thanks a lot.

pvtom, 2024-06-19:
Hello Jens,
you must switch on "Idle Periods" / "Sperrzeiten" at your S10X device.
Nothing else.
Turn on the limiter for charging the battery up to i.e. 80%, use
mosquitto_pub -h mqtt_host -t "e3dc/set/limit/charge/soc" -m 80
If you like to set it also for the next days, use
mosquitto_pub -h mqtt_host -t "e3dc/set/limit/charge/durable" -m true
(if it's "false" (default) the limiter will only work for the current day)

Best regards
Thomas

JensHauser, 2024-06-20:
Hi Thomas,
thanks, it works like you described. I just missed to switch it on.
BR Jens

JensHauser, 2025-05-19:
One more question related to limit charging. As proposed, I used the advice from above. Charging power, regulary limited to 3000W, is now way slower after applying the SOC limiter, from 0 to 1200 W.

Say, over the day some percentages (could be up to 20%) charging is still done. Any idea, why the limiter is not fully applied?

pvtom, 2025-05-19:
Hallo Jens,
(ich mach's mal in deutsch)
Der SOC Limiter ist eine Funktion im rscp2mqtt, nicht im Hauskraftwerk.
Wenn als Lade-Limit 60% eingestellt wurde und die 60% SOC erreicht werden, setzt rscp2mqtt die Ladesperrzeit für den aktuellen Tag auf die Zeitspanne von 0:00 bis 23:59 Uhr. So sollte das Hauskraft dann das Laden stoppen. Wird der eingestellte SOC unterschritten, nimmt rscp2mqtt die Sperrzeit wieder raus.
Die Ladeleistung wird durch rscp2mqtt dabei nicht manipuliert.
Kannst Du prüfen, ob es andere Sperrzeitendefinitionen gibt?
Nutzt Du dyn. Tarife und AI 360? Hier werden Ladungen ja auch in andere Zeiten des Tages verschoben. Vielleicht kollidiert das dann mit Sperrzeit-Defintionen.
Zu Deinem zweiten Bild: Hattest Du den Limiter rausgenommen, bevor die Ladung in Richtung 65% ging?
Wozu hast Du denn das Limit im Laufe des Tages geändert (bezieht sich auf "I adjusted the SOC limited during the day ")?

Gruß Thomas

JensHauser, 2025-05-19:
Hallo Thomas,

AI 360/dyn. Tarife nutze ich nicht. Es gibt Sperrzeiten aus den vergangenen Tagen, ja. Wetterabhängiges Laden habe ich rausgenommen. Den Limiter habe ich Gestern heruntergesetzt, um zu Probieren. Seitdem unangetastet gelassen. Das ist ja das, was ich nicht verstehe :-(

VG Jens

pvtom, 2025-05-19:
Hallo Jens,
kannst Du die Sperrzeiten löschen?
Ich werde den SOC Limiter bei mir mal einschalten. Mit dem neuen Software-Stand des Hauskraftwerks (S10_2025_028) habe ich den Limiter noch nicht im Einsatz gehabt. Vielleicht gibt es da ja eine Unverträglichkeit.

Gruß Thomas

JensHauser, 2025-05-19:
Hallo Thomas,

danke für´s Forschen :-)
Ich lösche die Sperrzeiten (bis auf heute) und melde mich mit dem Ergebnis.

VG Jens

pvtom, 2025-05-20:
Moin,
kannst Du die Topics e3dc/idle_period/# beobachten?
Z.B. mit mosquitto_sub -h your_mqtthost -t "e3dc/idle_period/#" -F "%j"

Gruß Thomas

JensHauser, 2025-05-20:
Läuft prima, wenn ich das System einmal einstelle und unangetastet lasse:
[image]

Idle period:
[image]

Ich versuche, das Problem einzugrenzen.
Habe OpenHAB neu gestartet, da ich den Fehler nun eher in meiner Installation sehe. 
Vermutung #1: SOC limiter in % muß wohl neu gesetzt werden. Initial "0". Mache ich das, wird weiter geladen, aber mit geringerer Leistung.
Vermutung #2: Die Idle Periods werden scheinbar nicht mehr berücksichtigt. Obwohl noch in MQTT sichtbar
Vermutung #3: rscp2mqtt wird von Openhab übersteuert oder kommt durcheinander.

pvtom, 2025-05-20:
Hallo Jens,

ich kenne OpenHAB nicht und weiß nicht, was da Richtung rscp2mqtt gemacht wird.
Nutzt OpenHAB die SOC Limiter Funktionalität von rscp2mqtt oder setzt es selbst die Sperrzeiten?
Kannst Du mal "e3dc/limit/charge/soc" beobachten. Hier steht der SOC-Wert drin, den rscp2mqtt halten soll.
Nur ein "e3dc/set/limit/charge" kann den Wert ändern oder ein Tageswechsel, wenn "e3dc/set/limit/charge/durable" false ist.
(die Werte 0 und 100 bedeuten, dass der Limiter aus ist und die Batterie auf 100% geladen werden soll)

Mein Test von heute: Die eingestellten 80% wurden ohne "Flattern" gehalten. Dieser Sägezahn kam bei mir nicht vor... die Sonne hat prall geschienen, da musste nichts aus der Batterie kommen.

Du erwähnst oben den Wert "e3dc/idle_period/change". Das ist ein laufender Zähler, der bei Änderungen der idle_period-Werte immer um 1 erhöht wird. Das ist nicht der SOC-Wert!
Vielleicht wertet OpenHAB hier etwas falsch aus.

Gruß Thomas

JensHauser, 2025-05-21:
Hallo Thomas,

danke für Deine Hinweise, die haben mich auf folgende Spur gebracht:

`FORCE_PUB=e3dc/limit/.*`

war nicht aktiv. D.h., ich habe im MQTT Explorer und damit auch in OpenHAB immer nur den alten Wert gesehen. Nun bin ich besserer Hoffnung, werde es beobachten.

VG Jens

closed 2025-07-07 by pvtom