pvtom/rscp2mqtt #80: Beibehaltung der Begrenzereinstellung auch nach dem Tageswechsel ohne Funktion

lorbetzki, 2024-07-10:
Hallo,

wenn ich ein Ladelimit festlege, sagen wir mal 30%, wird dies nach dem Tageswechsel zurückgesetzt.
mosquitto_pub -h localhost -p 1883 -t "e3dc/set/limit/charge/soc" -m 30

diese Einstellung soll das verhindern.
mosquitto_pub -h localhost -p 1883 -t "e3dc/set/limit/charge/durable" -m true

nach dem Tagswechsel ist der Wert von
"e3dc/limit/charge/soc" auf 0%, der e3dc/set/limit/charge/durable weiterhin auf true. 

Das Ladelimit wird ignoriert. Ich schätze das ist ein Fehler.

Viele Grüße

pvtom, 2024-07-10:
Hi,

komisch, bei mir funktioniert das problemlos.

Wenn ich ins Log schaue 
mosquitto_pub -h localhost -p 1883 -t "e3dc/set/log/cache" -m 1
und
journalctl -f _SYSTEMD_UNIT=rscp2mqtt.service (ich nutze systemd)

sehe ich folgendes:

Jul 10 18:41:46 r7 rscp2mqtt[22543]: container >0< tag >0< index >0< topic >e3dc/limit/charge/soc< payload >80< unit >%<
...
Jul 10 18:41:46 r7 rscp2mqtt[22543]: container >0< tag >0< index >2< topic >e3dc/limit/charge/durable< payload >1< unit ><

D.h. ich habe eine Ladegrenze von 80% eingestellt und die läuft auch an den nächsten Tagen noch (e3dc/limit/charge/durable payload = 1 entspricht true).

Gruß Thomas

lorbetzki, 2024-07-10:
Hi,

danke für deine Rückmeldung. Du hast vollkommen recht, ich habe den Fehler bei meiner Erweiterung entdeckt. Sorry.

closed 2024-07-10 by lorbetzki