pvtom/rscp2mqtt #82: help update or portianer ?

bmause, 2024-07-26:


bmause, 2024-07-26:
Hello, I use rscp2mqtt and it helps me collect data via MQTT and InfluxDB V2. I'm not a programmer or a professional, so I find it quite challenging to keep rscp2mqtt up to date.

I once managed to install it under Proxmox and an LXC container, but I can't seem to get it working again. I've tried multiple times to install a Docker LXC with Debian and Portainer, but I can't get rscp2mqtt to work. The pull doesn’t work for me, and I can’t get it to work through Portainer either, even using a stack.

Currently, I keep deleting everything and manually following your installation instructions. Is there another way to do this? Or could you give me a tip to help me out? Ideally, I'd like to manage updates through Portainer.

Thanks in advance and sorry for the silly question :-)

pvtom, 2024-07-26:
[rscp2mqtt_mit_portainer.pdf](https://github.com/user-attachments/files/16395609/rscp2mqtt_mit_portainer.pdf)

bmause, 2024-07-26:
Hallo Thomas, 

Vielen Dank für die Anleitung, habe es damit hinbekommen.
Großes Lob 😀👌👍

pvtom, 2024-07-26:
Das freut mich. Ich habe rscp2mqtt dadurch heute das erste Mal mit dem Portainer betrieben und viel dabei gelernt.
Ich werde die Anleitung in das Repository übernehmen, dann haben andere auch etwas davon.
Übrigens kannst Du das Updaten in Kürze ausprobieren. Ich muss noch einen Fehler in der Software beheben und damit kommt dann ein neues Release ;-)
Gruß Thomas

bmause, 2024-07-27:
Update klappte so auch 😀

pvtom, 2024-08-04:
prima, ich schließe den Issue.
Gruß Thomas

closed 2024-08-04 by pvtom