pvtom/rscp2mqtt #84: Wallbox Mindest-Ladestrom

lorbetzki, 2024-08-03:
Hi,

im E3DC Portal und am Gerät kann man die "Min. Ladestrom" einstellen.
[image]

Meinst Du, es gibt eine Möglichkeit dies über RSCP einzustellen? 

Viele Grüße

pvtom, 2024-08-03:
Hallo Kristian,
ich habe einen Tag gefunden, der passen könnte.
Du kannst folgende Zeilen in die .config eintragen:

ADD_NEW_REQUEST=TAG_WB_REQ_DATA:TAG_WB_INDEX:0-0
ADD_NEW_REQUEST=TAG_WB_REQ_DATA:TAG_WB_REQ_MIN_CHARGE_CURRENT-1
ADD_NEW_TOPIC=TAG_WB_DATA:TAG_WB_MIN_CHARGE_CURRENT:A:1:0:wallbox/min_current

ADD_NEW_SET_TOPIC=TAG_WB_REQ_DATA:TAG_WB_REQ_SET_MIN_CHARGE_CURRENT:set/wallbox/min_current:^[0-9]{1,2}$-UChar8

Damit kann man die Wallbox mit Index 0 abfragen und setzen (hoffentlich, ich kann es ja nicht testen).

Wenn das funktioniert, baue ich das ins Tool direkt ein, damit auch weitere Wallboxen angesprochen werden können.

Welche Werte kann man am Gerät bzw. im Portal setzen 6 A bis 20 A, in Einerschritten?

Gruß Thomas

pvtom, 2024-08-04:
Hallo,
sorry, aber "ADD_NEW_SET_TOPIC=TAG_WB_REQ_DATA:TAG_WB_REQ_SET_MIN_CHARGE_CURRENT:set/wallbox/min_current:^[0-9]{1,2}$-UChar8" wird nicht funktionieren. Der String wird nicht richtig interpretiert.
Ich werde das mit dem kommenden Release fixen und dann gleich "min current" umsetzen.
Gruß Thomas

lorbetzki, 2024-08-04:
Super, danke. 

Im Portal kann man in 1A Schritten gehen, ab 6A geht es los und dann bis max. den eingestellten Ladestrom.

Viele grüße

pvtom, 2024-08-04:
v3.29 steht zur Verfügung. Hoffentlich funktioniert das Lesen und Setzen des min. Ladestroms:
- e3dc/wallbox/min_current
- e3dc/set/wallobox/min_current
- (und die Varianten für Multi-Wallbox-Umgebungen)

lorbetzki, 2024-08-04:
e3dc/wallbox/min_current
e3dc/set/wallobox/min_current

gerade an einer Easy Connect getestet und läuft. Super. Vielen Dank.

closed 2024-08-04 by lorbetzki

pvtom, 2024-08-04:
das freut mich!