pvtom/rscp2mqtt #87: unexpected error writing points to database:  Socket receive error. errno 104

zutewas, 2024-09-09:
Hallo,

mind. 1-2 mal am Tag werden keine Daten mehr via. MQTT übertragen und es kommt immer zu folgender Meldung:

`{"code":"internal error","message":"unexpected error writing points to database: timeout"}[2024-09-09 09:08:52] pid=1 ppid=0 RscpMqttMain.cpp(2634) Error: Socket receive error. errno 104`

Was könnte die Ursache sein bzw. wie kann das Problem weiter geprüft werden? Vielen Dank!

pvtom, 2024-09-09:
Hallo,
das sind zwei unterschiedliche Fehler (im Log nicht richtig getrennt):

1. {"code":"internal error","message":"unexpected error writing points to database: timeout"} kommt von der InfluxDB, die scheinbar Probleme beim Ablegen der Daten hat.
2. "Socket receive error. errno 104" ist ein Kommunikationsabbruch durch das Hauskraftwerk: 104 steht für "Connection reset by peer". Diesen Abbruch fängt rscp2mqtt nicht sauber ab und hängt sich nach Auftreten auf. Dieses Problem kann ich lösen, so dass nach Auftreten des Abbruchs ein Reconnect durchgeführt wird. -> v3.30 kommt dann in Kürze.

Warum die Abbrüche oder der Timeout bei der InfluxDB auftreten, kann ich nicht sagen. Eine wackelige Netzwerkanbindung vielleicht? Auf welcher Hardware läuft denn der Docker Container und die InfluxDB?

Gruß Thomas

zutewas, 2024-09-09:
Vielen Dank für die schnelle Rückmeldung.

InfluxDB (influxdb:latest) läuft ebenfalls als Docker mit rscp2mqtt auf einem Celeron J6412 Host. Beide Container sind im selben Docker-Netzwerk.

pvtom, 2024-09-14:
v3.30 ist jetzt verfügbar.
- errno 104 wird jetzt abgefangen. Das Programm hängt sich nicht mehr auf
- zum InfluxDB-Timeout wurde in der Doku eine Erhöhung der Batchsize empfohlen. Das habe ich ich umgesetzt. Hoffe, dass der Fehler nicht mehr auftritt

pvtom, 2024-10-26:
Ich hoffe, das hat sich erledigt.

closed 2024-10-26 by pvtom

zutewas, 2024-12-04:
Sorry für die sehr späte Rückmeldung. Es läuft bis jetzt stabil und ich hatte keine Probleme mehr. Vielen Dank für deine Arbeit.  :-)